armbian/build #436: building script error when building distro

LarsNorgaard, 2016-08-09:
Building Debian distro fails. The kernel builds fine without any errors, but if I want to build the complete distro it fails with this error:

```

[ o.k. ] Writing bootloader [ /dev/loop0 ]
[ error ] ERROR in function write_uboot [ common.sh:378 ]
[ error ] U-boot failed to install [ @host ]
[ o.k. ] Process terminated 
[ error ] ERROR in function unmount_on_exit [ debootstrap-ng.sh:549 ]
[ error ] debootstrap-ng was interrupted 
[ o.k. ] Process terminated 
lars@ubuntu:~$
```

I used a fresh install of Ubuntu 16.04.

JulesThuillier, 2016-08-09:
I just had the same problem and was writing and issue about it !
I am currently trying to build after changing line 377 in file common.sh for this one 

`write_uboot_platform "/tmp/u-boot/usr/lib/${CHOSEN_UBOOT}_${REVISION}_${ARCH}" "$loop"`

When you check the folder /tmp/usr/lib does not exists but /tmp/u-boot/usr/lib does. So I changed to that. Currently building... I'll keep you updated in 30mins ;)

JulesThuillier, 2016-08-10:
Succes ! 

To sum up here is the Fix :
Open file common.sh

Replace line 377
`write_uboot_platform "/tmp/usr/lib/${CHOSEN_UBOOT}_${REVISION}_${ARCH}" "$loop"`
For this one :
`write_uboot_platform "/tmp/usr/lib/${CHOSEN_UBOOT}_${REVISION}_${ARCH}" "$loop"`

Save and you're good ! :)

igorpecovnik, 2016-08-10:
Thanks, fixed.
https://github.com/igorpecovnik/lib/commit/1b8e6c194e31642f2d327fff63149fdfafa4d292

closed 2016-08-10 by igorpecovnik