armbian/build #437: ODROID C2 system didn't report CPU name and model

krzyszp, 2016-08-11:
less /proc/cpuinfo

"processor       : 0
BogoMIPS        : 2.00
Features        : fp asimd crc32
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 1
BogoMIPS        : 2.00
Features        : fp asimd crc32
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 2
BogoMIPS        : 2.00
Features        : fp asimd crc32
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 3
BogoMIPS        : 2.00
Features        : fp asimd crc32
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

Hardware        : ODROID-C2
Revision        : 020b
"

zador-blood-stained, 2016-08-11:
This info is provided by kernel, which is maintained by Hardkernel, not us (kernel repository is [here](https://github.com/hardkernel/linux/tree/odroidc2-3.14.y)). Since this problem does not cause any issues we probably won't touch this.

krzyszp, 2016-08-11:
I have reported it to Hardkernel now, but it is cause issues as applications won't work properly as didn't get proper data from system.
Right now, our BOINC client application doesn't work properly as server can't get right CPU info and can't attach work for it (Universe@Home project).

zador-blood-stained, 2016-08-18:
[This](http://forum.odroid.com/viewtopic.php?t=23101&p=155482) topic has some useful info on the problem:

- This format of `/proc/cpuinfo` is default for mainline arm64 kernel (even for 4.x versions)

- Software should not rely on parsing `/proc/cpuinfo` (so this is more BOINC fault and not OS/kernel)

- BOINC can be compiled with custom options to bypass this problem

Closing since there is nothing to fix on our (Armbian) side

closed 2016-08-18 by zador-blood-stained