armbian/build #438: Analog sound not working on NanoPi M1

multivac61, 2016-08-12:
I am not able to get analog sound out of the M1. Happens both when compiling my own kernel and using the `Armbian_5.14_Nanopim1_Ubuntu_xenial_3.4.112.raw` compressed image on your website. I have tried the core qt .img file from FriendlyARM and there the analog sound works great.

When I execute `aplay -l` on a freshly installed Ubuntu Xenial Server Armbian I get both the sunxi audiocodec and HDMI soundcards up as expected:
```
root@nanopim1:~# aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: audiocodec [audiocodec], device 0: SUNXI-CODEC sndcodec-0 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: sndhdmi [sndhdmi], device 0: SUNXI-HDMIAUDIO sndhdmi-0 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

but when I run `speaker-test -c2` or play some .wav file using `aplay` I get no sound from the audio jack.

The .fex file I have used is the default `nanopim1.fex` which has audio0 set as following:
```
[audio0]
audio_used = 1
lineout_vol = 31
cap_vol = 5
audio_hp_ldo = "none"
adcagc_used = 0
adcdrc_used = 0
dacdrc_used = 0
adchpf_used = 0
dachpf_used = 0
audio_pa_ctrl = port:PA16<1><default><default><0>
```

The [default .fex](https://github.com/friendlyarm/h3_lichee/blob/master/tools/pack/chips/sun8iw7p1/configs/nanopi-h3/sys_config.fex) config file for M1 in the FriendlyArm [h3_lichee](https://github.com/friendlyarm/h3_lichee) has audio0 configured as follow:

```
[audio0]
audio_used = 1
lineout_vol = 31
cap_vol = 5
audio_hp_ldo = "none"
adcagc_used = 0
adcdrc_used = 0
dacdrc_used = 0
adchpf_used = 0
dachpf_used = 0
;audio_pa_ctrl = port:PA16<1><default><default><0>
```

The only difference is the `audio_pa_ctrl` parameter which is commented out. There have been some [interesting commits](https://github.com/friendlyarm/h3_lichee/commit/d7f658426b06afb907f97ac3facf1acfd10ccc7e) in the h3_liche repo recently which might help to understand why the analog sound is not working on the Armbian. 

Please let me know how I can help getting sound to work.

ThomasKaiser, 2016-08-12:
Thanks for the description and ruling out hardware issues. Can you please upload the output of `sudo sunxi-pio -m print` from both Armbian and FA's OS image to pastebin.com or the like?

`sunxi-pio` is already installed in Armbian, if it's not available in FA's OS image please try whether you can simply install our [sunxi-tools_1.3-1_armhf.deb](http://apt.armbian.com/pool/main/s/sunxi-tools/sunxi-tools_1.3-1_armhf.deb) (download, then install using `sudo dpkg -i`)

multivac61, 2016-08-12:
Not a problem, very happy to help. Here are the dumps of the sunxi-pio command:

[armbianm1_sunxi-pio.log](http://pastebin.com/BrL3kLfv)
[fam1_sunxi-pio.log](http://pastebin.com/7Lp6nNXY)

ThomasKaiser, 2016-08-12:
Well, I have an M1 since 2 days and not even booted. Also my interest in audio is 0 and I already waste too much time on these H3 devices trying out other stuff. And unfortunately I don't get what FA's code change does wrt audio (I only see that they seem to need `uart3` on their new NanoPi Air to connect to AP6212 and then PA15/PA16 are conflicting (and am thinking now about why analog audio out is missing on Banana Pi M2+ ;) ).

Anyway, just a shot in the dark: do things improve when you do a `sudo sunxi-pio -m PH16'<7><0><1>' `?

zador-blood-stained, 2016-08-12:
@multivac61
Can you please attach full dmesg log from Armbian image and FA image and also output of `amixer` on both images?

multivac61, 2016-08-12:
@ThomasKaiser `sudo sunxi-pio -m PH16'<7><0><1>'` did not seem to help.

@zador-blood-stained Sure thing:
## Armbian image
[armbian_dmesg.log](http://pastebin.com/qL90cskG)
[armbian_amixer.log](http://pastebin.com/gNNq9Rpm)

## FA image
[fa_dmesg.log](http://pastebin.com/bjPn9wyJ)
[fa_amixer.log](http://pastebin.com/ajf37cXq)

multivac61, 2016-08-12:
There is a big difference between the two amixer logs. I noticed that you are using h3_lichee as the main kernel. Could it be that the file [sun8iw7_sndcodec.c](https://github.com/friendlyarm/h3_lichee/blob/d7f658426b06afb907f97ac3facf1acfd10ccc7e/linux-3.4/sound/soc/sunxi/audiocodec/sun8iw7_sndcodec.c) is somehow not being added/called/compiled correctly when the kernel is compiled?

zador-blood-stained, 2016-08-12:
> Could it be that the file sun8iw7_sndcodec.c is somehow not being added/called/compiled correctly when the kernel is compiled?

You could compare kernel configs on both images (i.e. from `/proc/config.gz`), or at least post here one from FA image.

multivac61, 2016-08-12:
The kernel configs don't exist anywhere on the FA filesystem (running `find / -name config.gz` return nada). Further running `modprobe configs` returns the error _modprobe: FATAL: Module config not found_.

I have however used the h3_lichee repo to compile some kernel modules before. [Here is the .config file](http://pastebin.com/gdrFxJyD) that was genereted when I compiled for `sun8iw7p1`

multivac61, 2016-08-13:
I have located a problem. When I went into `alsamixer` I noticed that a parameter called "Audio lineout" was set to false. In the TUI I was not able to changed the parameter but I was able to save the state in a file by using `alsactl --file ~/.config/asound.state store`. In the .state file I manually changed the "Audio lineout" parameter to true and reloaded the alsamixer with the ``alsactl --file ~/.config/asound.state reload` command. Now `speaker-test -D sysdefault:CARD=audiocodec -twav -c2` plays perfectly.

You should be able to solve this issue by storing different default settings for ALSA.

zador-blood-stained, 2016-08-13:
> You should be able to solve this issue by storing different default settings for ALSA.

You can try extracting [this](https://github.com/igorpecovnik/lib/files/416589/asound.zip) `/var/lib/alsa/asound.state` file from FA image to Armbian image to the same path and check if this solves the problem (after reloading ALSA configuration)

multivac61, 2016-08-13:
Extracting the FA asound.state does give me some erroneous output in the shell:

> No state is present for card U25
> Found hardware: "USB-Audio" "USB Mixer" "USB1397:01b0" "" ""
> Hardware is initialized using a generic method
> No state is present for card U25

This does solve the problem however regardless of the errors.

To make the analog output the system default you will need to have the file `/etc/asound.conf` configured as follows:

```
pcm.!default {
  type hw
  card 0
  device 0
}
ctl.!default {
  type hw
  card 0
}
```

igorpecovnik, 2016-08-16:
I don't get sound out on Armbian with this config or manually setting the right card.

zador-blood-stained, 2016-08-16:
> I don't get sound out on Armbian with this config or manually setting the right card.

Only with `asound.conf` or you tested `asound.state` too?

wuweidong0107, 2016-08-17:
pls pay attention to these file:

wuweidong0107, 2016-08-17:
guys, pls pay attention to these file:
/var/lib/alsa/asound.state
/etc/asound.conf
sys_config.fex

multivac61, 2016-08-29:
@zador-blood-stained Using a fresh image I was able to get analog sound working by extracting the .state files you provided. This issue should be closed.

zador-blood-stained, 2016-10-16:
Can anyone please test if this is resolved in latest images?

ThomasKaiser, 2016-10-23:
@multivac61: Any news on this? Did you try it with a fresh Armbian image in the meantime?

multivac61, 2016-10-23:
@ThomasKaiser I will check this tomorrow and report (y).

multivac61, 2016-10-24:
I tried `speaker-test` on a fresh beta image `Armbian_5.23.161024_Nanopim1_Ubuntu_xenial_3.4.112.img`. Works straight out of the box, nice work guys!

zador-blood-stained, 2016-10-24:
@multivac61 
Thanks for testing!

closed 2016-10-24 by zador-blood-stained