armbian/build #442: Problem in the latest commit 7598c44 : resize screen

Paul75, 2016-08-18:
Hello,

I have a problem with the latest version of: when I did a "h3disp" to take the screen right dimention I get an error saying that it can convert bin because fex not found.

Thank you for your help.

igorpecovnik, 2016-08-18:
Are you using Armbian and if you do which version? Which image? Self build?

Paul75, 2016-08-18:
Yes use armbian, 5.17 jessie default desktop. Yes I self do image. On board Orange Pi plus 2E

igorpecovnik, 2016-08-18:
We are doing some changes those days and we could break something. Will take a look.

Paul75, 2016-08-18:
thanks ! in wait I take the image already done and it works.

igorpecovnik, 2016-08-18:
Yes, this is actually a cosmetic issue.

Paul75, 2016-09-02:
I must use image and not compile ?

igorpecovnik, 2016-09-02:
You need to compile - we haven't update any images yet. Soon.

Paul75, 2016-09-02:
but with the new compile the h3disp not works....

WereCatf, 2016-09-02:
@igorpecovnik  It would appear that sunxi-tools are missing from the images. Certainly, doing `find -iname "bin2fex"` in /usr doesn't return a thing, nor does the same for fex2bin. Can't say for Jessie, but this does apply at least to Xenial 16.04 - image with desktop.

zador-blood-stained, 2016-09-02:
@WereCatf 
sunxi-tools are now provided from apt.armbian.com repository or via new packages building process. It's not enabled by default yet.

zador-blood-stained, 2016-09-16:
Should be fixed with 5.20 release.

closed 2016-09-16 by zador-blood-stained