armbian/build #443: 5" HDMI LCD (800x480) non-functional on H3

WereCatf, 2016-08-20:
I recently bought an LCD similar to the one here http://www.waveshare.com/wiki/5inch_HDMI_LCD thinking it'd be a good companion for my Orange Pis. The display sorta-kinda works when enabling DVI-mode over HDMI, but you only get a portion of the image visible. I tried to fiddle with the HDMI-timings, but, quite frankly, I'm totally out of my depth there and couldn't make things any better. I also tried the settings from http://www.orangepi.org/orangepibbsen/forum.php?mod=viewthread&tid=444 but no dice. The display does work just fine on Raspberry Pi, so the display isn't broken.

Now, as far as I understand it, this is on the TODO-list already, but I am opening this ticket as kind of a reminder and to let anyone more knowledgeable to know that I have one of these display and I can test things with it.

igorpecovnik, 2016-08-20:
Check this, it might be helpful.
http://trustfarm.net/index.php/2016/08/17/orangepi-plus-2e-with-5inchlcd/
I discover it today as a ping back to my blog :)

WereCatf, 2016-08-20:
I shall take a look and report back if I can make it work.

WereCatf, 2016-08-20:
I just checked it and, well, all those patches already exist in the Armbian-tree. That is to say, it's still a no-go.

zador-blood-stained, 2016-11-30:
@WereCatf 
Please check if your screen works with new u-boot (it should work if it provides EDID data)
U-boot display driver now should support all kinds of non-standard resolutions.

WereCatf, 2016-11-30:
@zador-blood-stained Uboot does get the display-settings right, the display actually works fine. Once it loads up the kernel the display goes all bollocks-up again, though. This is with the legacy-kernel, though, I'll test the mainline in a bit.

WereCatf, 2016-11-30:
Hmm, no wonder h3disp isn't working, bin2fex reports "Malformed data" on orangepipc.bin

WereCatf, 2016-11-30:
No better luck with mainline-kernel.

zador-blood-stained, 2016-11-30:
@jernejsk said that u-boot display configuration can be passed to mainline simplefb driver. So eventually this should be resolved one way or the other.

jernejsk, 2016-11-30:
U-Boot simplefb support for H3 is already in my repo. Kernel side patches are included in above link. There is everything you need for testing that already. Only issue now is what to do with DRM driver if it is included. I guess using U-Boot DT patching ability to disable it is one way...

WereCatf, 2016-11-30:
Parsing my display's EDID multiple different ways I'm finding that it's an "ADA Model 4 Serial Number 1", I get a `Modeline        "Mode 0" 32.00 800 840 888 928 480 493 496 525 -hsync -vsync` and `HDMI:EDID failed to find a matching detail format for 800x480p hfp:40 hs:48 hbp:40 vfp:13 vs:3 vbp:29 pixel clock:32 MHz`

Alas, I don't know how to turn any of that into anything useful.

jernejsk, 2016-11-30:
Nice thing about simplefb is that you just doesn't care anymore about timings. Be sure to include kernel patches zador mentioned, compile image and that should generally be it. I'm still not sure about disabling DRM driver though.

zador-blood-stained, 2016-12-01:
I simply disabled DRM driver and added simplefb patch: 85bb349c70f5037a310ca74eee8e30e33d2c4b01
Works for me on OPi One.

So the only thing left is patching the legacy kernel 😞

WereCatf, 2016-12-01:
Does DPMS work with simplefb? If not, then that's enough of a reason to only consider it a short-term workaround, IMHO.

jernejsk, 2016-12-01:
No. Simplefb explanation is: U-Boot gave me some memory region. If I write into it, something will pop out on monitor.

jernejsk, 2016-12-01:
I forgot to mention two things:
- simplefb is basically good enough just for early fb support and boot screens
- don't make it as main display driver on legacy. It won't work together with libvdpau-sunxi and probably also with mali driver. A lot of users will get upset.

zador-blood-stained, 2016-12-01:
@jernejsk 
I know about this. We don't touch legacy yet, and I'm even not sure you can easily pass framebuffer configuration to legacy kernel. By "Patching the legacy kernel" I meant implementing EDID and custom modes support for disp2

jernejsk, 2016-12-01:
Oh, I'm losing interest in legacy more and more. Once proper way is determined by mainline guys and first work is done, I will switch.

zador-blood-stained, 2016-12-01:
> Once proper way is determined by mainline guys and first work is done, I will switch.

I'm not sure Cedrus/video decoding stuff will be added soon enough (if you are planning to switch OpenELEC to mainline), but other than that mainline kernel with all the patches looks good enough.

WereCatf, 2016-12-01:
I did get a random crash on boot with the megous-fork, so I'm not sure I'd exactly consider it stable yet.

jernejsk, 2016-12-01:
I have some hope that request API will land as planed (January if I'm not mistaken) First codecs are already implemented for A10 and A20. I guess porting to H3 should not be hard. But yes, this will definetly take some time.

zador-blood-stained, 2017-01-06:
I guess for the original topic (HDMI LCD support) @jernejsk's proposal [here](https://github.com/igorpecovnik/lib/issues/594#issuecomment-270881555) should allow getting correct values for the legacy driver, and mainline should work fine enough out of the box, so this can be closed.

closed 2017-01-06 by zador-blood-stained