armbian/build #446: can't build pine64

kometchtech, 2016-08-21:
Error after trying to build a pine64 SD images in the following options.

`$ sudo ./compile.sh BUILD_DESKTOP=no KERNEL_ONLY=no BOARD=pine64 BRANCH=default RELEASE=xenial PROGRESS_LOG_TO_FILE=yes PROGRESS_DISPLAY=plain`

```
  LD      drivers/video/sunxi/disp2/built-in.o
  LD      drivers/video/sunxi/built-in.o
  LD      drivers/video/built-in.o
  LD      drivers/built-in.o
  CC      examples/standalone/stubs.o
  CC      examples/standalone/hello_world.o
  LD      examples/standalone/libstubs.o
  LD      examples/standalone/hello_world
  OBJCOPY examples/standalone/hello_world.srec
  OBJCOPY examples/standalone/hello_world.bin
  LD      u-boot
  OBJCOPY u-boot.srec
  OBJCOPY u-boot.bin
./tools/add_hash_uboot.sh -f u-boot.bin -m uboot
 build_uboot 
‘u-boot.bin’ -> ‘u-boot-sun50iw1p1.bin’
make[1]: Leaving directory `/home/kometch/sources/u-boot-pine64/master/u-boot-pine64'
[ error ] ERROR in function compile_uboot [ common.sh:96 ]
[ error ] U-boot file not found [ u-boot-with-dtb.bin ]
[ o.k. ] Process terminated 
```
the build environment is as follows.
```
$ lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 14.04.5 LTS
Release:        14.04
Codename:       trusty
```
[logs-21_08_2016-22_26_48.zip](https://github.com/igorpecovnik/lib/files/428870/logs-21_08_2016-22_26_48.zip)

martinayotte, 2016-08-21:
I gave it a try, and saw that "boot0img" is crashing (which it is not shown in the above log) 

`make[1]: Leaving directory `/home/martin/git-work/igorpecovnik-armbian/sources/u-boot-pine64/master/u-boot-pine64'
./boot0img -B blobs/boot0.bin -s blobs/scp.bin -d arm-trusted-firmware/build/sun50iw1p1/debug/bl31.bin -u u-boot-pine64/u-boot.bin -e -F dt.dtb -o u-boot-with-dtb.bin
U-Boot: u-boot-pine64/u-boot.bin: 725952 Bytes
FDT   : dt.dtb: 68992 Bytes
make: *** [make_image] Segmentation fault (core dumped)
[ error ] ERROR in function compile_uboot [ common.sh:96 ]
[ error ] U-boot file not found [ u-boot-with-dtb.bin ]
[ o.k. ] Process terminated 
`
I still don't know the reason of that crash.

martinayotte, 2016-08-21:
I think I've found the error !
In boot0img.c line # 280, the memset was using the old ptr instead of the reallocated one in ret : 
`
memset((char*)ret + *sizeptr, 0, newsize - *sizeptr);
`

martinayotte, 2016-08-21:
Zador has merged the fix !
https://github.com/zador-blood-stained/u-boot-pine64-armbian/issues/1
You can retry your build ...

kometchtech, 2016-08-22:
build through.
Thank you very much.

Be reported as a separate issue is accessible via SSH from the outside, better?

martinayotte, 2016-08-22:
Glad to hear that your build succeeded !

About SSH, what do you mean ?

kometchtech, 2016-08-22:
I'd forget to take it up, cannot connect with SSH from the outside and thought from the UART console login, confirmed the openssh-server package were not available.

This is the package list error?

martinayotte, 2016-08-22:
I'm usually building Mainline, and SSH daemon is always there.
I will have to try a Legacy to check, but I think that would be odd if it is not there, or maybe this is an issue specific to PineA64.
I will report back ...

kometchtech, 2016-08-22:
Thank you in advance.

kometchtech, 2016-08-22:
The installation stage, had become as follows.
Is this Will is bad my build environment?

```
root@pine64:~# dpkg -l | grep openssh                                           
ii  openssh-client                     1:7.2p2-4ubuntu2.1                      s
iF  openssh-server                     1:7.2p2-4ubuntu2.1                      s
ii  openssh-sftp-server                1:7.2p2-4ubuntu2.1                      s
```

martinayotte, 2016-08-22:
Yes ! The package is installed but daemon failed to start, giving the "Privilege separation user sshd does not exist", error.
Digging the Web, I've figured out that it was simply an error in /etc/passwd where ssh user was member of the wrong group, it should match the ssh group found in /etc/group.
Then, a simple "/etc/init.d/ssh restart" make it worked successfully.

kometchtech, 2016-08-22:
Anyway, log in from once Console, it is that there is a need to run the following command?
`/etc/init.d/ssh restart`

martinayotte, 2016-08-22:
That only required after fixing the group, just to make sure the fix is working, then at next reboot, it will be already started.

martinayotte, 2016-08-22:
now closing the issue since it is now fixed ...

closed 2016-08-22 by martinayotte

kometchtech, 2016-08-27:
Seems to fail to restart the service.
```
kometch@pine64:~$ sudo /etc/init.d/ssh restart                                  
Restarting ssh (via systemctl): ssh.serviceJob for ssh.service failed because t.
 failed!    
```
```
kometch@pine64:~$ sudo systemctl status ssh -l                                  
��● ssh.service - OpenBSD Secure Shell server                                   
   Loaded: loaded (/lib/systemd/system/ssh.service; disabled; vendor preset: ena
   Active: failed (Result: exit-code) since Sat 2016-08-27 14:02:53 UTC; 20s ago
  Process: 3384 ExecStart=/usr/sbin/sshd -D $SSHD_OPTS (code=exited, status=255)
 Main PID: 3384 (code=exited, status=255)                                       
                                                                                
Aug 27 14:02:52 pine64 systemd[1]: Starting OpenBSD Secure Shell server...      
Aug 27 14:02:53 pine64 sshd[3384]: Privilege separation user sshd does not exist
Aug 27 14:02:53 pine64 systemd[1]: ssh.service: Main process exited, code=exited
Aug 27 14:02:53 pine64 systemd[1]: Failed to start OpenBSD Secure Shell server. 
Aug 27 14:02:53 pine64 systemd[1]: ssh.service: Unit entered failed state.      
Aug 27 14:02:53 pine64 systemd[1]: ssh.service: Failed with result 'exit-code'.
```

martinayotte, 2016-08-27:
As I've mentioned above, did you check the user/group in /etc/passwd ?
Because when I've also faced the issue of "Privilege separation", I've simply fixed the wrong group membership and restart the service and the problem was gone.

zador-blood-stained, 2016-08-27:
@kometchtech 
If you want to build images for Xenial target OS, please upgrade your build environment to Xenial first. AFAIK debootstrap process may not work correctly in this combination (Trusty host and Xenial target)

kometchtech, 2016-08-27:
@zador-blood-stained 

Thanks for the info.
Certainly the build was done at 14.04.
I would like to try at 16.04.

martinayotte, 2016-08-27:
Personally, the target I've built was for Jessie.
I presume the "Privilege separation" issue isn't related to specific target since I've faced it.
Did you check the user/group in /etc/passwd for the sshd user ?