armbian/build #450: [pine64][dev] fixed the link speed is not

kometchtech, 2016-08-27:
**issue**
experiencing the symptoms cannot be fixed 100 M Ethernet speed, as the result of ethtool.
`# ethtool -s eth0 autoneg on speed 100 duplex full `  
```
# ethtool eth0                                                     
Settings for eth0:                                                              
        Supported ports: [ TP MII ]                                             
        Supported link modes:   10baseT/Half 10baseT/Full                       
                                100baseT/Half 100baseT/Full                     
                                1000baseT/Half 1000baseT/Full                   
        Supported pause frame use: No                                           
        Supports auto-negotiation: Yes                                          
        Advertised link modes:  100baseT/Full                                   
        Advertised pause frame use: No                                          
        Advertised auto-negotiation: Yes                                        
        Link partner advertised link modes:  10baseT/Half 10baseT/Full          
                                             100baseT/Half 100baseT/Full        
                                             1000baseT/Half 1000baseT/Full      
        Link partner advertised pause frame use: Symmetric                      
        Link partner advertised auto-negotiation: Yes                           
        Speed: 1000Mb/s                                                         
        Duplex: Full                                                            
        Port: MII                                                               
        PHYAD: 1                                                                
        Transceiver: external                                                   
        Auto-negotiation: on                                                    
        Current message level: 0x00000000 (0)                                   
                                                                                
        Link detected: yes
```
**env**
```
# cat /etc/os-release                                              
NAME="Ubuntu"                                                                   
VERSION="16.04.1 LTS (Xenial Xerus)"                                            
ID=ubuntu                                                                       
ID_LIKE=debian                                                                  
PRETTY_NAME="Ubuntu 16.04.1 LTS"                                                
VERSION_ID="16.04"                                                              
HOME_URL="http://www.ubuntu.com/"                                               
SUPPORT_URL="http://help.ubuntu.com/"                                           
BUG_REPORT_URL="http://bugs.launchpad.net/ubuntu/"                              
UBUNTU_CODENAME=xenial 
```
```
# uname -a                                                         
Linux pine64 4.7.0-pine64 #1 SMP PREEMPT Sat Aug 27 21:17:52 JST 2016 aarch64 ax
```

zador-blood-stained, 2016-08-27:
Pine64 support in mainline is still experimental, and H3/A64 Ethernet driver is probably not complete, so it's normal that is doesn't support some features (and I'm not sure what driver revision is used in this kernel tree).

martinayotte, 2016-08-27:
My PineA64 has the networking working fine, but defaulted to 100Mb/s, probably because it is currently connected to a 100Mb/s switch.

kometchtech, 2016-08-27:
Official u-boot problem, has confirmed that 1Gbps is not stable.
Easy it is still connected to the 100Mbps switch.

ThomasKaiser, 2016-08-28:
Just FYI: I get pretty constant 435 Mbits/sec in Pine64 -> client direction and 620 Mbits/sec with slight fluctuations in the other direction. With 4.7 -- with 3.10.102 it's  906 and 940 Mbits/sec instead.

kometchtech, 2016-08-28:
@ThomasKaiser 

Thanks for the info.

When trying to communicate at 1 GB/s, and have a problem that I use is Pine64 plus this Ethernet and packet loss is.

zador-blood-stained, 2016-09-10:
`ethtool` problem is most likely caused by H3/A64 Ethernet driver in mainline not having such features yet, so there is nothing we can do. 

As for packet loss - it is possible that your board is defective: http://forum.armbian.com/index.php/topic/1917-armbian-running-on-pine64-and-other-a64h5-devices/page-2#entry15173

ThomasKaiser, 2016-09-11:
Mikhail's right, please check your board with legacy image, if there packet losses happen too but the whole thing can be cured with ethtool there, then your board most probably is defective. In case you have a forum account at pine64.org please send TL Lim a PM and ask for refund/return.

Using their forum to discuss the Ethernet problem is useless since there are a few guys active constantly censoring everything they don't understand. Watch what happened to [post 60](http://forum.pine64.org/showthread.php?tid=835&pid=19510#pid19510) for example. Before:
[image]

And after:
[image]

amc2002, 2016-09-12:
The moderator in that forum is totally and completely out of control.

ThomasKaiser, 2016-09-13:
> The moderator in that forum is totally and completely out of control.

He clearly don't know what he's doing (censorship, preventing any progress), but he did it again. Before:
[image]

And now:
[image]

He deleted several post (not only) of mine and edited the relevart parts of others away.

Since it's confirmed that at least on one Pine64+ the GbE symptoms are related to PSU/voltage (see [here](http://forum.armbian.com/index.php/topic/1917-armbian-running-on-pine64-and-other-a64h5-devices/?p=15173) or [here](http://forum.armbian.com/index.php/topic/1917-armbian-running-on-pine64-and-other-a64h5-devices/?p=15187) or [this as the most extreme example since with mainline kernel using the wrong PSU means no access to the GbE PHY at all and not just errors](https://irclog.whitequark.org/linux-sunxi/2016-09-09#17505162) it would be great if Pine64+ users that *do not* use any Fast Ethernet work-arounds (destroying their cables, using `ethtool` to force 100 Mbits/sec) start to measure `PHY_VDD33`.

Regarding test point please see [Armbian forum](http://forum.armbian.com/index.php/topic/1917-armbian-running-on-pine64-and-other-a64h5-devices/?p=15345). @kometchtech are you able to do some measurements and report back in the forum? (I fear it's not possible to try to do anything useful in pine64 forum as long as this person is enabled to censor everything away that's beyond his imagination)

kometchtech, 2016-09-13:
@ThomasKaiser 

Thank you for information.

Since English is not good, for it does not have confidence you are able to read well, may be the recognition that I use the armbian image of mailline kernel is to understand, and may if it is possible to apply a load more than 5.2V Is not it?

Then, the benchmark using the iperf, do you report to Armbian forum?

Best regards .

ThomasKaiser, 2016-10-04:
This specidic Ethernet issue can be considered a hardware flaw that needs physical replacement of the affected boards. Please check the [appropriate thread in Pine64 forum](http://forum.pine64.org/showthread.php?tid=835&pid=20702#pid20702) and ask their for refund/replacment.

closed 2016-10-04 by ThomasKaiser