armbian/build #452: [Help] Could not build Odroid C2 Xenial image with desktop

api-walker, 2016-08-28:
Hello,
I tried building an image for the Odroid C2, but everything fails due u-boot problems. I tried with Jessie and Xenial.
# Log
Displaying message: Preparing host info
Displaying message: Build host OS release xenial info
Displaying message: Downloading toolchain gcc-linaro-4.9-2016.02-x86_64_aarch64-linux-gnu info
Displaying message: Verifying
Displaying message: Extracting
Displaying message: Download complete  info
Displaying message: Downloading toolchain gcc-linaro-4.9-2016.02-x86_64_arm-linux-gnueabi info
Displaying message: Verifying
Displaying message: Extracting
Displaying message: Download complete  info
Displaying message: Downloading toolchain gcc-linaro-4.9-2016.02-x86_64_arm-linux-gnueabihf info
Displaying message: Verifying
Displaying message: Extracting
Displaying message: Download complete  info
Displaying message: Downloading toolchain gcc-linaro-5.2-2015.11-2-x86_64_arm-linux-gnueabihf info
Displaying message: Verifying
Displaying message: Extracting
Displaying message: Download complete  info
Displaying message: Downloading toolchain gcc-linaro-arm-linux-gnueabihf-4.8-2014.04_linux info
Displaying message: Verifying
gpg: WARNUNG: Unsicheres Besitzverhältnis der Konfigurationsdatei `/home/developer/.gnupg/gpg.conf'
gpg: error reading key: Öffentlicher Schlüssel nicht gefunden
gpg: WARNUNG: Unsicheres Besitzverhältnis der Konfigurationsdatei `/home/developer/.gnupg/gpg.conf'
gpg: Ausführen von externen Programmen ist ausgeschaltet, da die Dateirechte nicht sicher sind
gpg: Schlüsselserver-Datenübertragunsfehler: Allgemeiner Fehler
gpg: Empfangen vom Schlüsselserver fehlgeschlagen: Allgemeiner Fehler
gpg: WARNUNG: Unsicheres Besitzverhältnis der Konfigurationsdatei `/home/developer/.gnupg/gpg.conf'
gpg: die unterzeichneten Daten sind wohl in 'gcc-linaro-arm-linux-gnueabihf-4.8-2014.04_linux.tar.xz'
gpg: Unterschrift vom Mi 16 Apr 2014 20:29:14 CEST mittels RSA-Schlüssel ID 8F427EAF
gpg: Unterschrift kann nicht geprüft werden: Öffentlicher Schlüssel nicht gefunden
Displaying message: Verification failed  wrn
## BUILD CONFIGURATION
Kernel configuration:
Repository: https://github.com/hardkernel/linux
Branch: branch:odroidc2-3.14.y
Config file: linux-odroidc2-default

U-boot configuration:
Repository: https://github.com/hardkernel/u-boot.git
Branch: branch:odroidc2-v2015.01
Offset: 1
Size: 64

CPU configuration:
500000 - 2016000 with interactive
28.08.2016 12:46:15 Armbian 5.17 Odroidc2 Ubuntu xenial default
Build script version: 8762cf1a73c9b987641691460674335230c723be
Displaying message: Starting Armbian build script @host info
Displaying message: Building Armbian 5.17 Odroidc2 Ubuntu xenial default info
Displaying message: Syncing clock host info
Displaying message: Downloading sources  info
Displaying message: Checking git sources u-boot-odroidc2 odroidc2-v2015.01 info
Displaying message: Creating local copy
Displaying message: Fetching updates
Displaying message: Checking out
Displaying message: Checking git sources linux-odroidc2 odroidc2-3.14.y info
Displaying message: Creating local copy
Displaying message: Fetching updates
Displaying message: Checking out
Displaying message: ... creating a shallow clone sunxi-display-changer info
Displaying message: Cleaning u-boot-odroidc2/odroidc2-v2015.01 info
Displaying message: Cleaning linux-odroidc2/odroidc2-3.14.y info
Displaying message: Cleaning output/debs for odroidc2 default info
Displaying message: Started patching process for u-boot u-boot-odroidc2-default 2015.01 info
Displaying message: Looking for user patches in userpatches/u-boot/u-boot-odroidc2-default info
Displaying message: ... clean-sdfuse.patch succeeded info
Displaying message: ... firmware-override-toolchain-version.patch succeeded info
Displaying message: Compiling uboot 2015.01 info
Displaying message: Compiler version aarch64-linux-gnu-gcc 4.9.4 info
Displaying message: ERROR in function compile_uboot common.sh:96 err
Displaying message: U-boot file not found u-boot.bin err
Displaying message: Process terminated  info

# MyOS
Distributor ID: Ubuntu
Description:    Ubuntu 16.04.1 LTS
Release:        16.04
Codename:       xenial

zador-blood-stained, 2016-08-28:
Should be fixed now, please try again.

api-walker, 2016-08-28:
Sorry. Same error.

zador-blood-stained, 2016-09-01:
Should be fixed with 95aa647ac587838a70c818a592e42e44cf766c0e and fe83291c5cb130e6de8dd97d5f93060c5094ad37

api-walker, 2016-09-01:
Thank you very much. Build works now as expected

closed 2016-09-01 by api-walker