armbian/build #453: "Copying files to image [ tmprootfs.raw ]" stalls

WereCatf, 2016-09-01:
Building Xenial 16.04 desktop-image for Orange Pi Plus 2e on a Xenial 16.04 host works otherwise, but something recently broke the last part where it'd actually create the image. Looks like the image-size is calculated wrong, it just stops at 93% and CPU-usage drops to 0% as nothing is actively doing anything.

igorpecovnik, 2016-09-01:
Hmm. It looks fine to me, I have always 99%.

```
[ o.k. ] Creating blank image for rootfs [ 2401 MiB ]
2.34GiB [ 170MiB/s] [=======================================================================================================================>] 100%
[ o.k. ] Creating partitions [ root: ext4 ]
[ o.k. ] Copying files to image [ tmprootfs.raw ]
          1.66G  99%  158.08MB/s    0:00:09 (xfr#93278, to-chk=0/142446)   

sent 1.66G bytes  received 1.95M bytes  158.70M bytes/sec
total size is 1.66G  speedup is 1.00
[ o.k. ] Copying files to /boot partition [ tmprootfs.raw ]
         17.89M  99%  415.30MB/s    0:00:00 (xfr#44, to-chk=0/48) 

sent 17.89M bytes  received 867 bytes  35.79M bytes/sec
total size is 17.89M  speedup is 1.00
[ o.k. ] Free space: [ SD card ]
tmpfs           2.5G  1.8G  683M  73% /root/output/cache/sdcard
/dev/loop0p1    2.3G  1.9G  370M  84% /root/output/cache/mount
[ o.k. ] Writing bootloader [ /dev/loop0 ]
[ o.k. ] Signing and compressing [ Please wait! ]
[ o.k. ] Done building [ /root/output/images/Armbian_5.17_Lime2_Debian_jessie_4.7.2_desktop.7z [491M] ]
[ o.k. ] Runtime [ 7 min ]
```

WereCatf, 2016-09-01:
You built Jessie, I was talking about Xenial. Perhaps that's the issue?

zador-blood-stained, 2016-09-01:
@WereCatf 
Xenial desktop images were running out of space (I don't know why rsync hanged for you instead of throwing errors), so I reverted free space calculation to older value, now it should work fine.

WereCatf, 2016-09-01:
@zador-blood-stained Just got to test it and yeah, that fixed it. Closing issue.

closed 2016-09-01 by WereCatf