armbian/build #456: armbianmonitor exec fail

kometchtech, 2016-09-02:
In the following environment and try to run the armbianmonitor fail.
```
root@dns02:/home/kometch# armbianmonitor -c ./test 
Checking disks is not permitted as root or through sudo. Exiting
```
**env**
```
root@dns02:/home/kometch# hostnamectl 
   Static hostname: dns02
         Icon name: computer
        Machine ID: 25d4b0f89b3e49f0a1df9b406cc1e0f2
           Boot ID: 4ef80cdc21fb420ebb7e4dda8dfaf104
  Operating System: Ubuntu 16.04.1 LTS
            Kernel: Linux 3.10.102-pine64
      Architecture: arm64
```

ThomasKaiser, 2016-09-02:
Well you tried this feature as 'root or through sudo' and the message tries to tell you that this is not good (it really isn't since as root you're enabled to wipe out your whole installation when something goes wrong). 

Please try again as normal user and if that works please close the issue.

kometchtech, 2016-09-02:
@ThomasKaiser 
Thank you.
Worked fine so far executed by normal users.
However, shown below is expands the contents of the script, this is normal behavior?
```
kometch@dns02:~$ armbianmonitor -c ./
BASH=/bin/bash
BASHOPTS=cmdhist:complete_fullquote:extquote:force_fignore:hostcomplete:interactive_comments:progcomp:promptvars:sourcepath
BASH_ALIASES=()
BASH_ARGC=([0]="2")
BASH_ARGV=([0]="./" [1]="-c")
BASH_CMDS=()
BASH_LINENO=([0]="222" [1]="100" [2]="838" [3]="0")
BASH_SOURCE=([0]="/usr/local/bin/armbianmonitor" [1]="/usr/local/bin/armbianmonitor" [2]="/usr/local/bin/armbianmonitor" [3]="/usr/local/bin/armbianmonitor")
BASH_VERSINFO=([0]="4" [1]="3" [2]="46" [3]="1" [4]="release" [5]="aarch64-unknown-linux-gnu")
BASH_VERSION='4.3.46(1)-release'
BOLD=$'\E[1m'
DIRSTACK=()
<snip>
```

ThomasKaiser, 2016-09-02:
Nope, that's not right. Please try to execute `/bin/bash -x armbianmonitor -c ./` and post the results here. And please be aware that at least I might add this only to the TODO list (since this function is for identifying broken SD cards and when I started with that using crap cards I got so many weird symptoms that I only started to make a list what might going wrong here and postponed any fixing for now).

Thanks for reporting this!

kometchtech, 2016-09-02:
Output a with eternity.
```
kometch@dns02:~$ /bin/bash -x armbianmonitor -c ./
+ Main -c ./
+ export PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
+ PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
+ test -t 1
++ tput colors
+ ncolors=8
+ test -n 8
+ test 8 -ge 8
++ tput bold
+ BOLD=''
+ NC='\033[0m'
+ LGREEN='\033[1;32m'
+ LRED='\e[0;91m'
+ '[' 2 -eq 0 ']'
+ ParseOptions -c ./
+ getopts hHbBuUrRmMdDc:C:pP c
+ case ${c} in
+ CheckCard ./
++ id -u
+ '[' 1000 = 0 ']'
+ '[' '!' -d ./ ']'
+ TargetDir=./
+ which f3write
+ which iozone
+ '[' X '!=' X ']'
++ GetDevice ./
+++ findmnt ./
+++ awk '-F ' '/\/dev\// {print $2"\t"$3}'
++ TestPath=
++ [[ -z '' ]]
++ [[ -n . ]]
++ GetDevice .
+++ findmnt .
+++ awk '-F ' '/\/dev\// {print $2"\t"$3}'
++ TestPath=
++ [[ -z '' ]]
++ [[ -n . ]]
++ GetDevice .
+++ findmnt .
+++ awk '-F ' '/\/dev\// {print $2"\t"$3}'
++ TestPath=
++ [[ -z '' ]]
++ [[ -n . ]]
++ GetDevice .
+++ findmnt .
+++ awk '-F ' '/\/dev\// {print $2"\t"$3}'
++ TestPath=
++ [[ -z '' ]]
++ [[ -n . ]]
++ GetDevice .
+++ findmnt .
+++ awk '-F ' '/\/dev\// {print $2"\t"$3}'
++ TestPath=
++ [[ -z '' ]]
++ [[ -n . ]]
++ GetDevice .
+++ findmnt .
+++ awk '-F ' '/\/dev\// {print $2"\t"$3}'
++ TestPath=
++ [[ -z '' ]]
++ [[ -n . ]]
++ GetDevice .
+++ findmnt .
+++ awk '-F ' '/\/dev\// {print $2"\t"$3}'
++ TestPath=
++ [[ -z '' ]]
++ [[ -n . ]]
++ GetDevice .
+++ findmnt .
+++ awk '-F ' '/\/dev\// {print $2"\t"$3}'
++ TestPath=
++ [[ -z '' ]]
++ [[ -n . ]]
++ GetDevice .
+++ findmnt .
+++ awk '-F ' '/\/dev\// {print $2"\t"$3}'
++ TestPath=
++ [[ -z '' ]]
++ [[ -n . ]]
++ GetDevice .
+++ findmnt .
+++ awk '-F ' '/\/dev\// {print $2"\t"$3}'
++ TestPath=
++ [[ -z '' ]]
++ [[ -n . ]]
++ GetDevice .
+++ findmnt .
+++ awk '-F ' '/\/dev\// {print $2"\t"$3}'
++ TestPath=
++ [[ -z '' ]]
++ [[ -n . ]]
++ GetDevice .
+++ findmnt .
+++ awk '-F ' '/\/dev\// {print $2"\t"$3}'
++ TestPath=
++ [[ -z '' ]]
++ [[ -n . ]]
++ GetDevice .
+++ findmnt .
+++ awk '-F ' '/\/dev\// {print $2"\t"$3}'
++ TestPath=
++ [[ -z '' ]]
++ [[ -n . ]]
++ GetDevice .
+++ findmnt .
+++ awk '-F ' '/\/dev\// {print $2"\t"$3}'
++ TestPath=
++ [[ -z '' ]]
++ [[ -n . ]]
++ GetDevice .
+++ findmnt .
+++ awk '-F ' '/\/dev\// {print $2"\t"$3}'
++ TestPath=
++ [[ -z '' ]]
++ [[ -n . ]]
++ GetDevice .
+++ findmnt .
+++ awk '-F ' '/\/dev\// {print $2"\t"$3}'
++ TestPath=
++ [[ -z '' ]]
++ [[ -n . ]]
++ GetDevice .
+++ findmnt .
+++ awk '-F ' '/\/dev\// {print $2"\t"$3}'
++ TestPath=
++ [[ -z '' ]]
++ [[ -n . ]]
++ GetDevice .
+++ findmnt .
+++ awk '-F ' '/\/dev\// {print $2"\t"$3}'
++ TestPath=
++ [[ -z '' ]]
++ [[ -n . ]]
++ GetDevice .
+++ findmnt .
+++ awk '-F ' '/\/dev\// {print $2"\t"$3}'
++ TestPath=
++ [[ -z '' ]]
++ [[ -n . ]]
++ GetDevice .
+++ findmnt .
+++ awk '-F ' '/\/dev\// {print $2"\t"$3}'
++ TestPath=
++ [[ -z '' ]]
++ [[ -n . ]]
++ GetDevice .
+++ findmnt .
+++ awk '-F ' '/\/dev\// {print $2"\t"$3}'
^C
```

ThomasKaiser, 2016-09-03:
Ok, thanks. But if you try a simple `armbianmonitor -c $HOME` or `armbianmonitor -c "$(pwd)"` everything works as expected? Since this is the recommended mode anyway (if people start to provide different paths like eg `/tmp` results will be wrong anyway, so maybe we should change that. But then testing of external disks is not possible any longer).

ThomasKaiser, 2016-09-03:
Sorry, closed the issue by accident :)

kometchtech, 2016-09-03:
@ThomasKaiser 

The brief report.

```
kometch@dns02:/tmp$ armbianmonitor -c "$(pwd)"
Starting to fill /dev/mmcblk0p1 with test patterns, please be patient this might take a very long time
Free space: 958.03 MB
Creating file 1.h2w ... OK!
Free space: 0.00 Byte
Writing speed not available

Now verifying the written data:
                  SECTORS      ok/corrupted/changed/overwritten
Validating file 1.h2w ... 1962040/        0/      0/      0

  Data OK: 958.03 MB (1962040 sectors)
Data LOST: 0.00 Byte (0 sectors)
               Corrupted: 0.00 Byte (0 sectors)
        Slightly changed: 0.00 Byte (0 sectors)
             Overwritten: 0.00 Byte (0 sectors)
Average reading speed: 369.95 MB/s

Starting iozone tests. Be patient, this can take a very long time to complete:
        Iozone: Performance Test of File I/O
                Version $Revision: 3.429 $
                Compiled for 64 bit mode.
                Build: linux 

        Contributors:William Norcott, Don Capps, Isom Crawford, Kirby Collins
                     Al Slater, Scott Rhine, Mike Wisner, Ken Goss
                     Steve Landherr, Brad Smith, Mark Kelly, Dr. Alain CYR,
                     Randy Dunlap, Mark Montague, Dan Million, Gavin Brebner,
                     Jean-Marc Zucconi, Jeff Blomberg, Benny Halevy, Dave Boone,
                     Erik Habbinga, Kris Strecker, Walter Wong, Joshua Root,
                     Fabrice Bacchella, Zhenghua Xue, Qin Li, Darren Sawyer,
                     Vangel Bojaxhi, Ben England, Vikentsi Lapa.

        Run began: Sat Sep  3 22:55:05 2016

        Include fsync in write timing
        O_DIRECT feature enabled
        Auto Mode
        File size set to 102400 kB
        Record Size 4 kB
        Record Size 512 kB
        Record Size 16384 kB
        Command line used: iozone -e -I -a -s 100M -r 4k -r 512k -r 16M -i 0 -i 1 -i 2
        Output is in kBytes/sec
        Time Resolution = 0.000001 seconds.
        Processor cache size set to 1024 kBytes.
        Processor cache line size set to 32 bytes.
        File stride size set to 17 * record size.
                                                              random    random     bkwd    record    stride                                    
              kB  reclen    write  rewrite    read    reread    read     write     read   rewrite      read   fwrite frewrite    fread  freread
          102400       4
Can not open temp file: iozone.tmp
open: Invalid argument

The results from testing /dev/mmcblk0p1 (ext4):
  Data OK: 958.03 MB (1962040 sectors)
Data LOST: 0.00 Byte (0 sectors)
Average reading speed: 369.95 MB/s
                                            random    random
reclen    write  rewrite    read    reread    read     write
     4

Health summary: OK
BASH=/bin/bash
BASHOPTS=cmdhist:complete_fullquote:extquote:force_fignore:hostcomplete:interactive_comments:progcomp:promptvars:sourcepath
BASH_ALIASES=()
BASH_ARGC=([0]="2")
BASH_ARGV=([0]="/tmp" [1]="-c")
BASH_CMDS=()
BASH_LINENO=([0]="222" [1]="100" [2]="838" [3]="0")
BASH_SOURCE=([0]="/usr/local/bin/armbianmonitor" [1]="/usr/local/bin/armbianmonitor" [2]="/usr/local/bin/armbianmonitor" [3]="/usr/local/bin/armbianmonitor")
BASH_VERSINFO=([0]="4" [1]="3" [2]="46" [3]="1" [4]="release" [5]="aarch64-unknown-linux-gnu")
BASH_VERSION='4.3.46(1)-release'
BOLD=$'\E[1m'
DIRSTACK=()
Device=$'/dev/mmcblk0p1\text4'
DeviceName=/dev/mmcblk0p1
EUID=1000
FUNCNAME=([0]="CheckCard" [1]="ParseOptions" [2]="Main" [3]="main")
FileSystem=ext4
GROUPS=()
HOME=/home/kometch
HOSTNAME=dns02
HOSTTYPE=aarch64
IFS=$' \t\n'
LANG=en_US.UTF-8
LANGUAGE=en_US.UTF-8
LC_MESSAGES=en_US.UTF-8
```

kometchtech, 2016-09-03:
When you run this command runs fine.
Could not expand the contents of the shell script.
```
kometch@dns02:/tmp$ armbianmonitor -c $HOME 
Starting to fill /dev/mmcblk0p1 with test patterns, please be patient this might take a very long time
Free space: 12.81 GB
Creating file 1.h2w ... OK!
Creating file 2.h2w ... OK!
Creating file 3.h2w ... OK!
Creating file 4.h2w ... OK!
Creating file 5.h2w ... OK!
Creating file 6.h2w ... OK!
Creating file 7.h2w ... OK!
Creating file 8.h2w ... OK!
Creating file 9.h2w ... OK!
Creating file 10.h2w ... OK!
Creating file 11.h2w ... OK!
Creating file 12.h2w ... OK!
Creating file 13.h2w ... OK!
Free space: 177.32 MB
Average writing speed: 6.74 MB/s

Now verifying the written data:
                  SECTORS      ok/corrupted/changed/overwritten
Validating file 1.h2w ... 2097152/        0/      0/      0
Validating file 2.h2w ... 2097152/        0/      0/      0
Validating file 3.h2w ... 2097152/        0/      0/      0
Validating file 4.h2w ... 2097152/        0/      0/      0
Validating file 5.h2w ... 2097152/        0/      0/      0
Validating file 6.h2w ... 2097152/        0/      0/      0
Validating file 7.h2w ... 2097152/        0/      0/      0
Validating file 8.h2w ... 2097152/        0/      0/      0
Validating file 9.h2w ... 2097152/        0/      0/      0
Validating file 10.h2w ... 2097152/        0/      0/      0
Validating file 11.h2w ... 2097152/        0/      0/      0
Validating file 12.h2w ... 2097152/        0/      0/      0
Validating file 13.h2w ... 1341328/        0/      0/      0

  Data OK: 12.64 GB (26507152 sectors)
Data LOST: 0.00 Byte (0 sectors)
               Corrupted: 0.00 Byte (0 sectors)
        Slightly changed: 0.00 Byte (0 sectors)
             Overwritten: 0.00 Byte (0 sectors)
Average reading speed: 21.08 MB/s

Starting iozone tests. Be patient, this can take a very long time to complete:
        Iozone: Performance Test of File I/O
                Version $Revision: 3.429 $
                Compiled for 64 bit mode.
                Build: linux 

        Contributors:William Norcott, Don Capps, Isom Crawford, Kirby Collins
                     Al Slater, Scott Rhine, Mike Wisner, Ken Goss
                     Steve Landherr, Brad Smith, Mark Kelly, Dr. Alain CYR,
                     Randy Dunlap, Mark Montague, Dan Million, Gavin Brebner,
                     Jean-Marc Zucconi, Jeff Blomberg, Benny Halevy, Dave Boone,
                     Erik Habbinga, Kris Strecker, Walter Wong, Joshua Root,
                     Fabrice Bacchella, Zhenghua Xue, Qin Li, Darren Sawyer,
                     Vangel Bojaxhi, Ben England, Vikentsi Lapa.

        Run began: Sat Sep  3 23:47:41 2016

        Include fsync in write timing
        O_DIRECT feature enabled
        Auto Mode
        File size set to 102400 kB
        Record Size 4 kB
        Record Size 512 kB
        Record Size 16384 kB
        Command line used: iozone -e -I -a -s 100M -r 4k -r 512k -r 16M -i 0 -i 1 -i 2
        Output is in kBytes/sec
        Time Resolution = 0.000001 seconds.
        Processor cache size set to 1024 kBytes.
        Processor cache line size set to 32 bytes.
        File stride size set to 17 * record size.
                                                              random    random     bkwd    record    stride                                    
              kB  reclen    write  rewrite    read    reread    read     write     read   rewrite      read   fwrite frewrite    fread  freread
          102400       4      261      244     6897     6879     5998      243                                                          
          102400     512     6822     6036    22506    22505    22402     1517                                                          
          102400   16384     6642     5913    22546    22540    22534     5958                                                          

iozone test complete.

The results from testing /dev/mmcblk0p1 (ext4):
  Data OK: 12.64 GB (26507152 sectors)
Data LOST: 0.00 Byte (0 sectors)
Average writing speed: 6.74 MB/s
Average reading speed: 21.08 MB/s
                                            random    random
reclen    write  rewrite    read    reread    read     write
     4      261      244     6897     6879     5998      243                                                          
   512     6822     6036    22506    22505    22402     1517                                                          
 16384     6642     5913    22546    22540    22534     5958                                                          

Health summary: OK

Performance summary:
Sequential reading speed: 21.08 MB/s 
 4K random reading speed:  5998 KB/s 
Sequential writing speed:  6.74 MB/s 
 4K random writing speed:   243 KB/s (way too low)

The device you tested seems to perform too slow to be used with Armbian.
This applies especially to desktop images where slow storage is responsible
for sluggish behaviour. If you want to have fun with your device do NOT use
this media to put the OS image or the user homedirs on.

To interpret the results above correctly or search for better storage
alternatives please refer to http://oss.digirati.com.br/f3/ and also
http://www.jeffgeerling.com/blogs/jeff-geerling/raspberry-pi-microsd-card
and http://thewirecutter.com/reviews/best-microsd-card/
```

ThomasKaiser, 2016-09-04:
So, the script works ok. It should be called with absolute paths to directories pointing to storage media. Testing out /tmp (tmpfs, RAM) is useless anyway (reliability testing of what and why?)

> `Free space: 958.03 MB`
> `...`
> `Average reading speed: 369.95 MB/s`

This is no useful mode and it works on real storage media. Closing.

closed 2016-09-04 by ThomasKaiser