armbian/build #457: Armbianmonitor - wrong thermal_zone on Pine64

pfeerick, 2016-09-03:
On the legacy jessie image for the Pine64, running Armbianmonitor -m will not show the correct temperature. A steady 25C is shown, whereas the CPU temperature was around 44C. 

It appears that  lib/scripts/update-motd.d/30-sysinfo  is not picking the correct thermal zone - for some reason the test at ln 127 appears to be 'overwriting' ln 121... as commenting the tests at ln 127 and 133 result in correct temperatures being shown.

ThomasKaiser, 2016-09-07:
Thanks for reporting this. The test in line 127 seems to fail since `$board_temp` is 25000 and not 0 what is tested for:

    if [[ -d "/sys/devices/virtual/thermal/thermal_zone1/" && $board_temp == "0" ]]; then

Unfortunately setting this link from within `30-sysinfo` was a temporary quick hack and should've been avoided just like any quick hack. But since this also affects temperature displayed at login with Pine64 (steady 25°C) the problem has to be resolved anyway. The only quick solution is again an ugly hack:

    # lemaker guitar / roseapple pi
    source /etc/armbian-release
    if [[ -d "/sys/devices/virtual/thermal/thermal_zone1/" && ${LINUXFAMILY} == "s500" ]]; then
    	board_temp=$(awk '{printf("%d",$1/1000)}' </sys/devices/virtual/thermal/thermal_zone1/temp)
    	ln -fs /sys/devices/virtual/thermal/thermal_zone1/temp /etc/armbianmonitor/datasources/soctemp
    fi

Up to @igorpecovnik  to decide.

ThomasKaiser, 2016-09-14:
Just a small reminder @igorpecovnik -- would be great if this could be fixed prior to rolling out new releases. It's still about replacing [these few lines](https://github.com/igorpecovnik/lib/blob/master/scripts/update-motd.d/30-sysinfo#L126-L130)

igorpecovnik, 2016-09-14:
I don't get this. The other thermal zone is read here: https://github.com/igorpecovnik/lib/blob/master/scripts/update-motd.d/30-sysinfo#L121

zador-blood-stained, 2016-09-14:
> I don't get this. The other thermal zone is read here: 

But if both zones exist, second one (`thermal_zone1`) will be preferred unconditionally

igorpecovnik, 2016-09-14:
Aha. Then we set exceptions to a board name? I am running final build later tonight at around 10pm, than publishing in the morning if thing goes well.

ThomasKaiser, 2016-09-14:
> Then we set exceptions to a board name?

I would prefer `${LINUXFAMILY} == "s500"` as above (just in case someone comes around the corner and adds support for [Allo SPARKY](http://www.cnx-software.com/2015/09/14/allo-sparky-board-supports-raspberry-pi-hats-for-multimedia-iot-and-voip-applications/) to Armbian ;)

zador-blood-stained, 2016-09-14:
Ideally in the future we should keep these settings in board configuration files - splitting configuration and logic is a good way to simplify adding new boards and families. For now hardcoded exception for board name or $LINUXFAMILY should work

igorpecovnik, 2016-09-14:
I think this is related solo to Guitar but to be sure I would need to boot exotic s500 boards :)

zador-blood-stained, 2016-09-28:
Should be fixed by commit above

closed 2016-09-28 by zador-blood-stained