armbian/build #459: [OPIPCPLUS] Kernel Panic when setting up `hostapd-rt`

lenormf, 2016-09-05:
Hi,

I've been having issues with version 5.17 of Armbian (jessy, vanilla) provided by armbian.com. I'm trying to setup an access point that will provide connected clients with an IP address fetched from a DHCP server running on a system connected to the OrangePI by ethernet:

[opi **wlan0** (`hostapd`) → **br0** ← **eth0** ] ← [external system **eth0** (`dnsmasq`) ]

When a client connects to `hostapd` through the board's `wlan0` interface, I can see a broadcast hitting the DHCP server (which means `hostapd` has accepted the peer, and the IP request has gone through the bridge properly and reached the external system connected by ethernet), but as soon as the DHCP server replies back (or maybe right before, I'm not sure), the OrangePI crashes with a kernel `BUG`.

I've tried having a look at the file mentioned in the stacktrace (c.f. below), but nothing came out of it (I haven't tried any other image of armbian either).

OrangePI PC Plus, Armbian 5.17 (jessy) vanilla
`hostapd` 2.5 for Realtek rtl871xdrv
Linux 4.6.7-sun8i

**Bridge setup**
```
bridge name	bridge id		STP enabled	interfaces
br0		8000.00e04c6cc661	no		eth0, wlan0
```

**hostapd.conf**
```
ssid=SSID
interface=wlan0
hw_mode=g
channel=5
bridge=br0
logger_syslog=0
logger_syslog_level=0
wmm_enabled=1
ieee80211n=1
noscan=1
driver=rtl871xdrv
```

**output**
```
root@orangepipcplus:~# hostapd -dd /etc/hostapd.conf 
random: Trying to read entropy from /dev/random
Configuration file: /etc/hostapd.conf
drv->ifindex=3
Configure bridge br0 for EAPOL traffic.
BSS count 1, BSSID mask 00:00:00:00:00:00 (0 bits)
Allowed channel: mode=1 chan=1 freq=2412 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=2 freq=2417 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=3 freq=2422 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=4 freq=2427 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=5 freq=2432 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=6 freq=2437 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=7 freq=2442 MHz max_tx_power=[  638.583226] br0: port 2(wlan0) entered blocking state
0 dBm
Allowed channel: mode=1 chan=8 freq=2447 MHz max_tx_power[  638.588689] br0: port 2(wlan0) entered forwarding state
=0 dBm
Allowed channel: mode=1 chan=9 freq=2452 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=10 freq=2457 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=11 freq=2462 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=12 freq=2467 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=13 freq=2472 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=1 freq=2412 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=2 freq=2417 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=3 freq=2422 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=4 freq=2427 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=5 freq=2432 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=6 freq=2437 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=7 freq=2442 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=8 freq=2447 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=9 freq=2452 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=10 freq=2457 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=11 freq=2462 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=36 freq=5180 MHz max_tx_power=0 dBm
[... 181 lines cut ...]
[  681.940527] 7fc0: c010811c c0108120 60010013 ffffffff
[  681.945614] [<c010bdd4>] (__irq_svc) from [<c0108120>] (arch_cpu_idle+0x38/0x3c)
[  681.953062] [<c0108120>] (arch_cpu_idle) from [<c015cc58>] (cpu_startup_entry+0x1b8/0x214)
[  681.961375] [<c015cc58>] (cpu_startup_entry) from [<4010154c>] (0x4010154c)
[  681.968362] CPU2: stopping
[  681.971107] CPU: 2 PID: 0 Comm: swapper/2 Tainted: G      D W       4.6.7-sun8i #2
[  681.978715] Hardware name: Allwinner sun8i Family
[  681.983460] [<c010e9c8>] (unwind_backtrace) from [<c010b308>] (show_stack+0x10/0x14)
[  681.991253] [<c010b308>] (show_stack) from [<c052fcc4>] (dump_stack+0x84/0x98)
[  681.998526] [<c052fcc4>] (dump_stack) from [<c010d850>] (handle_IPI+0x168/0x188)
[  682.005971] [<c010d850>] (handle_IPI) from [<c01014bc>] (gic_handle_irq+0x88/0x8c)
[  682.013590] [<c01014bc>] (gic_handle_irq) from [<c010bdd4>] (__irq_svc+0x54/0x70)
[  682.021111] Exception stack(0xef0a1f88 to 0xef0a1fd0)
[  682.026194] 1f80:                   00000001 00000000 00000000 c0119620 ef0a0000 c0d0249c
[  682.034419] 1fa0: 00000000 00000000 c0c5c578 c0d02500 c0d49f1b c0d02508 00000000 ef0a1fd8
[  682.042638] 1fc0: c010811c c0108120 60010013 ffffffff
[  682.047726] [<c010bdd4>] (__irq_svc) from [<c0108120>] (arch_cpu_idle+0x38/0x3c)
[  682.055174] [<c0108120>] (arch_cpu_idle) from [<c015cc58>] (cpu_startup_entry+0x1b8/0x214)
[  682.063487] [<c015cc58>] (cpu_startup_entry) from [<4010154c>] (0x4010154c)
[  682.070478] Rebooting in 10 seconds..
```

Any idea what's up ?

Thanks.

igorpecovnik, 2016-09-05:
You are running an image with first **beta / preview** version of kernel, which is already EOL ... I hope you acknowledged that? It's good to know this, but I would try to build more recent kernel and try the same thing first.

lenormf, 2016-09-05:
Since the issue happens on the images you provide, I thought you would want to know so that you could fix it in your releases.

golfromeo-fr, 2016-09-05:
Since the trace shows a kernel bug, it would be more interesting to report to kernel.org, so Igor added "not-our-bug" 

But as Igor says, you may use the Armbian tool to build an up-to-date kernel image (in case this bug has been fixed by kernel.org) or wait for an "official" Armbian build with a stable linux kernel.

igorpecovnik, 2016-09-05:
I could be wrong and the bug is ours since we do certain adaptations to the kernel. We took a kernel from kernel.org and add certain things to the kernel, that those H3 boards works. We add wireless driver and of course the problem could be there ... so I am deciding based on a hunch, experience and small real data. I will leave issue open to see if we receive more input on topic.

lenormf, 2016-09-05:
@igorpecovnik what else do you need? I found [this issue](https://bugzilla.kernel.org/show_bug.cgi?id=98521) on the kernel bugzilla that really sounds similar to mine, except the stacktrace is quite different from mine.

I've tried the `4.7.2` and `4.6.7` kernels, the vanilla and legacy versions, and the bug still occurs. Note that I am unable to build an image myself because of the docker issues (c.f. #385), but I'm still trying to use another distro on my OPI PC+ to try to reproduce the issue (I would know whether this is a genuine Linux issue, or something we can fix in Armbian).

golfromeo-fr, 2016-09-05:
yes, right, if we do changes in modules 8189fs sun8i_ths sunxi_cir cpufreq_dt thermal_sys 

yes possible. duckduckgo search points to Armbian for 8189fs :)

is a test on OPI2e+ could be of use? if yes, I can build an image and test on the board I have

lenormf, 2016-09-05:
@igorpecovnik without talking about trying to fix the issue off the bat, it'd be good if someone could confirm that the issue comes from armbian (or prove otherwise).

igorpecovnik, 2016-09-05:
Again. This image is a preview and presumably full of bugs. Yes, it's a good thing that bugs are recorded. That's why they were made in the first place. Issues within are low priority since in most cases we can't be sure whether it's our or general bug or we are facing bug that's already been fixed upstream.

I hope this is acceptable and understandable. 

Nobody prohibits anyone to find and fix problems - Linux is a community project.

lenormf, 2016-09-06:
It looks like the packets going through the bridge have their checksum offset beyond the size of the current frame, which triggers the kernel `BUG`. I'm leaning towards a network driver bug, particularly considering the issues everybody has with realtek chips, but since I don't really need to get bridging to work, I'm going to assume this is not an issue related to kernel changes made on armbian's side.

Closing.

closed 2016-09-06 by lenormf

frankipl, 2018-06-18:
Hi
I had the same issue with Armbian 5.38 on Orange PI+, meaning: kernel panic after a few seconds of hostapd working. I have tried hostapd and hostapd-realtek, compiled different versions of hostapd. 
Finally I have compiled hostapd 2.0 with nl driver and apt-get remove --purge network-manager dnsmasq* (I'm using isc-dhcp)
If I'm not logged into system (via ssh) hostapd is running fine (a few hours now), when logged in, kernel still can panic after a minute.
Very strange!?

Best Regards,
Marek