armbian/build #460: consequtive upgrades to linux-jessie-root-cubietruck break due to mv /boot/bin to /boot/bin.old

mx-pycoder, 2016-09-05:
Somewhere in the upgrade process, the old /boot/bin directory is renamed (moved) to /boot/bin.old. Upon the next upgrade, the current /boot/bin directory is moved to /boot/bin.old/bin. The third time around, the upgrade process fails due to /boot/bin.old/bin not being empty. See http://forum.armbian.com/index.php/topic/690-upgrade-error-with-armbian-500-on-cubieboard2-running-from-nand/ thread

mx-pycoder, 2016-09-05:
Please keep all discussion for this issue on the forum topic available below:

[http://forum.armbian.com/index.php/topic/1963-460-consequtive-upgrades-to-linux-jessie-root-cubietruck-break-due-to-mv-bootbin-to-bootbinold/](http://forum.armbian.com/index.php/topic/1963-460-consequtive-upgrades-to-linux-jessie-root-cubietruck-break-due-to-mv-bootbin-to-bootbinold/)

zador-blood-stained, 2016-09-05:
For the record, you don't have to create forum topics for "normal" issues, this procedure is for assigned tasks only.

Commit 8ddc691 should fix this issue, needs testing.

mx-pycoder, 2016-09-05:
Thanks, sorry for not properly following procedures. I was trying to figure out how to file my first issue to the armbian project.

zador-blood-stained, 2016-09-05:
For issues with build script or build configuration GitHub issues are a good choice

mx-pycoder, 2016-09-05:
Okay, good to know!

zador-blood-stained, 2016-09-16:
Feel free to comment or reopen if latest release (5.20) didn't fix the issue.

closed 2016-09-16 by zador-blood-stained