armbian/build #461: Native support to at least one of these WiFi adapters

Trapizomba, 2016-09-05:
Can't understand why armbian don't have native support to at least one of these:

- Atheros AR9271
- Ralink RT3070
- Ralink RT3572

They are the most used USB WiFi adapters out there...

igorpecovnik, 2016-09-06:
http://forum.armbian.com/index.php/topic/1325-claim-a-task-set-due-date-and-start-doing-it/

zador-blood-stained, 2016-09-06:
> Can't understand why armbian don't have native support to at least one of these:
> Atheros AR9271
> Ralink RT3070
> Ralink RT3572

Because Armbian is non-commercial open-source project. We don't have resources and intentions to buy, test and prepare support for every third party hardware that is available/popular in every region of the world.

Especially when you can compile any driver (that is compatible with provided kernel version) on the board itself, using simple (or even generic) instructions.

ThomasKaiser, 2016-09-07:
Just for the record (and google users stumbling accross this): we added the firmware package for AR9271 back in March and both AR9271 and RT5370 are confirmed to work. So instead of opening an issue just to complain about missing _native support_ it might already help to know what you are talking about calling support _native_?

Drivers compiled, included and tested on all the +40 boards with both kernel versions and all 4 distros so that any user just has to insert the dongle and everything works out of the box?

itdaniher, 2016-11-01:
IIRC I had to drop a blob from built from https://github.com/qca/open-ath9k-htc-firmware for my AR9271. No big deal.

zador-blood-stained, 2016-11-24:
Merging this issue into #558

closed 2016-11-24 by zador-blood-stained