armbian/build #462: Cannot build for Orange Pi : Problem with U-Boot compilation

JulesThuillier, 2016-09-07:
Hello,

I have a problem when building for my orange pi pc + (though it used to work a few weeks ago). Here is the error trace during compilation. It happens even when I just try to build the kernel.

`DTC     arch/arm/dts/sun8i-h3-orangepi-2.dtb   In file included from arch/arm/dts/sun8i-h3-orangepi-2.dts:44:0:`
`                                      arch/arm/dts/sun8i-h3. `
`dts/Makefile:36: recipe for target 'arch-dtbs' failed `
`Makefile:816: recipe for target 'dts/dt.dtb' failed   `
`]: *** [arch-dtbs] Error 2`
`             compilation make: *** [dts/dt.dtb] Error 2`
`                                     make[2]: *** [arch/arm/dts/sun8i-h3-orangepi-2.dtb] Error 1`
`[ error ] ERROR in function compile_uboot [ common.sh:96 ]`
`[ error ] U-boot file not found [ u-boot-sunxi-with-spl.bin ]`
`[ o.k. ] Process terminated `

I think this problem has something to do with recent changes with U-Boot config.
BTW: I have been able to build for other boards.

zador-blood-stained, 2016-09-08:
Please provide full console output log and build logs. Both Orange Pi PC+ default and dev targets work for me.

JulesThuillier, 2016-09-08:
Here you go !
[logs.zip](https://github.com/igorpecovnik/lib/files/461218/logs.zip)

zador-blood-stained, 2016-09-08:
Looks like `add-emmc-orangepi.patch` causes issues for you for some reason. You should be able to build u-boot by creating empty file `userpatches/u-boot/u-boot-dev/add-emmc-orangepi.patch` as temporary workaround, i.e.
```
sudo touch /home/vrtracker/Documents/userpatches/u-boot/u-boot-dev/add-emmc-orangepi.patch
```
in your case

JulesThuillier, 2016-09-08:
I just tried that, it did what you expected (skip patch)

`add-emmc-orangepi.patch [ skipped ]`

But I ended up with the same error :/ I'd like to help you but I don't where to look

zador-blood-stained, 2016-09-08:
First, please try removing `sources/u-boot` directory to reset u-boot sources. If this doesn't help, try building with `PROGRESS_DISPLAY=plain` added to compile.sh and provide log from console to get full output with error messages.

JulesThuillier, 2016-09-08:
The source was fresh from yesterday, still it gave another error : 

```
[ o.k. ] Building deb [ linux-u-boot-dev-orangepipcplus_5.17_armhf.deb ]
dpkg-deb: error: control directory has bad permissions 770 (must be >=0755 and <=0775)
wc: /home/vrtracker/Documents/output/debs/linux-u-boot-dev-orangepipcplus_5.17_armhf.deb: No such file or directory
rm: cannot remove '/home/vrtracker/Documents/output/debs/linux-u-boot-dev-orangepipcplus_5.17_armhf.deb': No such file or directory
[ error ] ERROR in function compile_uboot [ common.sh:109 ]
[ error ] Building u-boot failed, check configuration 
[ o.k. ] Process terminated 
```

After a chmod, that went away to FINALLY WORK !
I compiled U-Boot properly and now it's compiling the kernel.

There must have been a fix in U-Boot source yesterday or this morning because I wiped cloned the git again yesterday morning.

Well thanks again for the assistance, and amazing job on Armbian !

closed 2016-09-08 by zador-blood-stained