armbian/build #463: u-boot build fails for lime2-emmc:  u-boot-sunxi-with-spl.bin file not found

lauhub, 2016-09-07:
When trying to compile with the current version of build tool for lime2-emmc, build process stops with this error:
```
  │   DTC     arch/arm/dts/sun7i-a20-yones-toptech-bd1078.dtb                                                                                                                                            │  
  └─────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────Device Tree Source is not correctly specified.                                                                                                                                                                    
              Please define 'CONFIG_DEFAULT_DEVICE_TREE'
                             or build with 'DEVICE_TREE=<device_tree>' argument

                                                      make[1]: *** [arch/arm/dts/sun7i-a20-olinuxino-lime2-emmc.dtb] Error 1
                                                                                make: *** [dts/dt.dtb] Error 2
[ error ] ERROR in function compile_uboot [ common.sh:96 ]
[ error ] U-boot file not found [ u-boot-sunxi-with-spl.bin ]
[ o.k. ] Process terminated 
```
while selecting lime2 (no emmc) works well.

This is for u-boot v2016.07 and linux-vanilla.

See output.log file and a copy of the console during build.

[output.log](https://github.com/igorpecovnik/lib/files/460486/output.log.lime2-emmc.txt)

[console-output.txt.zip](https://github.com/igorpecovnik/lib/files/460489/console-output.txt.zip)

JulesThuillier, 2016-09-08:
Exactly same issue as I posted yesterday on different board ;)
[https://github.com/igorpecovnik/lib/issues/462](url)

zador-blood-stained, 2016-09-08:
Lime2-eMMC target is experimental and it's possible that it will be removed in the future. Please use Lime2 target instead for now.

lauhub, 2016-09-08:
@JulesThuillier : not exactly the same, the errors prior to the last one are different. You have 

`dts/Makefile:36: recipe for target 'arch-dtbs' failed` 

 `Makefile:816: recipe for target 'dts/dt.dtb' failed`

I don't have these errors so I believe cause is different with same result

@zador-blood-stained :
Lime2 target does not work well for now for me either : system does not boot completely and I fallback to previous image I had (kernel 4.6.4). I don't have much time to test further for now.

zador-blood-stained, 2016-09-08:
@lauhub 
Please clean rootfs cache (depete files in `output/cache/rootfs`) and try again, there was an issue with wrong packages lists that should be fixed now. If not, please provide boot logs.

lauhub, 2016-09-08:
@zador-blood-stained 
The system (4.7.3) boots with SD card but I have no more board available to test on eMMC for now. As soon as I have one, I will test

FYI the issue appeared once after I burnt the image to the eMMC and made the [modification given here](http://forum.armbian.com/index.php/topic/853-armbian-customization/page-4#entry8032). Booting from eMMC then hanged after a while.

neomanic, 2016-10-11:
As per Armbian 5.21, building with `./compile.sh BOARD=lime2-emmc BRANCH=next` (so kernel 4.8.1) works just fine. Suggest closing this issue.

igorpecovnik, 2016-10-11:
OK

closed 2016-10-11 by igorpecovnik