armbian/build #464: [feature request] implement CCACHE_DIR to put ccache directory along compile.sh

golfromeo-fr, 2016-09-09:
There is an way to avoid ccache write files as root : setting CCACHE_DIR in the same folder as compile.sh

https://ccache.samba.org/manual.html

So we can avoid to have "root" owned files in $HOME/.ccache

zador-blood-stained, 2016-09-09:
Yes, but there are different ways to approach this problem:

- Set umask in config and setgit bit on ccache directory (see section "SHARING A CACHE") manually if you need this

- Export any environment variables that you need (like `CCACHE_DIR` or `CCACHE_CONFIGPATH` from userpatches/lib.config

- Add extra setting to Armbian build script (Which one? With what default value?) to handle this problem

golfromeo-fr, 2016-09-09:
@zador-blood-stained
Mikhail, I do agree with you

option 1 is messy (mixing local user + root owned files in $HOME/.ccache is dirty)

option 2 alone is possible without changing Armbian but the issue is not fixed for other users.

option 2+3 is better, Armbian tool manage its own ccache folder (and deleting this folder is ok)

=> Armbian tool create a local "ccache" folder along the folders (lib, output, toolchains sources  userpatches) where the working copy of compile.sh lies => CCACHE_DIR=$SRC/ccache 

Additionally, the user could setup a RAMDISK of size [CCACHE_MAXSIZE=1GB] probably with CCACHE_COMPRESS=true (and symlinks $SRC/ccache to the RAMDISK)
(never tested but it could be somewhat faster for mass generation of images -- if the images are compiled/generated by SoC familly/type -- A20, H3, Exynos, etc. to take advantage of ccache)
 
(Is it something we could agree on? So I may propose a patch)

zador-blood-stained, 2016-09-09:
> option 1 is messy (mixing local user + root owned files in $HOME/.ccache is dirty)

Dirty, but it works and I don't see any real security considerations for this

> option 2 alone is possible without changing Armbian but the issue is not fixed for other users.

There were no user reports that compilation broke for them since ccache was added by default, so I think most users use this build script in virtualized environment where this is not an issue at all.

> => Armbian tool create a local "ccache" folder along the folders (lib, output, toolchains, sources, userpatches) where the working copy of compile.sh lies => CCACHE_DIR=$SRC/ccache

It has some disadvantages:

- Separate directory means I have to set extra environment variable each time I want to see cache statistics or compile kernel/u-boot manually (and get any cache hits).

- Separate config file means you have to set your options (file limit, size limit, compression, temporary directory) in both config files.

> Additionally, the user could setup a RAMDISK of size [CCACHE_MAXSIZE=1GB] probably with CCACHE_COMPRESS=true (and symlinks $SRC/ccache to the RAMDISK)

"Warm" cache will stay in RAM in any case if you have enough of it, even "cold" cache is fast on SSD, and getting cache from previous runs is more important since you don't build the same kernel more than once with BUILD_ALL option.

zador-blood-stained, 2016-09-09:
> Is it something we could agree on?

My proposal:

- Build script option `PRIVATE_CCACHE`

- Disabled by default

- Description: Use separate ccache directory located in build tree (`$DEST/ccache`). Can be used to avoid problems with ownershipt on `$HOME/.ccache`. Set to "yes" if you are using build script as non-root user with "sudo"

golfromeo-fr, 2016-09-11:
Done as proposed. 
Left the option to create a symlink to the local ccache
Left some commented lines in ccache.conf (so the user can tweak ccache)

zador-blood-stained, 2016-09-25:
Closed with 870a7ac
Let's leave all cleanup and configuration work for end user to keep the code simple

closed 2016-09-25 by zador-blood-stained