armbian/build #474: Kernel compiling dev version of orangepi one shows warning

globalkrishna, 2016-09-22:
[ warn ] ... scripts-dtc-Update-to-version-with-overlays.patch [ failed ]

martinayotte, 2016-09-23:
Someone already reported in the forum itself.
The failure is fortunately only about the version tags found version_gen.h, which didn't prevent the build to be successful.
I will provide soon a fix for this tag change required with recent switch from 4.6.x and 4.7.x.

globalkrishna, 2016-09-23:
Even though it compiled, it did not boot after burning the SD Card.

martinayotte, 2016-09-23:
The fix for the above patch rejection is now committed :
https://github.com/igorpecovnik/lib/commit/0d2fa654cfb72d46f45842911228369432e58658
For the fact that your image didn't boot, we will have to dig more, probably you will need to provide Serial Debug output log while opening another issue, since it could probably be elsewhere, such DRAM clock switching in U-Boot.

closed 2016-09-23 by martinayotte