armbian/build #485: nand-sata-install set incorrectly u-boot path

martinayotte, 2016-10-03:
More than a week ago, I've posted this thread : 
http://forum.armbian.com/index.php/topic/2044-some-discovery-while-trying-520-builds/#entry16214
I've provided a potential fix, which takes cares of handling branch name to be used in the path where the u-boot is located.
But it needs to be tested also in Legacy, and it seems that no one seen that post ...

`
diff --git a/scripts/nand-sata-install/usr/lib/nand-sata-install/nand-sata-install.sh b/scripts/nand-sata-install/usr/lib/nand-sata-install/nand-sata-install.sh
index bab4960..f49e1e9 100644
--- a/scripts/nand-sata-install/usr/lib/nand-sata-install/nand-sata-install.sh
+++ b/scripts/nand-sata-install/usr/lib/nand-sata-install/nand-sata-install.sh
@@ -124,9 +124,10 @@ EOF
                
                # determine u-boot and write it
                name_of_ubootpackage=$(aptitude versions '~i linux-u-boot*'| head -1 | awk '{print $2}' | sed 's/linux-u-boot-//g' | cut -f1 -d"-")
+               name_of_ubootbranch=$(aptitude versions '~i linux-u-boot*'| head -1 | awk '{print $2}' | sed 's/linux-u-boot-.*-//g' | cut -f1 -d":")
                version_of_ubootpkg=$(aptitude versions '~i linux-u-boot*'| tail -1 |  awk '{print $2}')
                architecture=$(dpkg --print-architecture)
-               uboot="/usr/lib/linux-u-boot-"$name_of_ubootpackage"_"$version_of_ubootpkg"_"$architecture""/u-boot-sunxi-with-spl.bin
+               uboot="/usr/lib/linux-u-boot-"$name_of_ubootbranch"-"$name_of_ubootpackage"_"$version_of_ubootpkg"_"$architecture""/u-boot-sunxi-with-spl.bin
                dd if=$uboot of=$emmccheck bs=1024 seek=8  >/dev/null 2>&1 || (echo "Error"; exit 0)
                
        elif [[ -f /boot/boot.cmd ]]; then
`

zador-blood-stained, 2016-10-03:
I have a better idea for this (more universal), I'll try to implement it by the end of this week.

zador-blood-stained, 2016-10-06:
This should work: d42c867
Needs testing

igorpecovnik, 2016-10-06:
Installation with our latest dev branch (4.7.6-sun8i) works, after fix.

martinayotte, 2016-10-10:
Yes ! I've tested this latest commit, it is working properly now.

martinayotte, 2016-10-12:
Sadly, I discovered today while doing new build for my OrangePiPlus2E that "boot.cmd" has not been parsed and upgraded to have the following value for rootdev (it was still at /dev/mmcblk0) : 
```
setenv rootdev "/dev/mmcblk2p1"
```

zador-blood-stained, 2016-10-15:
Boot script updating is not related to the original issue, so closing this while leaving #497 open as a reminder.

closed 2016-10-15 by zador-blood-stained