armbian/build #487: RfC: Documentation improvements

ThomasKaiser, 2016-10-04:
I would propose a separate documentation hierarchy `lib.docs/docs/board_details` where we can add board specific information and software tweaks. As an example: For Olimex A10-Lime I would want to add 

- link to linux-sunxi wiki page
- information about consumption and some performance numbers
- info to disable display to improve memory throughput with legacy kernel: `echo 1 > /sys/devices/platform/disp/graphics/fb0/blank`
- links to interesting forum threads covering specific aspects (USB OTG, DRAM reliability)
- hardware specifics like both USB ports and SATA power using step-up converters and being able to run on battery
- tricks like the following to switch on/off SATA disk and USB ports:

    sunxi-pio -m PC03'<default><default<default><0>'
    sunxi-pio -m PH03'<default><default<default><0>'
    sunxi-pio -m PH06'<default><default<default><0>'

IMO from the download pages (in this case http://www.armbian.com/olimex-lime-a10/) the 'Hardware details' link should point to this specific device page (which then links to linux-sunxi wiki for further details and focuses on Armbian specific tweaks).

For the moment all I would need is conformation about the path. We can later decide how to integrate these new device pages below docs.armbian.com (or do we have to make a decision already)?

igorpecovnik, 2016-10-05:
https://github.com/igorpecovnik/lib.docs/tree/master/docs/boards

If it's added here, it will be displayed at download by default. So you want that we have separate file for this?

That right box on the download panel is currently generated this way - I can put it to GitHub that changes can be applied from others if this is desired:
```
case "beelinkx2":
        $HARDWARE="https://linux-sunxi.org/Beelink_X2";
        $CHIP="http://docs.armbian.com/Hardware_Allwinner-H3/";
        $FORUMS="http://forum.armbian.com/index.php/forum/13-allwinner-h3";
        $MAINLINE="hide";
        break;
```

ThomasKaiser, 2016-10-05:
> So you want that we have separate file for this and for docs only?

Exactly, I don't want that the download pages are turned into walls of text again (where essential information is missing, please be aware that a fraction of users is not able or willing to read anything that exceeds the length of a tweet)

That's why I propose a separate hierarchy where board specific information can be collected and if it's ready it would be great to adjust `$HARDWARE` link you shoved above. So having these links editable on Github would be great too.

ThomasKaiser, 2016-10-06:
[x] done: https://github.com/igorpecovnik/lib.docs/commit/b337bb4e6d26bafccc1827f2fb6359703d45f0af

closed 2016-10-06 by ThomasKaiser