armbian/build #491: Build failure: Odroid C1 | Jessie | Desktop

Fourdee, 2016-10-09:
Commands run from a fresh git pull:
```
git clone https://github.com/fourdee/lib --depth 1
cp lib/compile.sh .

sed -i 's/en_US/en_GB/' compile.sh
sed -i '/BOOTSIZE=/c\BOOTSIZE=64' lib/configuration.sh
sed -i '/ROOTPWD=/c\ROOTPWD="dietpi"' lib/configuration.sh

./compile.sh
```

```
[ o.k. ] Ending debootstrap process and preparing cache [ jessie ]
jessie-ng-armhf.230...c48.tgz:  251MB [ 144MB/s] [========================] 106%
[ o.k. ] Applying distribution specific tweaks for [ jessie ]
sed: can't read /root/output/cache/sdcard/etc/ssh/sshd_config: No such file or d                                                                 irectory
[ o.k. ] Applying common tweaks
sed: can't read /root/output/cache/sdcard/etc/default/console-setup: No such fil                                                                 e or directory
[ o.k. ] Installing kernel [ linux-image-odroidc1 ]
[ o.k. ] Installing u-boot [ linux-u-boot-odroidc1 ]
[ o.k. ] Installing headers [ linux-headers-odroidc1 ]
[ o.k. ] Installing DTB [ linux-dtb-odroidc1 ]
[ o.k. ] Installing board support package [ odroidc1 ]
[ o.k. ] Installing extra applications and drivers
[ o.k. ] Merging and packaging linux firmware [ @host ]
[ o.k. ] Installing linux firmware [ 5.21 ]
[ o.k. ] Installing additional driver [ MT7601U ]
[ o.k. ] Checking git sources [ mt7601 old ]
[ .... ] Creating local copy
[ .... ] Fetching updates
remote: Counting objects: 264, done.
remote: Compressing objects: 100% (207/207), done.
remote: Total 264 (delta 54), reused 255 (delta 53), pack-reused 0
Receiving objects: 100% (264/264), 1.29 MiB | 674.00 KiB/s, done.
Resolving deltas: 100% (54/54), done.
From https://github.com/igorpecovnik/mt7601
 * branch            old        -> FETCH_HEAD
 * [new branch]      old        -> origin/old
[ .... ] Checking out
cp: missing destination file operand after ‘/root/output/cache/sdcard/lib/module                                                                 s/3.10.103-odroidc1/kernel/net/wireless/’
Try 'cp --help' for more information.
[ o.k. ] Installing additional driver [ RT8192 ]
[ o.k. ] Checking git sources [ rt8192cu master ]
[ .... ] Creating local copy
[ .... ] Fetching updates
remote: Counting objects: 201, done.
remote: Compressing objects: 100% (177/177), done.
remote: Total 201 (delta 43), reused 71 (delta 21), pack-reused 0
Receiving objects: 100% (201/201), 1.17 MiB | 555.00 KiB/s, done.
Resolving deltas: 100% (43/43), done.
From https://github.com/dz0ny/rt8192cu
 * branch            master     -> FETCH_HEAD
 * [new branch]      master     -> origin/master
[ .... ] Checking out
cp: cannot create regular file ‘/root/output/cache/sdcard/lib/modules/3.10.103-o                                                                 droidc1/kernel/net/wireless/’: Not a directory
[ o.k. ] Building deb [ armbian-tools ]
[ o.k. ] ... downloading sources [ temper ]
[ o.k. ] ... downloading sources [ BT utils ]
[ o.k. ] ... compiling [ temper ]
[ error ] ERROR in function compiling [ tools.sh:70 ]
[ error ] Error building [ temper ]
[ o.k. ] Process terminated
rm: cannot remove ‘/root/output/cache/sdcard/tmp/debs’: Device or resource busy
[ error ] ERROR in function unmount_on_exit [ debootstrap-ng.sh:576 ]
[ error ] debootstrap-ng was interrupted
[ o.k. ] Process terminated
```

igorpecovnik, 2016-10-09:
I saw, working on it.

Fourdee, 2016-10-09:
Re-test results (if useful):
```
cat: /root/sources/u-boot-odroidc1/odroidc-v2011.03/Makefile: No such file or directory
cat: /root/sources/u-boot-odroidc1/odroidc-v2011.03/Makefile: No such file or directory
cat: /root/sources/u-boot-odroidc1/odroidc-v2011.03/Makefile: No such file or directory
cat: /root/sources/u-boot-odroidc1/odroidc-v2011.03/Makefile: No such file or directory
[ o.k. ] Compiling uboot
[ o.k. ] Compiler version [ arm-linux-gnueabihf-gcc 4.8 ]
make: *** No rule to make target `odroidc_config'. Stop.
make: ***

[ error ] ERROR in function compile_uboot [ common.sh:102 ]
[ error ] U-boot file not found [ bl1.bin.hardkernel ]
[ o.k. ] Process terminated
```

igorpecovnik, 2016-10-09:
That could be a GitHub broken download, network issue.

Fourdee, 2016-10-09:
@igorpecovnik 

4th test, same results as: https://github.com/igorpecovnik/lib/issues/491#issue-181886812

I am experiencing some issues with github connections timing out in general. Might be caused solely by that.
Not sure if its host/server cause. I'll reboot my network and run a retest tomorrow.

Fourdee, 2016-10-10:
Retest completed, same issue as https://github.com/igorpecovnik/lib/issues/491#issue-181886812

zador-blood-stained, 2016-10-10:
> Retest completed, same issue as #491 (comment)

Did you clear rootfs cache? Can you provide full build process output after that (after kernel compilation)?

igorpecovnik, 2016-10-10:
... or sources? I just built and boot C1 so system is working but I guess we need to prevent this happen.

Fourdee, 2016-10-10:
@igorpecovnik @zador-blood-stained 
Looks like an issue with package deps.

Output with dialog disabled:
```
[ o.k. ] Installing packages for [ Armbian ]
Reading package lists...
Building dependency tree...
Reading state information...
less is already the newest version.
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 mpv : Depends: libavcodec-ffmpeg56 but it is not going to be installed
       Depends: libavdevice-ffmpeg56 but it is not going to be installed
       Depends: libavfilter-ffmpeg5 but it is not going to be installed
       Depends: libavformat-ffmpeg56 but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
[ error ] ERROR in function create_rootfs_cache [ debootstrap-ng.sh:243 ]
[ error ] Installation of Armbian packages failed
[ o.k. ] Process terminated
[ error ] ERROR in function unmount_on_exit [ debootstrap-ng.sh:579 ]
[ error ] debootstrap-ng was interrupted
[ o.k. ] Process terminated
```

Local install of ```mpv``` package is fine:
```
root@vm-crosscomp:~/armbian# apt-get install mpv
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following extra packages will be installed:
  fontconfig-config fonts-dejavu-core libaacs0 libass4 libasyncns0
  libavcodec54 libavfilter3 libavformat54 libavresample1 libavutil52
  libbluray1 libbs2b0 libcdio-cdda1 libcdio-paranoia1 libcdio13 libdrm-intel1
  libdrm-nouveau2 libdrm-radeon1 libdvdnav4 libdvdread4 libegl1-mesa
  libegl1-mesa-drivers libenca0 libflac8 libfontconfig1 libgbm1
  libgl1-mesa-dri libgl1-mesa-glx libglapi-mesa libgsm1 libguess1
  libjack-jackd2-0 libjpeg-turbo8 libjpeg8 liblcms2-2 liblircclient0
  libllvm3.4 liblua5.2-0 libmowgli2 libmp3lame0 libmpg123-0 libogg0
  libopenjpeg2 libopenvg1-mesa libopus0 liborc-0.4-0 libpciaccess0
  libpostproc52 libpulse0 libquvi-scripts libquvi7 libsamplerate0
  libschroedinger-1.0-0 libsdl2-2.0-0 libsndfile1 libspeex1 libswscale2
  libtheora0 libtxc-dxtn-s2tc0 libva-glx1 libva-x11-1 libva1 libvdpau1
  libvorbis0a libvorbisenc2 libvpx1 libwayland-client0 libwayland-cursor0
  libwayland-egl1-mesa libwayland-server0 libx11-xcb1 libx264-142
  libxcb-dri2-0 libxcb-dri3-0 libxcb-glx0 libxcb-present0 libxcb-sync1
  libxcb-xfixes0 libxcursor1 libxdamage1 libxfixes3 libxi6 libxinerama1
  libxkbcommon0 libxrandr2 libxrender1 libxshmfence1 libxss1 libxv1
  libxvidcore4 libxxf86vm1 x11-common
Suggested packages:
  frei0r-plugins libbluray-bdj libdvdcss2 debhelper fakeroot libglide3 jackd2
  liblcms2-utils lirc opus-tools pulseaudio speex nvidia-vdpau-driver
  vdpau-driver
The following NEW packages will be installed
  fontconfig-config fonts-dejavu-core libaacs0 libass4 libasyncns0
  libavcodec54 libavfilter3 libavformat54 libavresample1 libavutil52
  libbluray1 libbs2b0 libcdio-cdda1 libcdio-paranoia1 libcdio13 libdrm-intel1
  libdrm-nouveau2 libdrm-radeon1 libdvdnav4 libdvdread4 libegl1-mesa
  libegl1-mesa-drivers libenca0 libflac8 libfontconfig1 libgbm1
  libgl1-mesa-dri libgl1-mesa-glx libglapi-mesa libgsm1 libguess1
  libjack-jackd2-0 libjpeg-turbo8 libjpeg8 liblcms2-2 liblircclient0
  libllvm3.4 liblua5.2-0 libmowgli2 libmp3lame0 libmpg123-0 libogg0
  libopenjpeg2 libopenvg1-mesa libopus0 liborc-0.4-0 libpciaccess0
  libpostproc52 libpulse0 libquvi-scripts libquvi7 libsamplerate0
[... 881 lines cut ...]
Processing triggers for libc-bin (2.19-18+deb8u6) ...
[ o.k. ] Installing packages for [ Armbian ]
Reading package lists...
Building dependency tree...
Reading state information...
less is already the newest version.
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 mpv : Depends: libavcodec-ffmpeg56 but it is not going to be installed
       Depends: libavdevice-ffmpeg56 but it is not going to be installed
       Depends: libavfilter-ffmpeg5 but it is not going to be installed
       Depends: libavformat-ffmpeg56 but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
[ error ] ERROR in function create_rootfs_cache [ debootstrap-ng.sh:243 ]
[ error ] Installation of Armbian packages failed
[ o.k. ] Process terminated
[ error ] ERROR in function unmount_on_exit [ debootstrap-ng.sh:579 ]
[ error ] debootstrap-ng was interrupted
[ o.k. ] Process terminated
```

Fourdee, 2016-10-10:
@zador-blood-stained 

> Did you clear rootfs cache?

Whats filepath used by ARMbian?
I always run from a fresh empty directory.

zador-blood-stained, 2016-10-10:
Desktop configuration is currently broken for most devices, this is a known bug.

igorpecovnik, 2016-10-10:
Will be fixed asap. I hope today.

zador-blood-stained, 2016-10-12:
Desktop packages dependency issue should be fixed.

Fourdee, 2016-10-12:
@zador-blood-stained 
Excellent, i'll run another test today to confirm.

Fourdee, 2016-10-12:
@zador-blood-stained 

> [ o.k. ] Done building [ /root/armbian/output/images/Armbian_5.21_Odroidc1_Debia       n_jessie_3.10.103_desktop.img ]
[ o.k. ] Runtime [ 39 min ]

Works a treat, thanks guys 👍

closed 2016-10-12 by igorpecovnik