armbian/build #493: [RfC] Using network-manager by default?

ThomasKaiser, 2016-10-10:
When exploring ways to support NanoPi AIR I played a bit around with `nmcli` on the weekend. IMO we should test using `network-manager` on our "server" builds too since NetworkManager's ability to constantly monitor network connections and re-establish broken connections seems mandatory to me on devices like the AIR that have only Wi-Fi capabilities and might be used in ways where connection drops happen regularly (imagine using an AIR as flight controller on a drone or stuff like that).

In fact on recent desktop images even Ethernet connections are managed by `network-manager` so looking a bit more into ways to set up and configure network connections (configs below `/etc/NetworkManager/system-connections/`) might be useful anyway especially when we also think about ways to allow our end users to pre-define Wi-Fi connections and stuff like that.

To create new setups from scratch with `nmcli` maybe we have to add `uuid` package also. I wanted to look into this a bit more and already prepared some scripted stuff but after [the mess](https://github.com/igorpecovnik/lib/pull/490) happened yesterday I threw the snippets away (silly me, I know). But it seems quite easy to let an AIR being flashed directly through FEL, then reboot and finish configuration through `g_serial` mode and a script asking for SSID/passphrase and creating the appropriate config below `/etc/NetworkManager/system-connections/` set to `0600` mode by default.

Only thing I remember: using `LC_ALL=C nmcli` is mandatory since otherwise i18n/localization do a pretty good job making output parsing harder ;)

ThomasKaiser, 2016-10-10:
Addendum: On a few devices we're able to setup multiple Wi-Fi devices (one in client and one in AP mode). Adding the AP interface based on MAC address would require adding it to `/etc/NetworkManager/NetworkManager.conf` in this way:

    [keyfile]
    unmanaged-devices=mac:<hwaddr>

This way we could enable both Wi-Fi interfaces by default and then it's up to the user to set up `hostapd` to use the specific one if needed.

ThomasKaiser, 2016-10-10:
Ah, I did it wrong when testing, establishing a new Wi-Fi connection (after optionally choosing from available networks by `nmcli dev wifi list`) is even more easy:

    nmcli con add con-name MyWiFi ifname wlan0 type wifi ssid MyWiFi ip4 192.168.1.101/24 gw4 192.168.1.1
    nmcli con modify MyWiFi wifi-sec.key-mgmt wpa-psk
    nmcli con modify MyWiFi wifi-sec.psk passphrase

`network-manager` is then doing the rest, creating PSK and storing the config using a new UUID. Since we should also keep in mind that we'll have to support at least [one Laptop](https://olimex.wordpress.com/2016/08/31/teres-i-diy-open-source-hardware-hackers-laptop-update/) soon ;) using profiles/configs with automatic connect depending on SSIDs found (at work Armbian will use SSID `office` while at home it's `My Cave` instead) is IMO the way to go (and static config files are unusable anyway).

ThomasKaiser, 2016-10-10:
Oh my... used to think way too complicated all the time, it's really just

    tk@orangepilite:~$ nmcli dev wifi list
    *  SSID                     MODE   CHAN  RATE       SIGNAL  BARS  SECURITY  
       EasyBox-116D28           Infra  9     54 Mbit/s  59      ▂▄▆_  WPA1 WPA2 
       Snort-Honeynet           Infra  13    54 Mbit/s  59      ▂▄▆_  WPA2      
       --                       Infra  6     54 Mbit/s  44      ▂▄__  WPA2      
       DISTORTEDPEOPLE          Infra  9     54 Mbit/s  40      ▂▄__  WPA1 WPA2 
       WLAN-341381              Infra  1     54 Mbit/s  22      ▂___  WPA2      
       FRITZ!Box Fon WLAN 7360  Infra  1     54 Mbit/s  17      ▂___  WPA1 WPA2 
    
    tk@orangepilite:~$ sudo nmcli --ask dev wifi connect 'Snort-Honeynet'
    [sudo] password for tk: 
    Password: 
    Device 'wlan0' successfully activated with 'ea8c06e7-4edd-4455-a336-4e736e9bef74'.

igorpecovnik, 2016-10-10:
Yes, nm is the way to go. I also did some tests (no advanced) and so far everything was working fine. It looks like it's matured and reliable but more tests has to be done. And we have an very easy tool for CLI interface: http://she-devel.com/nmtui.png

ThomasKaiser, 2016-10-10:
> I mean this is even more simple :)

O yeah, I fiddled around on OS X with `uuidgen` command and two javascripts (to create the PSK) to prepare NM configs and `wpa_supplicant.conf` there while it's that easy :)

Another nice thing is that NM is also monitoring link states so disconnecting Ethernet cable will automagically adjust routes too to use Wi-Fi interface instead if configured before. Wow, Wi-Fi on Linux the first time not PITA :)  (more or less like on OS X where `configd` does all the stuff in a similar way and can be accessed through CLI tools also)

But I agree: Needs a lot of testing.

ThomasKaiser, 2016-10-10:
Wah! Not even the name of the device is important, NM simply doesn't care how the device is called  and uses the most capable one. So even if the device names when using [jernej's `/proc/driver/wifi-pm/power` trick](http://forum.armbian.com/index.php/topic/872-beelink-x2-with-armbian-possible/page-8#entry16594) would be different or a users inserts every time different USB dongles not even a configuration change is required.

igorpecovnik, 2016-10-10:
I know about few of those fancy features but avoid it because it used to be buggy. Complex tools needs some time to become mature. Let our [volunteer testers](http://forum.armbian.com/index.php/topic/2062-test-team-testers/) join to this job? More people = more chances that we found something critical.

ThomasKaiser, 2016-10-10:
> Let our volunteer testers join to this job?

Good idea. I just added the necessary steps to deal with both Wi-Fis on Pine64+ (`wlan0` and `wlan1` being available). The [last paragraph here](https://github.com/igorpecovnik/lib.docs/blob/master/docs/board_details/pine64.md) shows how to install the stuff and how to properly blacklist/hide `wlan1`

ThomasKaiser, 2016-10-10:
To prepare some tests done by volunteers:

I tried the following on a pure headless installation (Xenial 'server', kernel 4.8 on an Orange Pi Plus 2E):

    apt-get --no-install-recommends install network-manager
    systemctl start network-manager
    nmcli dev wifi list
    nmtui
    nmcli device show 

Installation of `network-manager` will most probably inform you that all interfaces already configured using `/etc/network/interfaces` won't be available from NetworkManager so at least remove Wi-Fi interfaces from there before starting `network-manager`.

You can also exclude devices from being managed through NetworkManager bei either adding them to a file in `/etc/NetworkManager/conf.d` (see `man NetworkManager.conf` and search for `unmanaged-devices`).

Areas of testing:

- nmcli (examples above)
- nmtui (first choose 'Activate a connection' and later 'Edit a connection')
- tests with onboard Wi-Fi and additional USB Wi-Fi sticks
- managing Ethernet through `interfaces` file and Wi-Fi through `network-manager` and vice versa.
- manage all NICs using `network-manager`
- test link changes (disconnect Ethernet cable, increase distance to AP) and watch what happens (eg. checking routing table `netstat -rn`)
- interoperability with our `hostapd`
- (what did I forget?)

longcloudnz1, 2016-10-16:
I did some testing on my orangepi pc+ using jessie desktop.
Initially I set up a static address in interfaces and connected the wifi using a mixture of the network manager gui and the connections icon on the top right hand corner of the desktop.
I then disconnected the eth0 cable.
Although I could ping the access point I could not ping anything else.  The ping failed with a message from eth0. I assume network manager does not try and fiddle with default routes if it sees a connection in interfaces?
If I remove eth0 from interfaces and add it into network manager using nmtui it mostly works properly(see next paragraph)I can remove the lan cable and it will automatically start using wifi. If I turn wifi off eth0 will become active again.
There could be one small problem though, but possibly will not affect many people. During investigation into why I could not ping I tried connecting to another access point. So I had an eth0 and two wifi access points to connect to. If I tried to disconnect and reconnect between the two wifi access points it would ask me again for the wifi password of one access point. Maybe network manager only remembers one wifi password by default?
Also the network manager gui through the menus does not have root privs so I could not set anything up. Also network manger gui called through the root command line has problems discovering if you have changed field data as the ok button is often greyed out when it shouldn't be.
Better to use nmtui or the pure command line tools.
This testing was done via jessie desktop. The server os may perform differently. There seems to be some database duplication between the pure gui and the command line, with some sort of duplicates showing up in the desktop taskbar. It still works though.
At the moment I will continue to use network manager for everything as the quirks are not really a problem for me. If something major turns up in the future I couod reconsider. I want to get into containers or systemd nspawn in the future so hopefully everthing will still work properly.

longcloudnz1, 2016-10-17:
I added a 10/100mbs dongle to my orange pi and it recognised it and connected automatically.
I have unplugged ethernet cables, pinged the ethernet and the fallover to a working connection seems to work properly.
When I replug the cables back into the orangepi it also reconnects automatically.
I unplugged both ethernet cables and then disabled the access point. I couldn't ping and the desktop told me I wasn't connected anymore. I then re enabled the access point, wifi came back on automatically and I could ping again.
So NetworkManager seems to work as advertised, so far.

ThomasKaiser, 2016-10-20:
BTW: Average consumption of a Banana Pro with active `network-manager` and connected to a Wi-Fi network here is 250 mW **less** than without. No idea why but it seems `network-manager` does some hardware initialisation when Wi-Fi is brought up that is reponsible for the lowered consumption ([details](http://sprunge.us/JNLL)). Power management is active:

    root@bananapro:~# iwconfig wlan0 
    wlan0     IEEE 802.11  ESSID:"Snort-Honeynet"  
              Mode:Managed  Frequency:2.472 GHz  Access Point: BC:05:43:BE:C1:E7   
              Bit Rate=65 Mb/s   Tx-Power=31 dBm   
              Retry short limit:7   RTS thr:off   Fragment thr:off
              Encryption key:off
              Power Management:on
              Link Quality=55/70  Signal level=-55 dBm  
              Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
              Tx excessive retries:2  Invalid misc:0   Missed beacon:0

yon2004, 2016-10-23:
This is great can we change the config so that network manager can manage DNS for the system too. The static name-server approach does not work in lots of situations.

/etc/NetworkManager/NetworkManager.conf

[main]
plugins=ifupdown,keyfile,ofono
rc-manager=file

[ifupdown]
managed=false

zgoda, 2016-10-26:
Following @ThomasKaiser hint on forum, I switched my Jessie OPi PC with Ralink RT5370 USB wifi dongle to NM and all I can say it works now. First removed all configuration from `/etc/network/interfaces` and `/etc/wpa_supplicant/wpa_supplicant.conf`, gathered some information using `nmcli`, then used `nmtui` to setup connection.

ThomasKaiser, 2016-11-02:
Think this can be closed now.

closed 2016-11-02 by ThomasKaiser