armbian/build #496: Support  realtek RT5572

alexandre911, 2016-10-11:
It is possible to add support to RT5572?, this network card is in recent realeses of mainline kernel.

I read this post: http://forum.armbian.com/index.php/topic/157-add-wifi-driver-ralink-rt5572-rt2800usb/

Propietary driver works well using managed mode only but doesn't work in monitor mode or ap. Using backports RT5572 works well in all modes but I can't use internal wifi of Orange Pi PC+ because we need load mac80211 from backports and kernel give symbol error.

zador-blood-stained, 2016-10-11:
Once H3 gets proper mainline support it will be possible to use any drivers available in mainline and compatible with mainline. I'm not sure anybody will try to fix legacy kernel unless working patch is provided.

zador-blood-stained, 2016-11-24:
Merging this issue into #558

closed 2016-11-24 by zador-blood-stained