armbian/build #497: nand-sata-install.sh doesn't update boot script correctly

neomanic, 2016-10-12:
Newer Armbian versions use a boot.cmd with `setenv bootdev /dev/mmcblk0p1` at the top, rather than the old `root=` kernelarg directly.

The EMMC install in nand-sata-install.sh hasn't been updated this variable rather than directly.

Attached patch fixes the regexp, but you may want to alter it further?

```
diff --git a/scripts/nand-sata-install/usr/lib/nand-sata-install/nand-sata-install.sh b/scripts/nand-sata-install/usr/lib/nand-sata-install/nand-sata-install.sh
index 207ad0c..526b318 100644
--- a/scripts/nand-sata-install/usr/lib/nand-sata-install/nand-sata-install.sh
+++ b/scripts/nand-sata-install/usr/lib/nand-sata-install/nand-sata-install.sh
@@ -118,7 +118,7 @@ EOF
                cp -R /boot/ /mnt/bootfs
                
                # eMMC install          
-               sed -e 's,root='"$root_partition"',root='"$emmcuuid"',g' -i /mnt/bootfs/boot/boot.cmd           
+               sed -e 's,setenv rootdev '"$root_partition"',setenv rootdev '"$emmcuuid"',g' -i /mnt/bootfs/boot/boot.cmd               
                mkimage -C none -A arm -T script -d /mnt/bootfs/boot/boot.cmd /mnt/bootfs/boot/boot.scr >/dev/null 2>&1 || (echo "Error"; exit 0)       
                
                # fstab adj
```

[nand-sata-install-fixboot.patch.txt](https://github.com/igorpecovnik/lib/files/523272/nand-sata-install-fixboot.patch.txt)

neomanic, 2016-10-12:
Note that new kernels seem to always identify the eMMC as /dev/mmcblk1 regardless of whether a microSD card is present or not, so this is required in order to boot.

martinayotte, 2016-10-12:
With Mainline and OrangePi-PC+ or OrangePi-Plus2E, it is /dev/mmcblk2 (not /dev/mmcblk1).

neomanic, 2016-10-12:
So the same issue would apply in that case with those other boards, right @martinayotte?

martinayotte, 2016-10-12:
I can't confirm since I'm only using Mainline kernel.
From what I know, the device /dev/mmcblk2 is "2" on PC+ and Plus2E because SDIO1 is used by WiFi interfaces, so other boards which don't have such WiFi could have eMMC be still located at slot "1".

ThomasKaiser, 2016-10-12:
Please keep in mind that Zador's implementing a better approach currently: https://github.com/igorpecovnik/lib/commit/e48da4e991af20673b55c0d2be3db931d1f8c4ba

I would assume a lot of testing is necessary (read as: testers needed! ;) ) but at least we're then not affected by changing device names as it's now the case with vanilla kernel (change `mmc` stuff in DT and you bricked the board)

neomanic, 2016-10-12:
Ah, yes. I was only looking at the nand-sata-install script. We can just leave this issue alone for now until @zador-blood-stained finishes the work on the original install side of things.

zador-blood-stained, 2016-10-23:
This should be fixed now

closed 2016-10-23 by zador-blood-stained