armbian/build #498: Fix PWM for NanoPi Neo

roberthartung, 2016-10-12:
I've discovered that even if I load the pwm kernel module, nothing happens. It looks like there is just a define missing. When looking at the official FriendlyArm repo ( https://github.com/friendlyarm/h3_lichee/blob/master/linux-3.4/drivers/pwm/pwm-sunxi.c ) and comparing it to the official sunxi repo ( https://github.com/allwinner-zh/linux-3.4-sunxi/blob/master/drivers/pwm/pwm-sunxi.c ) you can see that the only difference is the define `CONFIG_ARCH_SUN8IW7P1` which is the arch from NanoPi Neo ( http://wiki.friendlyarm.com/wiki/index.php/NanoPi_NEO ).

My question here is can we just add a patch? Or can I fork some repo and change the URL for the repo for the kernel? I didn't find any information where the actual kernel modules are loaded from.

ThomasKaiser, 2016-10-12:
Yes, it seems FA patched this after we cloned their repo.

It would be great if you use our build system, create the [necessary patch](https://github.com/friendlyarm/h3_lichee/commit/9d43538162796100f16026a6ee26d826dd82320f#diff-3ec0ca26221dddd3796b7f169ed733a4.patch) and integrate it as `userpatches/kernel/sun8i-default/`0028-fix-pwm.patch` and then build a kernel with

    ./compile.sh KERNEL_ONLY=yes

(choosing NEO and default) and report back whether that fixes the issue. You get in `output/debs/` a linux-image `.deb` that can be installed using `dpkg -i`

ThomasKaiser, 2016-10-12:
Hmm... apart from the changes to `drivers/pwm/pwm-sunxi.c` the other stuff (eg. accessing DHT11/DHT22 through kernel driver) lookseven more interesting.

@zador-blood-stained @igorpecovnik any opinions on including the whole patch set into sun8i legacy?

igorpecovnik, 2016-10-12:
Added, try now.

roberthartung, 2016-10-13:
First of all I can tell that this patch does not resolve the issue. The patch only adds matrix_pwm if I am right here, but I was referring to pwm-sunxi.c - but the file isn't changed at all?

Furthermore only loading matrix_pwm gives me:

```
Hardware:sun8i
Revision:0000
BoardType:68330
Using default config: channel=0 freq=1000Hz duty=500
FAHW-Lib: Fail to set pwm
Fail to output PWM
Press enter to stop PWM
```

When additionally loading sunxi-pwm:

```
root@nanopineo:~# modprobe pwm-sunxi 
root@nanopineo:~# matrix-pwm 
Hardware:sun8i
Revision:0000
BoardType:68330
Using default config: channel=0 freq=1000Hz duty=500
Press enter to stop PWM
```

The error is gone, but still no PWM signal I can measure! I will try to use the ubuntu-core for now to see if it is working at all!

roberthartung, 2016-10-13:
Currently I am not able to use matrix_pwm on the ubuntu-core image as well ... I don't have to load sunxi_pwm here (at least I don't get an error). But still no PWM signal on the output pin!

I try to manually add the kernel patch to the build system now and try to make it work!

roberthartung, 2016-10-13:
So I put again hours into this. I don't get any PWM output on the pins. I can set it to High/Low  when using it as a GPIO. I don't know what's the problem here! the matrix-pwm module is loaded and doesnt report any errors...

igorpecovnik, 2016-10-13:
I get patch from here: 
https://github.com/friendlyarm/h3_lichee/commit/9d43538162796100f16026a6ee26d826dd82320f
Check @wwd Friendly Arm if it's working for them?

ThomasKaiser, 2016-10-13:
> the file didnt change in @igorpecovnik's commit

The patch missed the `.patch` suffix in the beginning which we realized and fixed a few hours later yesterday. 

Anyway, we're more server guys so it's really better to check basic issues with FriendlyARM first whether and especially how it should work there (my understanding is that the PWM pin is multiplexed on the UART header so without further tweaks in fex file it can't work since physically not exposed -- disclaimer: me an absolute noobie in this area)

roberthartung, 2016-10-13:
PWM1 isn't multiplexed. I guess I should be able to use it directly?

zador-blood-stained, 2017-01-09:
I just tested the PWM on Orange Pi Plus 2E, but configuration steps for other H3 boards should be similar.

Please note that hardware PWM is supported only on pin PA05 according to the tests that were done during the PWM driver mainlining, so UART0 should be disabled and output signal can be grabbed from RX pin of the debug UART (4-pin header near the USB port in case of the Neo).

Necessary steps for the legacy kernel:

1. Editing the `script.fex`:

- UART0 should be disabled
```
[uart0]
uart_used = 0
uart_port = 0
uart_type = 2
uart_tx = port:PA04<2><1><default><default>
uart_rx = port:PA05<2><1><default><default>
```
- PWM 0 should be enabled
```
[pwm0_para]
pwm_used = 1
pwm_positive = port:PA05<3><0><default><default>
```

- Reboot is needed to apply any changes to the script.bin

2. Installing the Matrix library

```
git clone https://github.com/friendlyarm/matrix.git --depth 1
cd matrix
make
make install
```

3. Testing

- Load the pwm_sunxi module
```
sudo modprobe pwm_sunxi
```

- Run the test program
```
cd matrix/demo/matrix-pwm/
# for the standard parameters - 1kHz, 50% duty cycle
./matrix-pwm
# for the user-supplied patameters - 5kHz, 25% duty cycle
./matrix-pwm 0 5000 750
```

- Observe the results
[image]
[image]

So this can be considered fixed.

closed 2017-01-09 by zador-blood-stained

neon12345, 2017-03-01:
Is it somehow possible to invert the pwm signal in cycle mode? Start at low and end at low if pwm is enabled/disabled?