armbian/build #500: Getting SPI to work on lime2 4.7.3-sunxi

multivac61, 2016-10-13:
So far I have been unsuccessful at getting the SPI bus to work on my Olimex Lime 2 eMMC. I have compiled [slightly modified .dts files](https://gist.github.com/multivac61/3ab2cbd75667f59c58d38a4aef26b38d) from the onces llisted [here](http://forum.armbian.com/index.php/topic/1537-397-maintainfix-dts-entries-for-some-devices-such-i2cspiw1/) and catenated them into `/sys/kernel/config/device-tree/overlays/mydtbo/dtbo` (or similar) as suggested by @martinayotte. I get the following error in dmesg:
```
[  948.948454] spidev spi32766.0: buggy DT: spidev listed directly in DT
[  948.955159] ------------[ cut here ]------------
[  948.959906] WARNING: CPU: 0 PID: 988 at drivers/spi/spidev.c:719 spidev_probe+0x1bc/0x1d8 [spidev]
[  948.969033] Modules linked in: spidev(+) fbtft(C) syscopyarea sysfillrect sysimgblt fb_sys_fops cfg80211 rfkill evdev sun4i_ts nvmem_sunxi_sid nvmem_core sun4i_ss cpufreq_dt thermal_sys uio_pdrv_genirq uio bonding fuse realtek
[  948.989707] CPU: 0 PID: 988 Comm: systemd-udevd Tainted: G         C      4.7.3-sunxi #26
[  948.997929] Hardware name: Allwinner sun7i (A20) Family
[  949.003219] [<c0110c1c>] (unwind_backtrace) from [<c010caac>] (show_stack+0x20/0x24)
[  949.011021] [<c010caac>] (show_stack) from [<c06185b0>] (dump_stack+0x8c/0xa0)
[  949.018300] [<c06185b0>] (dump_stack) from [<c0121c04>] (__warn+0xf8/0x110)
[  949.025299] [<c0121c04>] (__warn) from [<c0121cec>] (warn_slowpath_null+0x30/0x38)
[  949.032937] [<c0121cec>] (warn_slowpath_null) from [<bf184330>] (spidev_probe+0x1bc/0x1d8 [spidev])
[  949.042065] [<bf184330>] (spidev_probe [spidev]) from [<c0726fcc>] (spi_drv_probe+0x8c/0xb8)
[  949.050565] [<c0726fcc>] (spi_drv_probe) from [<c06bb0d8>] (driver_probe_device+0x21c/0x2c8)
[  949.059062] [<c06bb0d8>] (driver_probe_device) from [<c06bb24c>] (__driver_attach+0xc8/0xcc)
[  949.067555] [<c06bb24c>] (__driver_attach) from [<c06b9020>] (bus_for_each_dev+0x78/0xac)
[  949.075786] [<c06b9020>] (bus_for_each_dev) from [<c06ba86c>] (driver_attach+0x2c/0x30)
[  949.083844] [<c06ba86c>] (driver_attach) from [<c06ba3ec>] (bus_add_driver+0x1b4/0x22c)
[  949.091901] [<c06ba3ec>] (bus_add_driver) from [<c06bbcf4>] (driver_register+0x88/0x108)
[  949.100048] [<c06bbcf4>] (driver_register) from [<c0726ef8>] (__spi_register_driver+0x68/0x6c)
[  949.108722] [<c0726ef8>] (__spi_register_driver) from [<bf18c098>] (spidev_init+0x98/0xd0 [spidev])
[  949.117834] [<bf18c098>] (spidev_init [spidev]) from [<c0101c70>] (do_one_initcall+0x54/0x180)
[  949.126501] [<c0101c70>] (do_one_initcall) from [<c01ff548>] (do_init_module+0x74/0x3b0)
[  949.134650] [<c01ff548>] (do_init_module) from [<c01a7b60>] (load_module+0x19cc/0x20b8)
[  949.142706] [<c01a7b60>] (load_module) from [<c01a8470>] (SyS_finit_module+0xb8/0xc8)
[  949.150590] [<c01a8470>] (SyS_finit_module) from [<c0108240>] (ret_fast_syscall+0x0/0x3c)
[  949.158870] ---[ end trace 73eb6cb955088dde ]---
[  949.164169] spidev spi32765.0: buggy DT: spidev listed directly in DT
[  949.170661] ------------[ cut here ]------------
[  949.175447] WARNING: CPU: 0 PID: 988 at drivers/spi/spidev.c:719 spidev_probe+0x1bc/0x1d8 [spidev]
[  949.184487] Modules linked in: spidev(+) fbtft(C) syscopyarea sysfillrect sysimgblt fb_sys_fops cfg80211 rfkill evdev sun4i_ts nvmem_sunxi_sid nvmem_core sun4i_ss cpufreq_dt thermal_sys uio_pdrv_genirq uio bonding fuse realtek
[  949.204925] CPU: 0 PID: 988 Comm: systemd-udevd Tainted: G        WC      4.7.3-sunxi #26
[  949.213118] Hardware name: Allwinner sun7i (A20) Family
[  949.218378] [<c0110c1c>] (unwind_backtrace) from [<c010caac>] (show_stack+0x20/0x24)
[  949.226149] [<c010caac>] (show_stack) from [<c06185b0>] (dump_stack+0x8c/0xa0)
[  949.233398] [<c06185b0>] (dump_stack) from [<c0121c04>] (__warn+0xf8/0x110)
[  949.240375] [<c0121c04>] (__warn) from [<c0121cec>] (warn_slowpath_null+0x30/0x38)
[  949.247976] [<c0121cec>] (warn_slowpath_null) from [<bf184330>] (spidev_probe+0x1bc/0x1d8 [spidev])
[  949.257056] [<bf184330>] (spidev_probe [spidev]) from [<c0726fcc>] (spi_drv_probe+0x8c/0xb8)
[  949.265520] [<c0726fcc>] (spi_drv_probe) from [<c06bb0d8>] (driver_probe_device+0x21c/0x2c8)
[  949.273982] [<c06bb0d8>] (driver_probe_device) from [<c06bb24c>] (__driver_attach+0xc8/0xcc)
[  949.282441] [<c06bb24c>] (__driver_attach) from [<c06b9020>] (bus_for_each_dev+0x78/0xac)
[  949.290640] [<c06b9020>] (bus_for_each_dev) from [<c06ba86c>] (driver_attach+0x2c/0x30)
[  949.298666] [<c06ba86c>] (driver_attach) from [<c06ba3ec>] (bus_add_driver+0x1b4/0x22c)
[  949.306693] [<c06ba3ec>] (bus_add_driver) from [<c06bbcf4>] (driver_register+0x88/0x108)
[  949.314807] [<c06bbcf4>] (driver_register) from [<c0726ef8>] (__spi_register_driver+0x68/0x6c)
[  949.323446] [<c0726ef8>] (__spi_register_driver) from [<bf18c098>] (spidev_init+0x98/0xd0 [spidev])
[  949.332519] [<bf18c098>] (spidev_init [spidev]) from [<c0101c70>] (do_one_initcall+0x54/0x180)
[  949.341155] [<c0101c70>] (do_one_initcall) from [<c01ff548>] (do_init_module+0x74/0x3b0)
[  949.349270] [<c01ff548>] (do_init_module) from [<c01a7b60>] (load_module+0x19cc/0x20b8)
[  949.357296] [<c01a7b60>] (load_module) from [<c01a8470>] (SyS_finit_module+0xb8/0xc8)
[  949.365149] [<c01a8470>] (SyS_finit_module) from [<c0108240>] (ret_fast_syscall+0x0/0x3c)
[  949.373392] ---[ end trace 73eb6cb955088ddf ]---
```

Even on [linux-sunxi.org](http://linux-sunxi.org/SPIdev) there doesn't seam to be an awful lot of information on this topic. As far as I can tell the configuration files and .fex files for the Lime 2 are correct and so the thing that I am not able to work our correctly is the device tree overlays.

zador-blood-stained, 2016-10-13:
These are warnings, not errors. If SPI is configured correctly, it should work.
Please also check this topic: http://forum.armbian.com/index.php/topic/637-armbian-483-spi-driver/

martinayotte, 2016-10-14:
Exactly, "Buggy DT" is only a warning. 
On OrangePiPC, my spi-enable.dtbs overlay is working without any issue.

multivac61, 2016-10-14:
Alright, thank you for the help. I didn't realize that the normal `/dev/spi0.0` would be appearing under `spidev327xx.0`. I am still having problem with the SPI bus but you may close this issue. I will open a new issue  #501 addressing that particular case.

closed 2016-10-14 by zador-blood-stained

martinayotte, 2016-10-14:
The fact it is appearing as /dev/spidev327xx.0 instead of /dev/spi0.0 is due to the fact of missing alias in DT.