armbian/build #501: Linux lime2 4.7.3-sunxi - SPI transfer failed: -90

multivac61, 2016-10-14:
Now that I have SPI working on my Olimex Lime 2 #500 I would like to connect a TFT screen to it. My setup is surprisingly similar to the one @igorpecovnik had [here](http://forum.armbian.com/index.php/topic/77-spi-on-allwinner-banana-cubietruck/).

While Igor had a spi error code `-22` when running `modprobe fbtft_device name=adafruit22a busnum=32764 gpios=dc:128,reset:129,led:34 debug=7` I get `-90` which is [a system error code](http://www-numi.fnal.gov/offline_software/srt_public_context/WebDocs/Errors/unix_system_errors.html) that means "Message too long". 

I have tried increasing the SPI buffersize by running `modprobe spidev bufsiz=65536` but I still get the same errors:

```
...
[  134.677691] fb_ili9340 spi32766.0: fbtft_write_spi(len=3840): 00 00 00 00 ad 55 ad 55 ad 55 00 00 00 00 00 00 00 00 00 00 00 00 ad 55 ad 55 00 00 00 00 00 00 ...
[  134.677765] fb_ili9340 spi32766.0: fbtft_update_display: write_vmem failed to update display buffer
[  134.877063] fb_ili9340 spi32766.0: fbtft_update_display(start_line=232, end_line=239)
[  134.877101] fb_ili9340 spi32766.0: fbtft_write_reg8_bus8: 2a 00 00 00 ef
[  134.877131] fb_ili9340 spi32766.0: fbtft_write_spi(len=1): 2a
[  134.877265] fb_ili9340 spi32766.0: fbtft_write_spi(len=4): 00 00 00 ef
[  134.877396] fb_ili9340 spi32766.0: fbtft_write_reg8_bus8: 2b 00 e8 00 ef
[  134.877422] fb_ili9340 spi32766.0: fbtft_write_spi(len=1): 2b
[  134.877488] fb_ili9340 spi32766.0: fbtft_write_spi(len=4): 00 e8 00 ef
[  134.877545] fb_ili9340 spi32766.0: fbtft_write_reg8_bus8: 2c
[  134.877568] fb_ili9340 spi32766.0: fbtft_write_spi(len=1): 2c
[  134.877624] fb_ili9340 spi32766.0: fbtft_write_vmem16_bus8(offset=111360, len=3840)
[  134.877679] fb_ili9340 spi32766.0: fbtft_write_spi(len=3840): 00 00 00 00 ad 55 ad 55 ad 55 00 00 00 00 00 00 00 00 00 00 00 00 ad 55 ad 55 00 00 00 00 00 00 ...
[  134.877751] fb_ili9340 spi32766.0: fbtft_update_display: write_vmem failed to update display buffer
[  135.077071] fb_ili9340 spi32766.0: fbtft_update_display(start_line=232, end_line=239)
[  135.077109] fb_ili9340 spi32766.0: fbtft_write_reg8_bus8: 2a 00 00 00 ef
[  135.077139] fb_ili9340 spi32766.0: fbtft_write_spi(len=1): 2a
[  135.077275] fb_ili9340 spi32766.0: fbtft_write_spi(len=4): 00 00 00 ef
[  135.077402] fb_ili9340 spi32766.0: fbtft_write_reg8_bus8: 2b 00 e8 00 ef
[  135.077431] fb_ili9340 spi32766.0: fbtft_write_spi(len=1): 2b
[  135.077501] fb_ili9340 spi32766.0: fbtft_write_spi(len=4): 00 e8 00 ef
[  135.077558] fb_ili9340 spi32766.0: fbtft_write_reg8_bus8: 2c
[  135.077582] fb_ili9340 spi32766.0: fbtft_write_spi(len=1): 2c
[  135.077639] fb_ili9340 spi32766.0: fbtft_write_vmem16_bus8(offset=111360, len=3840)
[  135.077693] fb_ili9340 spi32766.0: fbtft_write_spi(len=3840): 00 00 00 00 ad 55 ad 55 ad 55 00 00 00 00 00 00 00 00 00 00 00 00 ad 55 ad 55 00 00 00 00 00 00 ...
[  135.077764] fb_ili9340 spi32766.0: fbtft_update_display: write_vmem failed to update display buffer
[  135.277083] fb_ili9340 spi32766.0: fbtft_update_display(start_line=232, end_line=239)
[  135.277121] fb_ili9340 spi32766.0: fbtft_write_reg8_bus8: 2a 00 00 00 ef
[  135.277149] fb_ili9340 spi32766.0: fbtft_write_spi(len=1): 2a
[  135.277277] fb_ili9340 spi32766.0: fbtft_write_spi(len=4): 00 00 00 ef
[  135.277429] fb_ili9340 spi32766.0: fbtft_write_reg8_bus8: 2b 00 e8 00 ef
[  135.277457] fb_ili9340 spi32766.0: fbtft_write_spi(len=1): 2b
[  135.277525] fb_ili9340 spi32766.0: fbtft_write_spi(len=4): 00 e8 00 ef
[  135.277582] fb_ili9340 spi32766.0: fbtft_write_reg8_bus8: 2c
[  135.277605] fb_ili9340 spi32766.0: fbtft_write_spi(len=1): 2c
[  135.277666] fb_ili9340 spi32766.0: fbtft_write_vmem16_bus8(offset=111360, len=3840)
[  135.277720] fb_ili9340 spi32766.0: fbtft_write_spi(len=3840): 00 00 00 00 ad 55 ad 55 ad 55 00 00 00 00 00 00 00 00 00 00 00 00 ad 55 ad 55 00 00 00 00 00 00 ...
[  135.277792] fb_ili9340 spi32766.0: fbtft_update_display: write_vmem failed to update display buffer
[  135.477127] fb_ili9340 spi32766.0: fbtft_update_display(start_line=232, end_line=239)
[  135.477170] fb_ili9340 spi32766.0: fbtft_write_reg8_bus8: 2a 00 00 00 ef
[  135.477200] fb_ili9340 spi32766.0: fbtft_write_spi(len=1): 2a
[  135.477385] fb_ili9340 spi32766.0: fbtft_write_spi(len=4): 00 00 00 ef
[  135.477461] fb_ili9340 spi32766.0: fbtft_write_reg8_bus8: 2b 00 e8 00 ef
[  135.477485] fb_ili9340 spi32766.0: fbtft_write_spi(len=1): 2b
[  135.477566] fb_ili9340 spi32766.0: fbtft_write_spi(len=4): 00 e8 00 ef
[  135.477626] fb_ili9340 spi32766.0: fbtft_write_reg8_bus8: 2c
[  135.477650] fb_ili9340 spi32766.0: fbtft_write_spi(len=1): 2c
[  135.477723] fb_ili9340 spi32766.0: fbtft_write_vmem16_bus8(offset=111360, len=3840)
[  135.477776] fb_ili9340 spi32766.0: fbtft_write_spi(len=3840): 00 00 00 00 ad 55 ad 55 ad 55 00 00 00 00 00 00 00 00 00 00 00 00 ad 55 ad 55 00 00 00 00 00 00 ...
[  135.477857] fb_ili9340 spi32766.0: fbtft_update_display: write_vmem failed to update display buffer
```

Has anyone here successfully connected a `ili9340` screen to their Olimex computer, or can lead me to the right direction?

multivac61, 2016-10-14:
Btw I know that the SPI bus is on `/dev/32764.0` because I have tun the ever so popular `spidev_test`.

zador-blood-stained, 2016-10-14:
Mainline SPI driver has rather small transfer size limits as far as I know.

multivac61, 2016-10-14:
@zador-blood-stained hmm, there must be some way to increase that? AFAIK the same works on the mainline driver for Raspberry Pis.

zador-blood-stained, 2016-10-14:
Yes, there were some experimental patches, but they never reached upstream and now they will need some rework since SPI driver code was changed.
For example, this: https://groups.google.com/forum/#!searchin/linux-sunxi/SPI$20sun4i|sort:relevance/linux-sunxi/M8ZSFzWKjfY/vlAzi7Yl66sJ
and this: https://groups.google.com/forum/#!searchin/linux-sunxi/SPI$20sun4i|sort:relevance/linux-sunxi/UAC2ayDFih4/Fc-MMackkyEJ

itdaniher, 2016-11-01:
There's also https://patchwork.kernel.org/patch/9173711/, which seems to be a more modern approach for adding long SPI transfers via DMA, instead of by CPU-backed interrupts.

itdaniher, 2016-11-04:
@multivac61 I have successfully updated a patch (v6-ARM-sun4i-spi-Allow-transfers-larger-than-FIFO-size.patch) to cleanly apply to sunxi-next and provide interrupt-driven SPI FIFO handling. I have tested this patch for several days of continuous SPI reads on a banana pi and seen no issues.

My patch is available at http://sprunge.us/ALKA - it's a pretty small edit on v6 to change some additional -90 checks added recently.

I will be considering exploring the DMA patch, but it did not cleanly apply and the much-simpler interrupt patch meets my needs. 

I do not think the interrupt patch is mainlineable given that DMA is the ideal solution, but it should meet your needs.

As I think you're also using an A20 board, I would be glad to provide the kernel .deb I built including the patch, let me know!

itdaniher, 2016-11-04:
@zador-blood-stained @igorpecovnik  is there a weaker burden of proof to include a patch with armbian? This is a pretty popular feature request on boards using sun4i-spi, and all mainlining efforts around the interrupt or DMA patches appear to have stalled.

multivac61, 2016-11-05:
@itdaniher I am away for the weekend but will check the patch out once I get back home. Looks good.

zador-blood-stained, 2016-11-05:
For the reference - this patch was added to sunxi-next branch: fdd69714af50ac0197c98b10b3cf8c44599e053d

zador-blood-stained, 2016-12-28:
Closing since the patch was reported as working by @itdaniher 
Now it will be maintained here unless SPI code changes will require too much work to have it running again

closed 2016-12-28 by zador-blood-stained