armbian/build #502: [pine64] boot issue (cpu stuck)

kometchtech, 2016-10-15:
Hello, all

When pine64+ the you try to start in Armbian 5.20, you can not start with an error such as the following.
```
Starting kernel ...

[mmc]: MMC Device 2 not found
[mmc]: mmc 2 not find, so not exit
INFO:    BL3-1: Next image address = 0x41080000
INFO:    BL3-1: Next image spsr = 0x3c9
Loading, please wait...
starting version 229
Begin: Loading essential drivers ... done.
Begin: Running /scripts/init-premount ... done.
Begin: Mounting root file system ... Begin: Running /scripts/local-top ... done.
Begin: Running /scripts/local-premount ... [   36.484607] BUG: soft lockup - CPU#0 stuck for 22s]
[   36.492947] Call trace:
[   36.495655] Call trace:
[   36.498351] Call trace:
[   36.501060] Call trace:
<snip>
```
Thanks,

zador-blood-stained, 2016-10-15:
Legacy or vanilla kernel? Image from download page or self built?

kometchtech, 2016-10-15:
Hello, @zador-blood-stained 

This problem is, what you have downloaded the legacy image (xenial) from the download page, is what was dd to the SD card.

Thanks,

martinayotte, 2016-10-15:
A bit strange that u-boot try to find mmc2.
(On my PinA64 5.17 self-build, the SDCard is on mmc0)

zador-blood-stained, 2016-10-15:
> A bit strange that u-boot try to find mmc2.
(On my PinA64 5.17 self-build, the SDCard is on mmc0)

Looks loke strangely implemented cleanup procedure: https://github.com/longsleep/u-boot-pine64/blob/55c9c8c8ac005b1c00ac948386c60c4a741ebaa9/drivers/mmc/mmc.c#L2852

Anyway I can reproduce this issue, so I'll try to check if it is caused by Armbian patches or DT changes.

zador-blood-stained, 2016-10-26:
After updating pine64-default kernel configuration I can't reproduce this issue anymore. Needs more testing to say for sure if it is fixed or not.

phr34k0, 2017-01-24:
Same errors/bugs as thread starter. Tried ALL legacy and mainline which everyone of them ends up with Segmentation Faults. PINE64+.. Desktop versions are out of the question here as every images i have tried on Armbian and the official site (with android) ends up with either bootloops or segmentation errors.

Tried with 64gb microsd that was shipped together with the board = Throws eMMC R/W errors in dmesg

Tried with an official 32gb non crappy microsd from kingston = This is what it spits out in dmesg .. Was trying to install PHP, Apache, Nginx etc

> [    5.806229] EXT4-fs (mmcblk0p1): re-mounted. Opts: commit=600,errors=remount-ro
[    6.876247] systemd-journald[1175]: Received request to flush runtime journal from PID 1
[    9.734234] libphy: 1c30000.ethernet: probed
[    9.869772] Generic PHY 1c30000.ethernet:00: attached PHY driver [Generic PHY] (mii_bus:phy_addr=1c30000.ethernet:00, irq=-1)
[    9.869788] sun8i-emac 1c30000.ethernet: device MAC address slot 0 02:ba:ba:cc:7b:7b
[   13.954048] sun8i-emac 1c30000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[   22.183451] EXT4-fs (mmcblk0p1): resizing filesystem from 7593472 to 7593728 blocks
[   22.185811] EXT4-fs (mmcblk0p1): resized filesystem to 7593728
[   31.266598] systemd[1]: apt-daily.timer: Adding 9h 13min 18.801918s random time.
[   31.673145] systemd[1]: apt-daily.timer: Adding 6h 24min 7.540240s random time.
[   31.830901] systemd[1]: apt-daily.timer: Adding 4h 28min 30.557636s random time.
[   55.310392] systemd[1]: apt-daily.timer: Adding 4h 36min 40.467474s random time.
[   55.568354] systemd[1]: apt-daily.timer: Adding 1h 14min 9.797248s random time.
[   56.390593] Adding 131068k swap on /var/swap.  Priority:-1 extents:2 across:139260k SS
[   56.816327] systemd[1]: apt-daily.timer: Adding 3h 23min 39.971205s random time.
[   57.237386] systemd[1]: apt-daily.timer: Adding 7h 18min 46.093248s random time.
[   57.442191] systemd[1]: apt-daily.timer: Adding 4h 23min 2.467090s random time.
[ 1894.672135] systemd[1]: apt-daily.timer: Adding 8h 2min 21.358940s random time.
[ 1904.406385] systemd[1]: apt-daily.timer: Adding 5h 11min 7.716058s random time.
[ 1904.668698] systemd[1]: apt-daily.timer: Adding 6h 48min 40.124516s random time.
[ 1906.601820] systemd[1]: apt-daily.timer: Adding 2h 46min 9.589164s random time.
[ 2004.021431] systemd[1]: apt-daily.timer: Adding 5h 58min 17.376619s random time.
[ 2010.315212] systemd[1]: apt-daily.timer: Adding 2h 57min 112.648ms random time.
[ 2010.572181] systemd[1]: apt-daily.timer: Adding 4h 4min 43.495638s random time.
[ 2011.798948] systemd[1]: apt-daily.timer: Adding 1h 49min 52.951532s random time.
[ 2103.927735] python[11107]: unhandled level 0 translation fault (11) at 0x1000f457ce67c, esr 0x92000004
[ 2103.927745] pgd = ffff800062bf1000
[ 2103.931249] [1000f457ce67c] *pgd=00000000a32b1003
[ 2103.935835] , *pud=0000000000000000

[ 2103.935853] CPU: 3 PID: 11107 Comm: python Not tainted 4.9.0-pine64 #3
[ 2103.935856] Hardware name: Pine64+ (DT)
[ 2103.935861] task: ffff80007bb0b200 task.stack: ffff800074134000
[ 2103.935867] PC is at 0xffff893f662c
[ 2103.935870] LR is at 0xffff893f65c4
[ 2103.935874] pc : [<0000ffff893f662c>] lr : [<0000ffff893f65c4>] pstate: 80000000
[ 2103.935876] sp : 0000fffffe3ab020
[ 2103.935879] x29: 0000fffffe3ab020 x28: 0000fffffe3ab218 
[ 2103.935885] x27: 0000ffff86cb76c4 x26: 00000000218a5010 
[ 2103.935892] x25: 0000fffffe3ab1b4 x24: 00000000219c49d0 
[ 2103.935897] x23: 0000000021335be0 x22: 00000000219c49d0 
[ 2103.935903] x21: 0000000000000018 x20: 0000ffff86d11ba8 
[ 2103.935910] x19: 0000ffff8945e000 x18: 00000000a5f2bf3a 
[ 2103.935915] x17: 0000ffff8a0a81c0 x16: 0000ffff8945e1e8 
[ 2103.935921] x15: 0000000011706380 x14: 000000004e0811a1 
[ 2103.935927] x13: 000000006f73f0f1 x12: 0000000000000000 
[ 2103.935933] x11: 0000000000000040 x10: 00000000ad297317 
[ 2103.935939] x9 : 00000000219d3480 x8 : 0000fffffe3aafe8 
[ 2103.935944] x7 : 0000000000000000 x6 : 000000000000002c 
[ 2103.935950] x5 : 0000000000000018 x4 : 0001000f457ce67c 
[ 2103.935956] x3 : 0000ffff843b8000 x2 : 0000000000001dd7 
[ 2103.935961] x1 : 0001000f457ce678 x0 : 0000000002959ba8 

[ 2172.376792] 40-updates[11192]: unhandled level 1 translation fault (11) at 0xffefb05fffd8, esr 0x92000005
[ 2172.376802] pgd = ffff800062bb9000
[ 2172.380296] [ffefb05fffd8] *pgd=00000000a29bc003
[ 2172.384755] , *pud=0000000000000000

[ 2172.384768] CPU: 1 PID: 11192 Comm: 40-updates Not tainted 4.9.0-pine64 #3
[ 2172.384771] Hardware name: Pine64+ (DT)
[ 2172.384775] task: ffff80007b6cbe80 task.stack: ffff8000741f0000
[ 2172.384780] PC is at 0xffffb0ec9828
[ 2172.384783] LR is at 0xffffb0a6dd80
[ 2172.384787] pc : [<0000ffffb0ec9828>] lr : [<0000ffffb0a6dd80>] pstate: 80000000
[... 55 lines cut ...]
[ 2272.063968] x29: 0000ffffe52a8e50 x28: 0000ffffe52a8f68 
[ 2272.063982] x27: 0001000059aca47c x26: 0000aaaae3fe9d30 
[ 2272.063988] x25: 0000ffffe52a8f50 x24: 0000ffffe52a8ee0 
[ 2272.063994] x23: 0000000000000050 x22: 0000ffffe52a8f78 
[ 2272.064000] x21: 0000aaaae3fd32c0 x20: 0000aaaae3fd33b0 
[ 2272.064006] x19: 0000ffff89006ff0 x18: 0000000000000000 
[ 2272.064012] x17: 0000ffff8a4ba900 x16: 0000ffff8a9090a8 
[ 2272.064018] x15: 0000000000000000 x14: ffffffffffffffff 
[ 2272.064024] x13: ffffffffff000000 x12: 0000000000000000 
[ 2272.064029] x11: 0000000000000018 x10: 0101010101010101 
[ 2272.064036] x9 : 0000ffffffffffff x8 : 0101010101010101 
[... cut ...]
Hit:2 http://ppa.launchpad.net/ansible/ansible/ubuntu xenial InRelease                                  
Hit:3 http://ports.ubuntu.com xenial-security InRelease                                                 
Hit:4 http://ports.ubuntu.com xenial-updates InRelease                             
Hit:5 https://repos.influxdata.com/ubuntu xenial InRelease                         
Hit:6 http://ports.ubuntu.com xenial-backports InRelease                          
Hit:7 http://apt.armbian.com xenial InRelease                                     
Segmentation faultsts... 0%

zador-blood-stained, 2017-01-24:
> Same errors/bugs as thread starter. Tried ALL legacy and mainline which everyone of them ends up with Segmentation Faults. PINE64+.. Desktop versions are out of the question here as every images i [...]

I suspect you have HW issues (or HW tolerance issues) since for other people these and other official images work fine. We can't help with thise, your best option is contacting the seller for a replacement or a refund.

phr34k0, 2017-01-24:
@zador-blood-stained Could be but what held me back was if i run a simple website on an Apache webserver with the legacy or mainline Armbian image, it could go on for hours and even days without errors nor bugs. I tried contact support but they said it could be power issue though i have checked that it's not the case.. might be just u-boot? I'm getting calltraces too on Legacy image upon boot

[   36.492947] Call trace:
[   36.495655] Call trace:
[   36.498351] Call trace:
[   36.501060] Call trace:
.....
Please enter password etc etc..

zador-blood-stained, 2017-01-24:
> I tried contact support but they said it could be power issue though i have checked that it's not the case.. 

How did you check this? The only **reliable** way is powering a board via GPIO pins with something like a lab power supply or ATX power supply.

The one thing I know - people tested different Armbian images and those images worked for them, but my board, for example, can't boot from SD anymore, and before that I had similar "Call trace" issues.

phr34k0, 2017-01-24:
@zador-blood-stained By using a stepup converter https://www.adafruit.com/product/2465 with 18650 Li-Po batteries.

zador-blood-stained, 2017-01-24:
> By using a stepup converter, 3.7v to 5v https://www.adafruit.com/product/2465 with 18650 Li-Po batteries.

Connected to the GPIO or to the microUSB port?

phr34k0, 2017-01-24:
@zador-blood-stained Tried both ways, was able to boot without problems. Armbian is the best image i have tried that nailed most frustrating problems like bootloops even with Desktop versions, only that at random times, might be 100% utilization, that it'll start to throw segmentation errors, other then that, it works like a charm. Dmesg shows mostly eMMC or Memory addressing issues

zador-blood-stained, 2017-04-26:
The original issue (soft lockup) was reproduced on longsleep's images, so this can be closed and upstream issue should be used for tracking the status of this bug: https://github.com/longsleep/build-pine64-image/issues/51

closed 2017-04-26 by zador-blood-stained

igorpecovnik, 2018-11-20:
We are getting a modern 4.19.y kernel: https://forum.armbian.com/topic/8795-next-lts-kernel-419y-allwinner-a10-a20-a64-h2-h3-h5-h6-debugging-party/ ... Don't have Pine64 but many other A64 based boards act much better.