armbian/build #503: (At least) on Xenial haveged is not running

ThomasKaiser, 2016-10-17:
FYI: At least on Xenial `haveged` seems not to run:

    root@orangepipcplus:~# systemctl status haveged
    ● haveged.service - Entropy daemon using the HAVEGE algorithm
       Loaded: loaded (/lib/systemd/system/haveged.service; enabled; vendor preset: enabled)
       Active: failed (Result: exit-code) since Mon 2016-10-17 12:51:26 UTC; 3min 21s ago
         Docs: man:haveged(8)
               http://www.issihosts.com/haveged/
     Main PID: 214 (code=exited, status=227/NO_NEW_PRIVILEGES)
    
    Oct 17 12:51:26 orangepipcplus systemd[1]: haveged.service: Main process exited, code=exited, status=227/NO_NEW_PRIVILEGES
    Oct 17 12:51:26 orangepipcplus systemd[1]: haveged.service: Unit entered failed state.
    Oct 17 12:51:26 orangepipcplus systemd[1]: haveged.service: Failed with result 'exit-code'.
    Warning: Journal has been rotated since unit was started. Log output is incomplete or unavailable.

I added logging of available entroy to `firstrun` script (commit https://github.com/igorpecovnik/lib/commit/5861e4088f7c8a772d50133d8aa28e3a0a661333) and got with an OPi PC Plus this:

    ### [firstrun] Recreated SSH keys (entropy: 131 162)

I have to admit that I don't get the idea how to diagnose the issue (still `systemd` noobie)

zador-blood-stained, 2016-10-17:
```
➜  ~  % cat /etc/systemd/system/haveged.service.d/10-no-new-privileges.conf
[Service]
NoNewPrivileges=false
➜  ~  %
```

ThomasKaiser, 2016-10-17:
Maybe adding [`pollinate`](http://blog.dustinkirkland.com/2014/02/random-seeds-in-ubuntu-1404-lts-cloud.html) on Ubuntu too?

zador-blood-stained, 2016-10-17:
> Maybe adding pollinate on Ubuntu too?

IMO one entropyrelated service should be enough, and sinc epollinate doesn't exist in Debian repository, I think haveged should be used for consistency.

ThomasKaiser, 2016-10-17:
Agree, and at least here it didn't help (rebuilt the image with `pollinate` also installed) but it still looks bad:

    root@orangepipcplus:~# grep 'SSH key' /var/log/armhwinfo.log
    ### [firstrun] Recreated SSH keys (entropy: 138 173)
    root@orangepipcplus:~# cat /proc/sys/kernel/random/entropy_avail
    128
    root@orangepipcplus:~# systemctl status pollinate
    ● pollinate.service - Pollinate to seed the pseudo random number generator
       Loaded: loaded (/lib/systemd/system/pollinate.service; enabled; vendor preset: enabled)
       Active: inactive (dead) since Mon 2016-10-17 14:21:49 UTC; 1min 57s ago
     Main PID: 644 (code=exited, status=0/SUCCESS)
    
    Oct 17 14:21:47 orangepipcplus systemd[1]: Starting Pollinate to seed the pseudo random number generator...
    Oct 17 14:21:49 orangepipcplus systemd[1]: Started Pollinate to seed the pseudo random number generator.
    root@orangepipcplus:~# systemctl status haveged
    ● haveged.service - Entropy daemon using the HAVEGE algorithm
       Loaded: loaded (/lib/systemd/system/haveged.service; enabled; vendor preset: enabled)
       Active: failed (Result: exit-code) since Mon 2016-10-17 14:21:43 UTC; 2min 27s ago
         Docs: man:haveged(8)
               http://www.issihosts.com/haveged/
     Main PID: 294 (code=exited, status=227/NO_NEW_PRIVILEGES)
    
    Oct 17 14:21:43 orangepipcplus systemd[1]: haveged.service: Main process exited, code=exited, status=227/NO_NEW_PRIVILEGES
    Oct 17 14:21:44 orangepipcplus systemd[1]: haveged.service: Unit entered failed state.
    Oct 17 14:21:44 orangepipcplus systemd[1]: haveged.service: Failed with result 'exit-code'.
    Warning: Journal has been rotated since unit was started. Log output is incomplete or unavailable.

We should add to some sort of 'Armbian default security' document that checking `grep 'SSH key' /var/log/armhwinfo.log` is a good idea and regenerating the keys if in doubt when more entropy is available (there mentioning 'root allowed to do SSH' also and so on).

Still don't understand why `haveged` is running on an ODROID-C1 (also Xenial) but not on OPi PC Plus with defaults (after populating `/etc/systemd/system/haveged.service.d/10-no-new-privileges.conf` is works there too now).

ThomasKaiser, 2016-10-17:
What about putting these 2 lines in front of `dpkg-reconfigure openssh-server` call?

    dd if=/dev/random bs=1 count=4096 of=/tmp/random-data
    export RANDFILE=/tmp/random-data

If `haveged` is running (or otherwise system has enough entropy sources available) then execution should be pretty fast otherwise it will a) take ages (so we get angry reports by users and are aware that there's an issue) and b) enough entropy is available to generate the good keys.

ThomasKaiser, 2016-10-17:
Nah, doesn't work and `export SSH_USE_STRONG_RNG=1` doesn't either (for key generation). So we should provide a fix, log entropy pool size and document behaviour? So that users who care about this stuff should re-generate keys if needed while we're trying to provide at least a bit better entropy pool size when running `firstrun`

ThomasKaiser, 2016-10-17:
Another thought: pre-populating `output/cache/sdcard/var/lib/urandom/random-seed` on the build host. But this should never happen on Igor's host (or whoever builds official Armbian images). 

Hmm... bad idea, too dangerous.

ThomasKaiser, 2016-10-18:
Fixed with https://github.com/igorpecovnik/lib/commit/40d69b395eed7c52a5550d9bc942965c52b9442e -- TODO: document use and purpose of [`haveged`](http://www.issihosts.com/haveged/index.html) in an 'Armbian default security' document

closed 2016-10-18 by ThomasKaiser

ThomasKaiser, 2016-10-18:
Looks better now:

    ### [firstrun] Recreated SSH keys (entropy: 3580 3835)