armbian/build #504: [RfE] Improve boot times?

ThomasKaiser, 2016-10-19:
What about patching all board configs that use mainline u-boot with this?

    +CONFIG_ZERO_BOOTDELAY_CHECK=y
    +CONFIG_BOOTDELAY=0

igorpecovnik, 2016-10-19:
Hmm, for end users this is o.k. but I assume it's possible to get into u-boot prompt?

ThomasKaiser, 2016-10-19:
Sure, you just have to hit any key prior to the `Hit any key to stop autoboot` line being displayed:

    U-Boot 2016.09-armbian (Oct 19 2016 - 16:47:25 +0200) Allwinner Technology
    
    CPU:   Allwinner A10 (SUN4I)
    Model: Olimex A10-OLinuXino-LIME
    I2C:   ready
    DRAM:  512 MiB
    MMC:   SUNXI SD/MMC: 0
    *** Warning - bad CRC, using default environment
    
    In:    serial
    Out:   serial
    Err:   serial
    SCSI:  SATA link 0 timeout.
    AHCI 0001.0100 32 slots 1 ports 3 Gbps 0x1 impl SATA mode
    flags: ncq stag pm led clo only pmp pio slum part ccc apst 
    Net:   eth0: ethernet@01c0b000
    Hit any key to stop autoboot:  0 
    => boot

As far as I understood we don't get any more chances for devices being stuck on u-boot prompt since if something is interpreted as keyboard input then now it's just a smaller time frame.

ThomasKaiser, 2016-10-19:
On a related note: Adding `quiet` to kernel command line in boot script saves 0.8 seconds with A10 at `verbosity = 1` level (quite surprising). Thinking about an addition for `armhwinfo` comparing timestamps of `boot.cmd` and `boot.scr` and in case no higher verbosity is needed using a quiet `boot.scr` instead. But I will test this some time and create another issue in a few days/weeks.

(ideas taken from http://free-electrons.com/doc/training/boot-time/boot-time-slides.pdf)

igorpecovnik, 2016-10-19:
OK, lets do it. I'll also do some more test too see if I can found a problem.

ThomasKaiser, 2016-10-20:
Hmm... trying to address this 'the usual way' seems almost broken to me (since we have just another patch flying around that will not be updated the next time a board will be added and might break). What about a mechanism walking through `config/boards/`, checking `$LINUXFAMILY` there, sourcing `config/sources/$LINUXFAMILY.conf` and examining `$BOOTBRANCH`? To do a

    sed '$ a CONFIG_ZERO_BOOTDELAY_CHECK=y\nCONFIG_BOOTDELAY=0'

to the configs in question right before u-boot compilation starts? Or maybe better adding a new variable `$ZEROBOOTDELAY` to family configs in `config/sources`?

ThomasKaiser, 2016-10-20:
Tried to prepare this with https://github.com/igorpecovnik/lib/commit/93b8ba356c7b45099723868ef67b7d5207b5c469

In case you agree please add `BOOTDELAY=0` to family configs where appropriate (will be away from keyboard for a while)

ThomasKaiser, 2016-10-20:
Hmm... and replacing line 60 with

    [[ -n ${BOOTDELAY} ]] && sed -i "s/^CONFIG_BOOTDELAY=.*/CONFIG_BOOTDELAY=${BOOTDELAY}/" .config || echo "CONFIG_BOOTDELAY=${BOOTDELAY}" >> .config

would then also be a good idea?

igorpecovnik, 2016-10-20:
I changed line60 and add to AW common. I haven't got a chance to test yet.
https://github.com/igorpecovnik/lib/commit/3b3b62e2a8e3339a72e579f6dfccfcbdb25025e4

ThomasKaiser, 2016-10-25:
I close this now with the option to re-open it if we succeeded getting more testers and a more sane way to test stuff prior to release (then it's just adding `BOOTDELAY=0` to all family configs, waiting three weeks for feedback from testers and fixing stuff where necessary).

closed 2016-10-25 by ThomasKaiser