armbian/build #505: Provide ability to disable Wi-Fi power management?

ThomasKaiser, 2016-10-20:
According to google a `mkdir -p -m755 /etc/pm/power.d/` and then adding the following to `/etc/pm/power.d/wireless` should be sufficient:

    #!/bin/bash
    export PATH=/usr/sbin:/usr/bin:/sbin:/bin
    ListOfWiFis=$(/sbin/iwconfig 2>&1 | egrep -v "lo|tunl|eth|bond" | grep -v "^ " | awk -F" " '{print $1}')
    for i in ${ListOfWiFis} ; do
    	iwconfig $i power off
    done

(didn't work for me but I'm using `network-manager` now). With `network-manager` it's adding 

    #!/bin/bash
    case "$2" in
    	up)
    		/sbin/iwconfig $1 power off
    		;;
    esac

to `/etc/NetworkManager/dispatcher.d/99disable-power-management`. Unless execute bit is set this doesn't get executed (should also apply to the first approach above). Maybe including this file and adding to documentation that it's a simple `chmod 755` to get rid of Wi-Fi power management?

A `udev` rule like this

    KERNEL=="wlan*", ACTION=="add", RUN+="/sbin/iwconfig wlan0 power off"

might also be sufficient but interface name isn't always `wlan0`.

ThomasKaiser, 2016-10-21:
After tests with AP6210, RTL8189FTV and RTL8192CU I came to the conclusion that `iwconfig $NIC power [on|off]` seems to work universally so no need to fiddle around with module parameters and reboots (as it's done in Fourdee's `firstrun` contribution now). Therefore I vote for adding `$CACHEDIR/sdcard/etc/NetworkManager/dispatcher.d/99disable-power-management` with mode `644` and the following contents:

    #!/bin/bash
    case "$2" in
        up)
            /sbin/iwconfig $1 power off || true
            ;;
        down)
        	/sbin/iwconfig $1 power on || true
            ;;
    esac

We can then document this behaviour and H3 users can control it from `h3consumption` (just added with https://github.com/igorpecovnik/lib/commit/c653fabfe3452ef7af5d346c1b20c465aa6a1498). Where is the best place to add this file in installation/debootstrap process?

BTW: Don't care about the non-network-manager mode. People who want to configure Wi-Fi the hard way should also be able to figure out how to adjust power management settings on their own (eg. `pre-up /sbin/iwconfig wlan0 power off` line in interfaces file)

ThomasKaiser, 2016-10-23:
Closed since https://github.com/igorpecovnik/lib/commit/c3ee014ec86cf3aa705e4f56e14ff8310cf54351

zador-blood-stained, 2016-12-29:
@ThomasKaiser 

After doing some tests on Opi Zero with mainline kernel and xradio I've found a proper setting for the power management in NM:
```
root@orangepizero:~# cat /etc/NetworkManager/conf.d/default-wifi-powersave-on.conf
[connection]
wifi.powersave = 3
```

This is the global value, and according to `man nm-settings`, following values are possible:
```
0 - use global/default value (for the individual connections)
1 - ignore power management
2 - disable power management
3 - enable power management
```
and distribution default (at least in Xenial) is 3.

Should we also disable PM by default on Opi Zero?

zador-blood-stained, 2016-12-29:
This also means that "sleep" hacks in armhwinfo are no longer needed and h3consumption should be updated to use this setting.

ThomasKaiser, 2016-12-29:
I would go with `wifi.powersave = 2` by default on every device since the 'usual' complaints are caused by people getting sluggish SSH sessions when doing initial steps through WiFi. If people know what they're doing then they can activate it later by choice.

I've to admit that all my test with NM are solely based on Xenial (Version: `1.1.93-0ubuntu4`) and I've no idea how Jessie behaves here (Version: `0.9.10.0-7`). Am also away from build host for some time so cant't/won't look into this anytime soon.

But if it's supported in Jessie too the necessary changes look rather small?

zador-blood-stained, 2016-12-29:
> I've to admit that all my test with NM are solely based on Xenial (Version: 1.1.93-0ubuntu4) and I've no idea how Jessie behaves here (Version: 0.9.10.0-7). Am also away from build host for some time so cant't/won't look into this anytime soon.

I'll make a Jessie based image to test this (at least by looking at package file list it seems that this configuration override is not provided by default) and update the build scripts, after that you'll be able to reflect these changes in `h3consumption` or I may try to update it too later.

zador-blood-stained, 2016-12-29:
Hm. I can't get a wireless connection working with NM in Jessie and `wifi.powersave` is not supported there anyway, but it should be fixed at least in Xenial and in Stretch in the future.

ThomasKaiser, 2016-12-29:
> I can't get a wireless connection working with NM in Jessie

Hmm... no output from `nmcli dev wifi list` either?

zador-blood-stained, 2016-12-29:
> Hmm... no output from nmcli dev wifi list either?

Reflashed the image and tried again, seems to work now. Before it just didn't want to associate with the network with something like "Connection failed" (after entering the WPA2 password).

ThomasKaiser, 2016-12-29:
> Reflashed the image and tried again, seems to work now

O, how I hate these SD card or 'image burning' related problems. Wasted so much time with this already. Now curious whether there's a backports NM variant that could be used? Quick search was negative. 

So in case `wifi.powersave` is not supported in Jessie (due to outdated packages)... how to proceed then? Live with our hack now and switch with Stretch release?

zador-blood-stained, 2016-12-29:
> Now curious whether there's a backports NM variant that could be used? Quick search was negative.

Don't see any backports for the network-manager: https://packages.debian.org/jessie/network-manager

> So in case wifi.powersave is not supported in Jessie (due to outdated packages)... how to proceed then?

Since nothing seems to be touching the power management, I'll add `/etc/NetworkManager/dispatcher.d/99disable-power-management` to Jessie without the "sleep" hack (and with executable bit set by default) since it works for me. Now only h3consumption needs updates since PM logic now depends on OS release.

zador-blood-stained, 2017-01-06:
Should be fixed by f2cb36b

closed 2017-01-06 by zador-blood-stained