armbian/build #506: Olimex A20-OLinuXino-MICRO does not boot with Armbian_5.20_Micro_Debian_jessie_4.7.3

santiagorr, 2016-10-21:
But it does with 3.4.112 kernel using the same SD card.

This is the last part of the boot messages:

```
...
[    3.922698] console [ttyS0] disabled
[    3.949676] 1c28000.serial: ttyS0 at MMIO 0x1c28000 (irq = 46, base_baud = 1500000) is a U6_16550A
[    4.859211] console [ttyS0] enabled
[    4.893501] 1c29800.serial: ttyS1 at MMIO 0x1c29800 (irq = 47, base_baud = 1500000) is a U6_16550A
[    4.933338] 1c29c00.serial: ttyS2 at MMIO 0x1c29c00 (irq = 48, base_baud = 1500000) is a U6_16550A
[    4.954758] brd: module loaded
[    4.973848] loop: module loaded
[    4.988609] sun7i-dwmac 1c50000.ethernet: no regulator found
[    5.001221] sun7i-dwmac 1c50000.ethernet: no reset control found
[    5.013873]  Ring mode enabled
[    5.023416]  DMA HW capability register supported
[    5.028097]  Normal descriptors
[    5.053671] libphy: stmmac: probed
[    5.063510] eth0: PHY ID 00008201 at 1 IRQ POLL (stmmac-0:01) active
[    5.076641] ehci_hcd: USB 2.0 'Enhanced' Host Controller (EHCI) Driver
[    5.089559] ehci-platform: EHCI generic platform driver
[    5.101643] ohci_hcd: USB 1.1 'Open' Host Controller (OHCI) Driver
[    5.114259] ohci-platform: OHCI generic platform driver
[    5.126338] usbcore: registered new interface driver usb-storage
[    5.140061] sunxi-rtc 1c20d00.rtc: rtc core: registered rtc-sunxi as rtc0
[    5.153351] sunxi-rtc 1c20d00.rtc: RTC enabled
[    5.164283] i2c /dev entries driver
[    5.175519] axp20x-i2c 0-0034: AXP20x variant AXP209 found
[    5.201933] input: axp20x-pek as /devices/platform/soc@01c00000/1c2ac00.i2c/i2c-0/0-0034/axp20x-pek/input/input0
[    5.219120] axp20x-regulator axp20x-regulator: regulators node not found
[    5.239127] axp20x-i2c 0-0034: Backup (RTC) battery charging is disabled
[    5.252766] axp20x-i2c 0-0034: AXP20X driver loaded
[    5.268176] sunxi-wdt 1c20c90.watchdog: Watchdog enabled (timeout=16 sec, nowayout=0)
[    5.284680] sunxi-mmc 1c0f000.mmc: Got CD GPIO
[    5.328004] sunxi-mmc 1c0f000.mmc: base:0xf08fa000 irq:28
[    5.341366] sunxi-mmc 1c12000.mmc: Got CD GPIO
[    5.379503] mmc0: host does not support reading read-only switch, assuming write-enable
[    5.387966] sunxi-mmc 1c12000.mmc: base:0xf08fe000 irq:29
[    5.389027] ledtrig-cpu: registered to indicate activity on CPUs
```

Cheers!

ThomasKaiser, 2016-10-21:
Did you already try a different PSU and/or powering through the Mini USB port? I know, that sounds weird but I had similar symptoms with an A10-Lime days ago when I tried out new 'power cables' (USB type A to 5.5/2.1 barrel plug). With mainline kernel Lime always got stuck when loading USB drivers (which should most probably happen after the `ledtrig-cpu` message above). Throwing the cables away solved the problem.

santiagorr, 2016-10-21:
It was powering through the mini USB port. I will try another barrel plug in some days.