armbian/build #507: [pine64] armbian package update issue

kometchtech, 2016-10-23:
Hello,

In these environments, PINE 64 seems to be in error after trying to update + ARMBIAN package.

Pine64
legacy packages

```
kometch@dns02:~$ sudo apt update
[sudo] password for kometch: 
Hit:1 http://ports.ubuntu.com xenial InRelease                                                          
Hit:2 http://ppa.launchpad.net/cz.nic-labs/knot-dns/ubuntu xenial InRelease                             
Hit:3 http://ports.ubuntu.com xenial-security InRelease                                                 
Hit:4 http://ppa.launchpad.net/cz.nic-labs/knot-resolver/ubuntu xenial InRelease                        
Hit:5 http://ports.ubuntu.com xenial-updates InRelease                                                  
Processing triggers for systemd (229-4ubuntu11) ...
Processing triggers for ureadahead (0.100.0-19) ...
Processing triggers for man-db (2.7.5-1) ...
Processing triggers for initramfs-tools (0.122ubuntu8.5) ...
update-initramfs: Generating /boot/initrd.img-3.10.102-pine64
depmod: WARNING: could not open /var/tmp/mkinitramfs_oLAxIf/lib/modules/3.10.102-pine64/modules.order: No such file or directory
depmod: WARNING: could not open /var/tmp/mkinitramfs_oLAxIf/lib/modules/3.10.102-pine64/modules.builtin: No such file or directory
Setting up armbian-firmware (5.22) ...
Setting up armbian-tools-xenial (5.22) ...
Setting up hostapd (1:2.5~armbian5.22+1) ...
Setting up linux-firmware-image-pine64 (5.22) ...
Setting up linux-headers-pine64 (5.22) ...
Compiling headers - please wait ...
Setting up linux-image-pine64 (5.22) ...
update-initramfs: Generating /boot/initrd.img-3.10.104-pine64
Setting up linux-u-boot-pine64-default (5.22) ...
/var/lib/dpkg/info/linux-u-boot-pine64-default.postinst: line 5: conditional binary operator expected
/var/lib/dpkg/info/linux-u-boot-pine64-default.postinst: line 5: syntax error near `function'
/var/lib/dpkg/info/linux-u-boot-pine64-default.postinst: line 5: `[[  == function ]] && setup_write_uboot_platform'
dpkg: error processing package linux-u-boot-pine64-default (--configure):
 subprocess installed post-installation script returned error exit status 2
Setting up linux-xenial-root-pine64 (5.22) ...
Setting up sunxi-tools (1.3-1~armbian5.22+1) ...
Processing triggers for initramfs-tools (0.122ubuntu8.5) ...
update-initramfs: Generating /boot/initrd.img-3.10.104-pine64
Errors were encountered while processing:
 linux-u-boot-pine64-default
E: Sub-process /usr/bin/dpkg returned an error code (1)
```
Thanks.

zador-blood-stained, 2016-10-23:
This is a known issue and it will be fixed in a bugfix release (hopefully today or tomorrow)

ThomasKaiser, 2016-10-23:
Fixed with 5.23:

    Setting up linux-u-boot-pine64-default (5.23) ...
    Updating u-boot on device /dev/mmcblk0

closed 2016-10-23 by ThomasKaiser