armbian/build #513: Odroid XU4 running hotter on 3.10.104

alexmat, 2016-10-25:
Last kernel upgrade is making my xu4s run about 5 degrees hotter with the same workload. I ran powertop and didn't see any huge red flags. Any idea what might have caused it?

Also tried booting a fresh install, same issue, so it's not anything I'm running on the board.

ThomasKaiser, 2016-10-26:
How do you measure temperatures? 

In case you blindly trust in a readout some driver produces please keep the following in mind:

Thermal readouts sometimes depend on clocks and stuff like that. With Allwinner's H3 SoC (used by a lot of devices Armbian supports) switching between Allwinner's u-boot and mainline u-boot leads to differences in thermal readouts by 10-15°C.

Also sometimes the thermal sensor inside the SoC sits nearby specific engines. Again with H3 and an old version of Allwinner's BSP kernel SoC temperature decreased by 3°C 10 minutes after booting **if** no display is connected (since then display engine has been powered off).

In other words: Nothing to be concerned about if it's internal readouts. That's just a sensor somewhere producing values that are interpreted (more or less correctly) by software. Apart from that we can't do anything since we only use the kernel but don't contribute.

zador-blood-stained, 2016-10-26:
Apart from this Mali kernel driver was updated recently in Hardkernel repository, so difference in temperature may be caused by different GPU DVFS settings, but we can't do anything here since we rely on upstream kernel settings.

alexmat, 2016-10-26:
@ThomasKaiser I noticed the temperature increase because I use a fanless heatsink case and have two xu4s running the exact same config. The one I updated is hot to the touch, whereas the outdated one is lukewarm. The armbian login message temperature readout when I ssh in to the boxes shows a difference of about 5 degrees Celsius between the two, which feels about right in terms of touch. I'll see if I can pickup a heat gun to confirm the specifics, but I can tell there is quite a bit of difference by touch alone. Here is the case I use on both: http://ameridroid.com/products/xu4-aluminum-cooling-case-black

@zador-blood-stained Assuming it is the Mali driver, what's the best way to disable it? I don't need the GPU since these are both headless units that don't use the gpu in any way. Do I need to recompile the kernel? Also I noticed in the changelog that using a vanilla kernel with an XU4 is now possible, however I didn't see it on the download page.

Thanks for any suggestions. I'll contact upstream if there's no easy fix for this. I'd rather not debug GPU issues when I don't need the GPU to begin with.

golfromeo-fr, 2016-10-26:
@alexmat 
I have tested this image some hours ago
http://guillaume.romagny.free.fr/Armbian_5.23-GRO_Odroidxu4_Debian_jessie_4.8.4.7z

May I suggest something like https://www.amazon.com/Etekcity-Lasergrip-774-Non-contact-Thermometer/dp/B00837ZGRY/ to shape up the difference between hot & lukewarm ;)

alexmat, 2016-10-26:
@golfromeo-fr Is that image using legacy or vanilla kernel?

haha, thanks for the link. I have a heat gun I'll use if I have to, but in this situation the temperature difference is very obvious. The readings from the onboard sensor have been reasonably accurate in the past, so I have no reason to doubt them.

alexmat, 2016-10-26:
@golfromeo-fr ah it's 4.8.4 so it must be vanilla. Thanks for sharing, I'll try it out shortly.

zador-blood-stained, 2016-12-29:
Closing since most likely this is not related to Armbian specific settings

closed 2016-12-29 by zador-blood-stained