armbian/build #514: Lamobo R1 image 5.20 ships with broken swconfig

hknaack, 2016-10-25:
After failing to upgrade from 5.16 to 5.23 (missing depencency for swconfig?) and downgrading back, I used the available SD-card image [1], knowing that swconfig would be integrated there. Switch configuration didn't work, since that version of swconfig just reported "failed" when trying to set "reset 1" or "apply 1". swconfig from 5.23 [2] however had that problem fixed.
This should be added to the known problems section in the changelog of 5.20, maybe even the firmware image removed, as it almost bricks the device (just serial console available).

[1]http://image.armbian.com/Armbian_5.20_Lamobo-r1_Debian_jessie_4.7.3.7z
[2]http://apt.armbian.com/pool/utils/s/swconfig/swconfig_15.04-2~armbian5.23%2B1_armhf.deb

golfromeo-fr, 2016-10-25:
When I had trouble this was working.
http://guillaume.romagny.free.fr/swconfig_0.1.0_armhf.deb
or the binary
http://guillaume.romagny.free.fr/swconfig.zip

igorpecovnik, 2016-10-26:
It has been documented. That's all what can be done at urgent notice.

hknaack, 2016-10-26:
@igorpecovnik : to clarify my situation: when I upgraded from 5.16 to 5.23, swconfig did not get installed, which meant no more network/internet connection of the device (and thus no convenient installation of swconfig through aptitude). This means 5.23 should at least recommend to install swconfig, if not even depend on it. But I did not test this, instead tried to revert to 5.16 in the cache. For whatever reason (maybe because unstable wifi disconnected me), compilation took hours, so after one night, I rebooted and it bricked. => in changelog I would recommend a warning and point out to explicitly select swconfig during update
So I put on the 5.20 image, which comes with swconfig, which was unable to reset or apply. So, there was no ethernet connection. Using swconfig from the 5.23 package got it working. => in changelog it should be mentioned, that it doesn't work out of the box.
@golfromeo-fr I read about it in the forum, but (no offense) I always have a bad feeling trusting third party binaries.

igorpecovnik, 2016-10-26:
If I understand this properly - our dependencies are wrong, not the kernel driver?

hknaack, 2016-10-26:
Exactly. I'm currently on 5.20 and could not find any dependency in aptitude between swconfig and any lamobo r1 specific package.
swconfig shipped in 5.20 is buggy, but the 5.23 version is fine.

ThomasKaiser, 2016-10-26:
If I understand correctly these are **two** problems: `swconfig` was broken in 5.20 and now kernel is broken/incompatible in 5.23. That's why staying at 5.20 (kernel 4.7?) together with `swconfig` from 5.23 works.

hknaack, 2016-10-26:
I can not comment on kernel quality of 5.23, since I tried to revert to 5.16 as soon as I realized that swconfig was not installed (and no simple way to install it without ethernet).
I plan to upgrade to 5.23 again, but need to figure a good backup strategy first (at best without downtime).

zador-blood-stained, 2016-11-29:
This should not be an issue with future releases and further discussion can be done in #511

closed 2016-11-29 by zador-blood-stained