armbian/build #515: C2 - Disk resize on first boot - leaves unallocated

Code4Sale, 2016-10-26:
Probably not worth a mention, however, the disk resize on first boot leaves some space unallocated. 

Wish I had time to look into it (I need to setup a new build VM) - Just trying to help, I did not see anything in the list about it, but did notice possible changes made in this area? I am trying to slowly get involved here :)

16 GB Sandisk Ultra Card 151.9 MB unallocated

64 GB EMMC "Odroid Black" (Samsung?) 596.4 MB unallocated

Using:

http://image.armbian.com/betaimages/Armbian_5.23.161025_Odroidc2_Ubuntu_xenial_3.14.79_desktop.7z

Joe

igorpecovnik, 2016-10-26:
It's a (hidden) feature, not a bug.
https://github.com/igorpecovnik/lib/blob/master/scripts/firstrun#L168-L172

Code4Sale, 2016-10-26:
Good explanation! Looks like we are both paying attention.

I see it was labeled as a question, so good! Looks like it may become a unhidden documented feature, meaning fewer folks posting questions about it.

Sorry it was lame of an issue - it was the only issue I have seen so far. I need to download the code.

ThomasKaiser, 2016-10-26:
The problem is that 2 different xy GB SD cards are not exactly xy GB in size but always differ by a few sectors. We want to prevent the situation that people can not clone their installations to different media 'of the same size' or restore a device image they made before on another card.

That's the whole reason a specific percentage won't be used since these size differences then don't matter any more. If we would document that we would deal with one issue per week to 'reclaim disk space wasted by Armbian'.

closed 2016-10-26 by ThomasKaiser