armbian/build #521: <built-in>: internal compiler error: in layout_type, at stor-layout.c:2111

noloader, 2016-10-28:
I'm working on an ODROID C2. It took 7 package updates on OCT 27, 2016. It looks like one of them broke the C++ compiler.

```
$ uname -a
Linux odroidc2 3.14.79-odroidc2 #5 SMP PREEMPT Sun Oct 23 16:52:51 CEST 2016 aarch64 GNU/Linux
```

And:

```
$ cat test.cxx
#include <string>
int main(int argc, char* argv[])
{
  return 0;
}
```

And:

```
$ g++ -march=armv8-a+crc test.cxx -o test.exe
<built-in>: internal compiler error: in layout_type, at stor-layout.c:2111
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-4.9/README.Bugs> for instructions.
Preprocessed source stored into /tmp/cckroNqQ.out file, please attach this to your bugreport.
```

And:

```
$ cat /proc/cpuinfo 
processor	: 0
BogoMIPS	: 48.00
Features	: fp asimd evtstrm crc32
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
BogoMIPS	: 48.00
Features	: fp asimd evtstrm crc32
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
BogoMIPS	: 48.00
Features	: fp asimd evtstrm crc32
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
BogoMIPS	: 48.00
Features	: fp asimd evtstrm crc32
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

Hardware	: ODROID-C2
Revision	: 020c
```

-----

File `cckroNqQ.out` zipped and attached. Its a plain text file.

[cckroNqQ.out.zip](https://github.com/igorpecovnik/lib/files/557621/cckroNqQ.out.zip)

zador-blood-stained, 2016-10-28:
> I'm working on an ODROID C2. It took 7 package updates on OCT 27, 2016. It looks like one of them broke the C++ compiler. Prior to the updates, -march=armv8-a+crc was consumed without troubles.

Can you please check what packages were updated? You can find latest package related events in `/var/log/dpkg.log`

noloader, 2016-10-28:
> Can you please check what packages were updated?

Yes, absolutely. The log below was trimmed. The attached zip is the full log.

```
$ sudo cat /var/log/dpkg.log
[sudo] password for jwalton: 
...
2016-10-27 21:52:00 startup archives unpack
2016-10-27 21:52:03 upgrade tzdata:all 2016f-0+deb8u1 2016h-0+deb8u1
2016-10-27 21:52:03 status half-configured tzdata:all 2016f-0+deb8u1
2016-10-27 21:52:03 status unpacked tzdata:all 2016f-0+deb8u1
2016-10-27 21:52:03 status half-installed tzdata:all 2016f-0+deb8u1
2016-10-27 21:52:10 status half-installed tzdata:all 2016f-0+deb8u1
2016-10-27 21:52:10 status unpacked tzdata:all 2016h-0+deb8u1
2016-10-27 21:52:11 status unpacked tzdata:all 2016h-0+deb8u1
2016-10-27 21:52:11 startup packages configure
2016-10-27 21:52:11 configure tzdata:all 2016h-0+deb8u1 <none>
2016-10-27 21:52:11 status unpacked tzdata:all 2016h-0+deb8u1
2016-10-27 21:52:11 status half-configured tzdata:all 2016h-0+deb8u1
2016-10-27 21:52:20 status installed tzdata:all 2016h-0+deb8u1
2016-10-27 21:52:20 startup archives unpack
2016-10-27 21:52:20 upgrade armbian-firmware:arm64 5.20 5.22
2016-10-27 21:52:20 status half-configured armbian-firmware:arm64 5.20
2016-10-27 21:52:20 status unpacked armbian-firmware:arm64 5.20
2016-10-27 21:52:20 status half-installed armbian-firmware:arm64 5.20
2016-10-27 21:52:21 status half-installed armbian-firmware:arm64 5.20
2016-10-27 21:52:22 status unpacked armbian-firmware:arm64 5.22
2016-10-27 21:52:22 status unpacked armbian-firmware:arm64 5.22
2016-10-27 21:52:22 upgrade linux-dtb-odroidc2:arm64 5.20 5.23
2016-10-27 21:52:22 status half-configured linux-dtb-odroidc2:arm64 5.20
2016-10-27 21:52:22 status unpacked linux-dtb-odroidc2:arm64 5.20
2016-10-27 21:52:22 status half-installed linux-dtb-odroidc2:arm64 5.20
2016-10-27 21:52:22 status half-installed linux-dtb-odroidc2:arm64 5.20
2016-10-27 21:52:22 status unpacked linux-dtb-odroidc2:arm64 5.23
2016-10-27 21:52:22 status unpacked linux-dtb-odroidc2:arm64 5.23
2016-10-27 21:52:24 upgrade linux-headers-odroidc2:arm64 5.20 5.23
2016-10-27 21:52:24 status half-configured linux-headers-odroidc2:arm64 5.20
2016-10-27 21:52:24 status unpacked linux-headers-odroidc2:arm64 5.20
2016-10-27 21:52:24 status half-installed linux-headers-odroidc2:arm64 5.20
2016-10-27 22:01:15 status half-installed linux-headers-odroidc2:arm64 5.20
2016-10-27 22:01:16 status unpacked linux-headers-odroidc2:arm64 5.23
2016-10-27 22:01:20 status unpacked linux-headers-odroidc2:arm64 5.23
2016-10-27 22:01:21 upgrade linux-image-odroidc2:arm64 5.20 5.23
2016-10-27 22:01:21 status half-configured linux-image-odroidc2:arm64 5.20
2016-10-27 22:01:21 status unpacked linux-image-odroidc2:arm64 5.20
2016-10-27 22:01:21 status half-installed linux-image-odroidc2:arm64 5.20
2016-10-27 22:02:37 status half-installed linux-image-odroidc2:arm64 5.20
2016-10-27 22:02:38 status unpacked linux-image-odroidc2:arm64 5.23
2016-10-27 22:02:38 status unpacked linux-image-odroidc2:arm64 5.23
2016-10-27 22:02:38 upgrade linux-jessie-root-odroidc2:arm64 5.20 5.23
2016-10-27 22:02:38 status triggers-pending initramfs-tools:all 0.120+deb8u2
2016-10-27 22:02:38 status triggers-awaited linux-jessie-root-odroidc2:arm64 5.20
2016-10-27 22:02:38 status half-configured linux-jessie-root-odroidc2:arm64 5.20
2016-10-27 22:02:38 status unpacked linux-jessie-root-odroidc2:arm64 5.20
2016-10-27 22:02:38 status half-installed linux-jessie-root-odroidc2:arm64 5.20
2016-10-27 22:02:38 status triggers-pending systemd:arm64 215-17+deb8u5
2016-10-27 22:02:38 status half-installed linux-jessie-root-odroidc2:arm64 5.20
2016-10-27 22:02:38 status unpacked linux-jessie-root-odroidc2:arm64 5.23
2016-10-27 22:02:38 status unpacked linux-jessie-root-odroidc2:arm64 5.23
2016-10-27 22:02:38 upgrade linux-u-boot-odroidc2-default:arm64 5.20 5.23
2016-10-27 22:02:38 status half-configured linux-u-boot-odroidc2-default:arm64 5.20
2016-10-27 22:02:38 status unpacked linux-u-boot-odroidc2-default:arm64 5.20
2016-10-27 22:02:39 status half-installed linux-u-boot-odroidc2-default:arm64 5.20
2016-10-27 22:02:40 status half-installed linux-u-boot-odroidc2-default:arm64 5.20
2016-10-27 22:02:40 status unpacked linux-u-boot-odroidc2-default:arm64 5.23
2016-10-27 22:02:40 status unpacked linux-u-boot-odroidc2-default:arm64 5.23
2016-10-27 22:02:40 trigproc initramfs-tools:all 0.120+deb8u2 <none>
2016-10-27 22:02:40 status half-configured initramfs-tools:all 0.120+deb8u2
2016-10-27 22:02:58 status installed initramfs-tools:all 0.120+deb8u2
[... 17 lines cut ...]
2016-10-27 22:03:41 status unpacked linux-image-odroidc2:arm64 5.23
2016-10-27 22:03:41 status half-configured linux-image-odroidc2:arm64 5.23
2016-10-27 22:03:49 status installed linux-image-odroidc2:arm64 5.23
2016-10-27 22:03:49 configure linux-jessie-root-odroidc2:arm64 5.23 <none>
2016-10-27 22:03:49 status triggers-pending initramfs-tools:all 0.120+deb8u2
2016-10-27 22:03:49 status unpacked linux-jessie-root-odroidc2:arm64 5.23
2016-10-27 22:03:49 status unpacked linux-jessie-root-odroidc2:arm64 5.23
2016-10-27 22:03:49 status half-configured linux-jessie-root-odroidc2:arm64 5.23
2016-10-27 22:03:50 status triggers-awaited linux-jessie-root-odroidc2:arm64 5.23
2016-10-27 22:03:50 configure linux-u-boot-odroidc2-default:arm64 5.23 <none>
2016-10-27 22:03:50 status unpacked linux-u-boot-odroidc2-default:arm64 5.23
2016-10-27 22:03:50 status half-configured linux-u-boot-odroidc2-default:arm64 5.23
[... cut ...] t:arm64 5.23
2016-10-27 22:03:50 trigproc initramfs-tools:all 0.120+deb8u2 <none>
2016-10-27 22:03:50 status half-configured initramfs-tools:all 0.120+deb8u2
2016-10-27 22:03:50 status installed linux-jessie-root-odroidc2:arm64 5.23
2016-10-27 22:04:00 status installed initramfs-tools:all 0.120+deb8u2
2016-10-27 22:04:01 startup packages configure
```

-----

File `dpkg.log` zipped and attached. Its a plain text file.

[dpkg.log.zip](https://github.com/igorpecovnik/lib/files/558952/dpkg.log.zip)

noloader, 2016-10-28:
The [interesting?] thing about this issue is:

* HiKey - ARMv8, Aarch64, CRC+Crypto - OK
* Pine64 - ARMv8, Aarch64, CRC+Crypto - OK
* ODROID C2 - ARMv8, Aarch64, CRC only - bad
* Raspberry PI3 - ARMv8, ARM-32 (not even Aarch32), CRC only - OK

Whatever has happened, it seems that it only affects ODROID C2 at the moment.

I alos

zador-blood-stained, 2016-10-30:
I just built a fresh image for Odroid C2 and I can't reproduce this error. 
```
root@odroidc2:~# g++ -dumpversion
4.9.2
root@odroidc2:~# cat test.cxx
#include <string>
int main(int argc, char* argv[])
{
  return 0;
}

root@odroidc2:~# g++ test.cxx
root@odroidc2:~# g++ -march=armv8-a+crc test.cxx
root@odroidc2:~# ./a.out
root@odroidc2:~# uname -r
3.14.79-odroidc2
root@odroidc2:~# cat /proc/cpuinfo
processor       : 0
BogoMIPS        : 48.00
Features        : fp asimd evtstrm crc32
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 1
BogoMIPS        : 48.00
Features        : fp asimd evtstrm crc32
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 2
BogoMIPS        : 48.00
Features        : fp asimd evtstrm crc32
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 3
BogoMIPS        : 48.00
Features        : fp asimd evtstrm crc32
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

Hardware        : ODROID-C2
Revision        : 020b
root@odroidc2:~#
```
Do you use SD or eMMC for your system? Can you please install `debsums` and run `debsums -sc` (as root) to check for corruption of system files?

noloader, 2016-10-31:
@zador-blood-stained,

Yeah, that could be it... I just brought a RPI3 back to life by tossing the SDcard. Let me try it and see what happens. I'll need a couple of days.

noloader, 2016-11-02:
Uninstalling and reinstalling `build-essential` did the trick. Thanks for the suggestions. Sorry about the report.

closed 2016-11-02 by noloader