armbian/build #522: [RfD] Doing a forced rmmod on shutdown?

ThomasKaiser, 2016-10-28:
NanoPi Air idling with active Wi-Fi (3.4.112) consumed 670 mW on average, after `poweroff` the board still used 280 mW. Now I added this to `armhwinfo`'s stop case:

	for module in $(cut -f1 -d' ' </proc/modules) ; do rmmod -f ${module} ; done
	echo -e "\n\nshutdown at $(date)" >>/var/log/shutdown.log
	lsmod >>/var/log/shutdown.log
	dmesg | tail -n 100 >>/var/log/shutdown.log
	sync

All modules (bmp085, pcf8591, g_serial, dhd, hidp, rfcomm, hci_uart, bluetooth) get unloaded now, whether poweroff consumption will also decrease I know in an hour (30m average consumption values). The last time I asked about a brute-force `rmmod -f` approach on shutdown Zador was against so I ask now whether it would be useful to do the following

- add the `for module in $(cut -f1 -d' ' </proc/modules) ; do rmmod -f ${module} ; done` line to `armhwinfo` by default
- based on `[[ -n $(cut -f1 -d' ' </proc/modules) ]]` 0.5 seconds later adding properly formatted 100 last lines from `dmesg` output to `/var/log/armhwinfo.log`

Idea behind: Minimizing kernel crashes on shutdown where possible and logging where problems occur. Since stuff gets into `/var/log/armhwinfo.log` there's log rotation active and people can easily submit stuff using `armbianmonitor -u`. Without that I find it hard to be able to even start to understand why and when these shutdown problems occur.

zador-blood-stained, 2016-10-28:
Instead of trying to remove every loaded module, we may use a list of modules (per-board) to be removed (that were also tested and confirmed to work), and end users will be able to add/delete entries in this list on the board.

ThomasKaiser, 2016-10-28:
> whether poweroff consumption will also decrease I know in an hour

Nope, didn't help, green led remained active and "poweroff" consumption was at 360 mW. Log showed this:

    shutdown at Fri Oct 28 07:22:46 UTC 2016
    Module                  Size  Used by
    [    7.791234] systemd[1]: Starting Load/Save Random Seed...
    [    7.890628] systemd[1]: Activating swap /var/swap...
    [    8.088102] Adding 131068k swap on /var/swap.  Priority:-1 extents:2 across:139260k SS
    [    8.101145] systemd[1]: Starting Create Static Device Nodes in /dev...
    [    8.138969] systemd[1]: Mounted FUSE Control File System.
    [    8.163830] systemd[1]: Activated swap /var/swap.
    [    8.430267] systemd[1]: Started Apply Kernel Variables.
    [    8.900431] systemd[1]: Started Load/Save Random Seed.
    [    9.070336] systemd[1]: Started Create Static Device Nodes in /dev.
    [    9.250252] systemd[1]: Started udev Coldplug all Devices.
    [    9.341844] systemd[1]: Reached target Local File Systems (Pre).
    [    9.460613] systemd[1]: Mounting /tmp...
    [    9.641357] systemd[1]: Starting udev Kernel Device Manager...
    [    9.660477] systemd[1]: Reached target Swap.
    [    9.741590] systemd[1]: Mounted /tmp.
    [    9.811225] systemd[1]: Started Entropy daemon using the HAVEGE algorithm.
    [    9.941423] systemd[1]: Starting Journal Service...
    [    9.960662] systemd[1]: Reached target Local File Systems.
    [   10.061904] systemd[1]: Starting Set console font and keymap...
    [   10.241146] systemd[1]: Starting Raise network interfaces...
    [   10.311658] systemd[1]: Started udev Kernel Device Manager.
    [   10.332032] systemd[1]: Started Journal Service.
    [   10.782536] twi_start()434 - [i2c0] START can't sendout!
    [   10.782834] twi_start()434 - [i2c0] START can't sendout!
    [   10.783131] twi_start()434 - [i2c0] START can't sendout!
    [   10.783431] twi_start()434 - [i2c0] START can't sendout!
    [   10.783727] twi_start()434 - [i2c0] START can't sendout!
    [   10.784025] twi_start()434 - [i2c0] START can't sendout!
    [   10.786735] twi_start()434 - [i2c0] START can't sendout!
    [   10.787038] twi_start()434 - [i2c0] START can't sendout!
    [   10.787330] twi_start()434 - [i2c0] START can't sendout!
    [   10.787488] bmp085: probe of 0-0077 failed with error -121
    [   10.920301] systemd-journald[218]: Received request to flush runtime journal from PID 1
    [   11.049925] vmouse_input_dev_open 
    [   11.110478] vmouse_input_dev_close 
    [   12.247101] [rfkill]: rfkill set power 1
    [   17.943523] dhd_open: Enter d4420800
    [   18.024874] dhd_open: Exit ret=0
    [   20.030731] Connectting with bc:05:43:be:c1:e7 channel (13) ssid "Snort-Honeynet", len (14)
    
    [   20.101785] wl_iw_event: Link UP with BSSID=BC:05:00:00:C1:E7
    [   20.101828] wl_bss_connect_done succeeded with bc:05:43:be:c1:e7
    [   20.170712] wl_bss_connect_done succeeded with bc:05:43:be:c1:e7
    [   21.185286] g_serial gadget: high-speed config #2: CDC ACM config
    [ 1363.015669] CFG80211-ERROR) wl_is_linkdown : Link down Reason : WLC_E_DEAUTH_IND
    [ 1363.015763] link down if wlan0 may call cfg80211_disconnected. event : 6, reason=6 from bc:05:43:be:c1:e7
    [ 1363.020539] wl_iw_event: Link Down with BSSID=BC:05:00:00:C1:E7
    [ 1363.020602] CFG80211-ERROR) wl_is_linkdown : Link down Reason : WLC_E_DEAUTH
    [ 1363.020669] CFG80211-ERROR) wl_is_linkdown : Link down Reason : WLC_E_LINK
    [ 1363.027387] cfg80211: Calling CRDA for country: DE
    [ 1363.056821] Connectting with bc:05:43:be:c1:e7 channel (13) ssid "Snort-Honeynet", len (14)
    
    [ 1363.070427] wl_iw_event: Link UP with BSSID=BC:05:00:00:C1:E7
    [ 1363.070487] wl_bss_connect_done succeeded with bc:05:43:be:c1:e7
    [ 1363.076920] cfg80211: Regulatory domain changed to country: DE
    [ 1363.076948] cfg80211:  DFS Master region ETSI
    [ 1363.076959] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp)
    [ 1363.076975] cfg80211:   (2400000 KHz - 2483500 KHz @ 40000 KHz), (N/A, 2000 mBm)
    [ 1363.076987] cfg80211:   (5150000 KHz - 5250000 KHz @ 80000 KHz), (N/A, 2000 mBm)
    [ 1363.076999] cfg80211:   (5250000 KHz - 5350000 KHz @ 80000 KHz), (N/A, 2000 mBm)
    [ 1363.077011] cfg80211:   (5470000 KHz - 5725000 KHz @ 160000 KHz), (N/A, 2698 mBm)
    [ 1363.077023] cfg80211:   (57000000 KHz - 66000000 KHz @ 2160000 KHz), (N/A, 4000 mBm)
    [ 1363.077136] cfg80211: Calling CRDA to update world regulatory domain
    [ 1363.098932] cfg80211: World regulatory domain updated:
[... 21 lines cut ...]
    [ 2552.811014] dhd_dpc_thread: Unexpected up_cnt 0
    [ 2552.811130] dhd_detach(): thread:dhd_dpc:b7 terminated OK
    [ 2552.811547] wl_event_handler: was terminated
    [ 2552.811693] wl_destroy_event_handler(): thread:wl_event_handler:b5 terminated OK
    [ 2552.911475] dhd_tcpack_suppress_set 376: already set to 0
    [ 2552.911765] dhd_os_prealloc: failed to alloc memory, section: 7, size: 0bytes
    [ 2552.912101] unregister wifi platform drivers
    [ 2552.912118] wifi_platform_set_power = 0
    [ 2552.912127] ======== PULL WL_REG_ON LOW! ========
    [ 2552.912137] wifi_platform_bus_enumerate device present 0
    [ 2552.912145] ======== Card detection to remove SDIO card! ========
    [ 2552.912199] dhd_module_cleanup: Exit
    [ 2552.912842] mmc1: card 0001 removed
    [ 2552.913174] [mmc]: sdc1 power_supply is null
[... cut ...]
Anyway: adding the following to `armhwinfo` seems like a good idea to me since we get some shutdown logging also in `armhwinfo.log`:

	for module in $(cut -f1 -d' ' </proc/modules) ; do rmmod -f ${module} ; done
	echo -e "\n### shutdown:\n\n$(dmesg | tail -n 100)" >>/var/log/armhwinfo.log
	sync

(maybe replacing this with a list of modules as suggested by Zador but I fail to understand why a forced `rmmod` shouldn't work and fear maintaining such a list will take ages especially since we have 40 boards + 2 kernels each and don't know which modules might cause trouble)

ThomasKaiser, 2016-10-28:
Hmm... by looking at the above it seems like a good idea to modify `armbianmonitor` in a way that `-u` switch clears out all IP and MAC addresses and also SSID/BSSID occurences while `-U` provides them unaltered. No time for that now or within the next week(s) though...

Shouldn't be that hard since it's just extending [the regex that is already in place](https://github.com/igorpecovnik/lib/blob/master/scripts/armbianmonitor/armbianmonitor#L164) to obfuscate IP addresses somewhat.

zador-blood-stained, 2016-10-28:
```
       -f, --force
           This option can be extremely dangerous: it has no effect unless
           CONFIG_MODULE_FORCE_UNLOAD was set when the kernel was compiled.
           With this option, you can remove modules which are being used, or
           which are not designed to be removed, or have been marked as unsafe
           (see lsmod(8)).
```
```
config MODULE_FORCE_UNLOAD
	bool "Forced module unloading"
	depends on MODULE_UNLOAD
	help
	  This option allows you to force a module to unload, even if the
	  kernel believes it is unsafe: the kernel will remove the module
	  without waiting for anyone to stop using it (using the -f option to
	  rmmod).  This is mainly for kernel developers and desperate users.
	  If unsure, say N.
```
Do you really want to try to remove **every** kernel module before they have a chance of unloading properly?

ThomasKaiser, 2016-10-28:
> Do you really want to try to remove every kernel module before they have a chance of unloading properly?

No, you're right. Currently it would only affect two families (BTW: renaming `pine64` to `sun50i` might be necessary soon?) so let's better remove the `-f` switch and fix both kernel configs below (since options is most probably there by accident)?

    linux-cubox-default.config:# CONFIG_MODULE_FORCE_UNLOAD is not set
    linux-cubox-dev.config:# CONFIG_MODULE_FORCE_UNLOAD is not set
    linux-cubox-next.config:# CONFIG_MODULE_FORCE_UNLOAD is not set
    linux-marvell-default.config:# CONFIG_MODULE_FORCE_UNLOAD is not set
    linux-marvell-dev.config:# CONFIG_MODULE_FORCE_UNLOAD is not set
    linux-marvell-next.config:# CONFIG_MODULE_FORCE_UNLOAD is not set
    linux-odroidc1-default.config:# CONFIG_MODULE_FORCE_UNLOAD is not set
    linux-odroidc2-default.config:# CONFIG_MODULE_FORCE_UNLOAD is not set
    linux-odroidxu4-default.config:CONFIG_MODULE_FORCE_UNLOAD=y
    linux-odroidxu4-next.config:# CONFIG_MODULE_FORCE_UNLOAD is not set
    linux-pine64-default.config:CONFIG_MODULE_FORCE_UNLOAD=y
    linux-pine64-dev.config:# CONFIG_MODULE_FORCE_UNLOAD is not set
    linux-s500-default.config:# CONFIG_MODULE_FORCE_UNLOAD is not set
    linux-sun4i-default.config:# CONFIG_MODULE_FORCE_UNLOAD is not set
    linux-sun5i-default.config:# CONFIG_MODULE_FORCE_UNLOAD is not set
    linux-sun7i-default.config:# CONFIG_MODULE_FORCE_UNLOAD is not set
    linux-sun8i-default.config:# CONFIG_MODULE_FORCE_UNLOAD is not set
    linux-sun8i-dev.config:# CONFIG_MODULE_FORCE_UNLOAD is not set
    linux-sunxi-dev.config:# CONFIG_MODULE_FORCE_UNLOAD is not set
    linux-sunxi-next.config:# CONFIG_MODULE_FORCE_UNLOAD is not set
    linux-toradex-default.config:# CONFIG_MODULE_FORCE_UNLOAD is not set
    linux-udoo-default.config:# CONFIG_MODULE_FORCE_UNLOAD is not set
    linux-udoo-neo-default.config:# CONFIG_MODULE_FORCE_UNLOAD is not set
    linux-udoo-next.config:# CONFIG_MODULE_FORCE_UNLOAD is not set

zador-blood-stained, 2016-10-28:
> No, you're right.

Then maybe use `modprobe -r` instead of plain rmmod too?
```

       rmmod is a trivial program to remove a module (when module unloading
       support is provided) from the kernel. Most users will want to use
       modprobe(8) with the -r option instead.
```

> BTW: renaming pine64 to sun50i might be necessary soon?

I thought about this, currently sun50i is different from other sunxi families - different u-boot file list and installation procedure, different boot script, different u-boot source, different architecture, ... so it won't include `sunxi_common.inc` but will be totally separate config for now, and maybe split into `sunxi_common64.inc` if Allwinner makes another 64bit SoC with ~~crappy~~ similar kernel and u-boot situation.

zador-blood-stained, 2016-10-28:
> I also had to realize that the USB device created by g_serial is still present after a shutdown -h now and only disconnecting USB cable from my Mac shows then the message Cannot open /dev/cu.usbmodem1411! (so maybe this is an issue caused by ARISC?).

`echo 0 > /sys/bus/platform/devices/sunxi_usb_udc/otg_role` on shutdown may fix this issue, I noticed this too while testing FEL mass storage

ThomasKaiser, 2016-10-28:
Ok, let's postpone everything related to Allwinner's 64-bit experiments (I thought about the other 3 A64 devices that are to be released soon since they'll share everything except DT bits) at least until we get an idea how H5 'BSP situattion' looks like ;)

Tried to address the other issues with https://github.com/igorpecovnik/lib/commit/7d54789a9e62e2e5463d4864ea8b1d63bfc342cc

closed 2016-10-28 by ThomasKaiser