armbian/build #524: U-Boot fdtfile is wrongly set to bad name on few boards

martinayotte, 2016-10-28:
With Armbian 5.24 with OrangePi-PC+, the fdtfile is set to sun8i-h3-orangepi-pc.dtb, instead of sun8i-h3-orangepi-pc-plus.dtb

With OrangePiPlus2E, it is set to sun8i-h3-orangepi-plus.dtb instead sun8i-h3-orangepi-plus2e.dtb

It is a bit annoying when doing DT changes and that we discovered that those changes were done in the wrong file. :-(

zador-blood-stained, 2016-10-28:
It depends on target kernel version. Currently sun8i-dev still points to 4.8 branch, but `sun8i-h3-orangepi-pc-plus.dts` and `sun8i-h3-orangepi-plus2e.dts` were added in only 4.9 if I'm reading git logs correctly.

martinayotte, 2016-10-28:
Ah ! Ok !
Since the 4.8.x seems to be completely broken on H3, at least for me, I've switched to 4.9.x ...
Like discussed in others threads, are we planning to switch and forget about getting 4.8.x working ?

zador-blood-stained, 2016-10-28:
Both 4.9 branches in https://github.com/megous/linux/ are broken for me too, but other people had success with https://github.com/montjoie/linux/tree/sun8i-emac-wip-v5 branch, which I wasn't able to test due to some cryptic hardware issues which I still have to investigate.

zador-blood-stained, 2016-10-28:
If https://github.com/montjoie/linux/tree/sun8i-emac-wip-v5 works on most supported boards, then we should switch to it since working branch without THS and some other stuff is still better that feature-complete, but not working.

martinayotte, 2016-10-28:
The https://github.com/montjoie/linux/tree/sun8i-emac-wip-v5 works for me on PC+ and Plus2E, I'm currently building also for PC/One/Lite and keep you posted.

martinayotte, 2016-10-28:
All my H3 boards are now on 4.9.0

zador-blood-stained, 2016-10-31:
I removed the patch that changed `fdtfile` on PC Plus and Plus 2E (80d1dec), so this can be considered "fixed".

closed 2016-10-31 by zador-blood-stained