armbian/build #526: Possible: 5.23 causes kernel exception (MMC), corrupts SD card on OPi One

Fourdee, 2016-10-29:
Hi guys,

I'am not 100% on this yet, i've tested with 2 PSU's and 3 different SD cards. But, my board could be faulty. So, need someone to confirm on their board.

#### Test to replicate:
- Build ARMbian image 5.23: OPi One, Jessie, Server image
- Write to SD card and power on.
- Wait 1-3 minutes
- Kernel exception with MMC errors reported on screen, system attempts to reboot, then fails to boot 

I think this may be related to ```linux-image-sun8i (5.23)```. Our DietPi image has ```linux-image-sun8i (5.21)```. After running ```apt-get upgrade``` which installs ```linux-image-sun8i (5.23)```, reboot, then able to replicate same results in above test.

igorpecovnik, 2016-10-29:
19:14:42 up 31 min,  1 user,  load average: 0.00, 0.01, 0.02
This is kernel and u-boot. Try.
http://image.armbian.com/test/debs524.zip

Fourdee, 2016-10-29:
@igorpecovnik 

Thanks 👍 , i'll try the 5.24 debs tomorrow. 
Did some further testing today, inconsistent results, but always occurs, so most likely board failure or both PSU's are faulty.

zador-blood-stained, 2016-10-31:
Tested `linux-image-sun8i` from the repository (5.23) on Orange Pi One, no such problems detected.
```
root@orangepione:~# uptime
 18:19:38 up 10 min,  1 user,  load average: 0.00, 0.02, 0.05
root@orangepione:~# uname -a
Linux orangepione 3.4.112-sun8i #10 SMP PREEMPT Sun Oct 23 16:06:55 CEST 2016 armv7l GNU/Linux
root@orangepione:~# dmesg | tail
[    7.802031] bmp085: probe of 0-0077 failed with error -70
[    8.254617] EXT4-fs (mmcblk0p1): re-mounted. Opts: commit=600,errors=remount-ro
[    8.520048] Adding 131068k swap on /var/swap.  Priority:-1 extents:1 across:131068k SS
[    9.937647] systemd-journald[179]: Received request to flush runtime journal from PID 1
[   10.353288] gmac0: probed
[   10.353650] gmac0 gmac0: eth0: eth0: PHY ID 00441400 at 0 IRQ poll (gmac0-0:00)
[   11.925865] Registered IR keymap rc-empty
[   11.926604] rc0: sunxi-ir as /devices/virtual/rc/rc0
[   11.927764] rc rc0: lirc_dev: driver ir-lirc-codec (sunxi-ir) registered at minor = 0
[   13.350255] PHY: gmac0-0:00 - Link is Up - 100/Full
root@orangepione:~#
```

Fourdee, 2016-10-31:
@igorpecovnik @zador-blood-stained 
Thanks guys, appreciate it. 👍 
Yep, this appears to be a board failure, tried multiple PSU's. Not the 1st time a OrangePi has failed on me 😞 

Please close ticket.

closed 2016-10-31 by zador-blood-stained