armbian/build #528: Banana Pi M2+ Armbian desktop compression error

SkutchElectronics, 2016-10-31:
I downloaded this version of the legacy OS and tried to uncompress it with 7Zip to no avail.  It reports "Unsupported Compression Method" for all of the files.  Even trying to just uncompress the img file, it reports that for sha256sum and the img file.  sha256sum shows a size of 128 and a packed size of 682 750 239 and a CRC of 078F0A77.

igorpecovnik, 2016-11-01:
```
sha256sum.sha:
OK:  Armbian_5.20_Bananapim2plus_Debian_jessie_3.4.112_desktop.img

Errors: 0
OK: 1, not found: 0, read error: 0, wrong checksum: 0
```
I made a download and a test ->everything looks o.k. It must be some problem on your side, corrupted download, bad media, ...?

zador-blood-stained, 2016-11-01:
@SkutchElectronics 
What version of 7-zip do you have?

SkutchElectronics, 2016-11-01:
That was it, thanks, had a really old version.

closed 2016-11-01 by zador-blood-stained

ThomasKaiser, 2016-11-01:
Might be interesting to check .xz instead since if we find compatible settings with Etcher only recommendation could be: "Download image, open Etcher and burn"

zador-blood-stained, 2016-11-02:
> Might be interesting to check .xz instead since if we find compatible settings with Etcher only recommendation could be: "Download image, open Etcher and burn"

If I understand this correctly, you can write compressed files with Etcher and Rufus only if there is a single image file inside.

ThomasKaiser, 2016-11-02:
At least Etcher should have support for some sort of a [container format](https://github.com/resin-io/etcher/issues?q=is%3Aopen+is%3Aissue+label%3A%22component%3Aextended+archives%22) currently only supporting ZIP. But maybe it's more efforts trying to get .xz support for these "Extended archives" and switching to this format than dealing with broken decompression/burning regularly?

igorpecovnik, 2016-11-02:
I think this is an isolated event so we don't need to change anything ... yet. A must is a format that uses more than one core to compress in first place. XZ looks promising but can it store more files or they must be tared? So we are talking about tar.xz here?