armbian/build #531: PA0 misconfigured on OrangePi+2

jimgregory, 2016-11-02:
Pin PA0 is misconfigured, so it won't behave as expected in the fex file (e.g., it won't function as a TX pin on UART2). In the fex file, PA0 has the following attributes:

```
$ bin2fex /boot/script.bin | grep PA00
jtag_ms = port:PA00<3><default><default><default>
uart_tx = port:PA00<2><1><default><default>
gpio_pin_6 = port:PA00<1><default><default><0>
```

But, according to sunxi-pio, it's an input pin:

```
$ sudo sunxi-pio -m print | grep -E "^PA[0-4]<.*"
PA0<0><0><0><0>
PA0<2><1><1>
PA1<2><1><1>
PA2<2><1><1>
PA3<2><1><1>
PA4<2><1><1>
```

Resetting the pin to match the settings of PA1 makes it work correctly as UART2_TX:
 
```
$ sudo sunxi-pio -m PA0'<2><1><1>'
$ sudo sunxi-pio -m print | grep -E "^PA[0-4]<.*"
PA0<2><1><1>
PA1<2><1><1>
PA2<2><1><1>
PA3<2><1><1>
PA4<2><1><1>

```

zador-blood-stained, 2016-11-09:
It may be the same problem as with M2+: https://forum.armbian.com/index.php/topic/2114-banana-pi-m2-uart2-only-works-one-way/#entry19343

Will be tested and fixed soon, since detection setting for mmc0 looks wrong to me too on boards with eMMC, and this needs to be checked before pushing the changes.

zador-blood-stained, 2016-11-10:
Should be fixed: f160a06b38bdebc267a357878d584660ffc5d955

closed 2016-11-10 by zador-blood-stained