armbian/build #533: Support for BPI M3

SkutchElectronics, 2016-11-02:
Any idea on when you'll have Armbian ready for Banana Pi M3, I like it over the rest of the distros, but noticed it isn't ready for M3 yet?

zador-blood-stained, 2016-11-02:
AFAIK nobody on our team wants to work with this device. It was discussed [here](http://forum.armbian.com/index.php/topic/474-quick-review-of-banana-pi-m3/) and [here](http://forum.armbian.com/index.php/topic/480-is-a83t-supported-banana-pi-m3/).

Until A83T gets [proper support in mainline kernel](https://linux-sunxi.org/Mainlining_Effort#Status_Matrix) this situation won't change.

ThomasKaiser, 2016-11-02:
http://forum.armbian.com/index.php/topic/474-quick-review-of-banana-pi-m3/ and http://linux-sunxi.org/Banana_Pi_M3 might give some reasons why there's no support for this device.

The hardware is a fail, IO bandwidth is so low, BPi M3 customers seem to be absolutely clueless (focused on specs and ready to believe every fairy tale). And supporting this device would send a wrong signal out. There are maybe 100 other SBC around worth a look before considering Armbian support for this piece of hardware.

closed 2016-11-06 by zador-blood-stained