armbian/build #534: Orange Pi One reboots on serial login

jadr2ddude, 2016-11-03:
For some reason, whenever I log in through the serial port, it immediately starts the shutdown process and reboots. This does not happen with SSH login, so it is not that important. This is happening with the dev kernel.

martinayotte, 2016-11-03:
That is strange !
I didn't faced that before, not even since I've upgraded my OPi-One last week.

svkatielee, 2016-11-04:
Forgive me if this simple, but did you review the RC files in the home directory of the account you are logging in? .profile .bashrc and all

jadr2ddude, 2016-11-04:
The RC files are all default - it is a fresh install. I have not modified anything yet.

igorpecovnik, 2016-11-05:
It could be a corrupted file system / SD card. Make sure you rule out HW related problems. Our script auto reboots if the process of creation default username fails, if the process was broken in the first place and you are trying to create user which already exits ...

jadr2ddude, 2016-11-05:
I will be installing it on 11 more orange pi's, so I will see what happens then.

jadr2ddude, 2016-11-09:
The SD card failed to mount on my laptop. It has been doing that quite a lot for some reason. It might be the SD cards.

jadr2ddude, 2016-11-10:
I made a whole new image on a new sd card and the same thing happened.

zador-blood-stained, 2016-11-10:
@jadr2ddude 
Do you use image from the download page, beta image or self-built image?

jadr2ddude, 2016-11-10:
I built the image, because I needed the dev kernel. I will upload the image later today.

igorpecovnik, 2016-11-10:
Have you used Ubuntu Xenial host?

jadr2ddude, 2016-11-10:
Yes on google cloud with a Xenial VM. No warnings I am aware of.

jadr2ddude, 2016-11-10:
Just uploaded image:
https://transfer.sh/TtGFq/test.img.xz

zador-blood-stained, 2016-11-10:
Works for me:
```

U-Boot SPL 2016.09.01-armbian (Nov 10 2016 - 03:17:45)
DRAM: 512 MiB
Trying to boot from MMC1

U-Boot 2016.09.01-armbian (Nov 10 2016 - 03:17:45 +0000) Allwinner Technology

CPU:   Allwinner H3 (SUN8I 1680)
Model: Xunlong Orange Pi One
DRAM:  512 MiB
MMC:   SUNXI SD/MMC: 0
*** Warning - bad CRC, using default environment

In:    serial
Out:   serial
Err:   serial
Net:   phy interface0
eth0: ethernet@1c30000
Hit any key to stop autoboot:  0
switch to partitions #0, OK
mmc0 is current device
Scanning mmc 0:1...
Found U-Boot script /boot/boot.scr
2092 bytes read in 215 ms (8.8 KiB/s)
## Executing script at 43100000
Booting from SD
114 bytes read in 182 ms (0 Bytes/s)
5129090 bytes read in 487 ms (10 MiB/s)
5996200 bytes read in 602 ms (9.5 MiB/s)
0 bytes read in 149 ms (0 Bytes/s)
20925 bytes read in 756 ms (26.4 KiB/s)
## Loading init Ramdisk from Legacy Image at 43300000 ...
   Image Name:   uInitrd
   Image Type:   ARM Linux RAMDisk Image (gzip compressed)
   Data Size:    5129026 Bytes = 4.9 MiB
   Load Address: 00000000
   Entry Point:  00000000
   Verifying Checksum ... OK
## Flattened Device Tree blob at 43000000
   Booting using the fdt blob at 0x43000000
   Loading Ramdisk to 49b1b000, end 49fff342 ... OK
   Loading Device Tree to 49b12000, end 49b1a1bc ... OK

Starting kernel ...

Uncompressing Linux... done, booting the kernel.
Loading, please wait...
starting version 229
Begin: Loading essential drivers ... done.
Begin: Running /scripts/init-premount ... done.
Begin: Mounting root file system ... Begin: Running /scripts/local-top ... done.
Begin: Running /scripts/local-premount ... Scanning for Btrfs filesystems
done.
Begin: Will now check root file system ... fsck from util-linux 2.27.1
[/sbin/fsck.ext4 (1) -- /dev/mmcblk0p1] fsck.ext4 -a -C0 /dev/mmcblk0p1
/dev/mmcblk0p1: clean, 77733/91968 files, 261951/367616 blocks
done.
done.
Begin: Running /scripts/local-bottom ... done.
Begin: Running /scripts/init-bottom ... done.

Welcome to Ubuntu 16.04.1 LTS!

[  OK  ] Listening on Syslog Socket.
[  OK  ] Created slice System Slice.
[  OK  ] Created slice system-serial\x2dgetty.slice.
[  OK  ] Listening on udev Kernel Socket.
[  OK  ] Started Forward Password Requests to Wall Directory Watch.
[... 467 lines cut ...]
root@orangepione:~# logout

Ubuntu 16.04.1 LTS orangepione ttyS0

orangepione login: qwerty
Password:
  ___                               ____  _    ___
 / _ \ _ __ __ _ _ __   __ _  ___  |  _ \(_)  / _ \ _ __   ___
| | | | '__/ _` | '_ \ / _` |/ _ \ | |_) | | | | | | '_ \ / _ \
| |_| | | | (_| | | | | (_| |  __/ |  __/| | | |_| | | | |  __/
 \___/|_|  \__,_|_| |_|\__, |\___| |_|   |_|  \___/|_| |_|\___|
                       |___/

Welcome to ARMBIAN Ubuntu 16.04.1 LTS 4.9.0-sun8i
System load:   1.03             Up time:       1 min
Memory usage:  4 % of 495Mb     IP:
Usage of /:    9% of 15G

[ 23 updates to install: apt-get upgrade ]

To run a command as administrator (user "root"), use "sudo <command>".
See "man sudo_root" for details.

qwerty@orangepione:~$
```

jadr2ddude, 2016-11-10:
Hmm. Maybe there is a problem with my serial adapter

closed 2016-11-10 by jadr2ddude