armbian/build #536: h3consumption permissions issue

rjbrown99, 2016-11-05:
Hi. The new h3consumption script has a function to disable wifi power management. The idea is by running  "/usr/bin/h3consumption -w off" it will create a script called /etc/NetworkManager/dispatcher.d/99disable-power-management. This is executed upon initial boot by NetworkManager.

Upon boot, I would expect wifi power management to be off. In my case, it is still set to "All packets received" when I run iwconfig wlan0.

When adding debug statements to the 99disable-power-management, I notice that at the end of that script the wireless power is definitely turned off (shows Power Management: off). However, between the end of that script and the completion of the boot process the power management is turned back on.  

Stay tuned for more, I don't yet know why that happens.

ThomasKaiser, 2016-11-07:
Can you please have a look whether https://github.com/igorpecovnik/lib/commit/fe8afb20f0c8d5f62f0d4e6fbc0f6f536faeb97d fixed it? Please note that on existing installations you've to edit `/etc/NetworkManager/dispatcher.d/99disable-power-management` manually so that it reads:

    #!/bin/bash
    case "$2" in
            up) (sleep 15 && /sbin/iwconfig $1 power off || true) & ;;
            down) (sleep 15 && /sbin/iwconfig $1 power on || true) & ;;
    esac

In my tests 10 seconds delay was already sufficient while 5 seconds was too low. Since we're searching for a generic approach that should work with every Wi-Fi implementation out there (even those crappy USB Wi-Fi dongles people seem to love for reasons unknown to me) I thought better save than sorry and increased the delay to 15 seconds.

rjbrown99, 2016-11-07:
Thanks, this seems to work for me as well through a half dozen reboots so far. I spent a ton of time troubleshooting this weekend and never thought to sleep ahead of the power tweak. Much appreciated.

rjbrown99, 2016-11-07:
Also, for what it's worth this feature was also added to NetworkManager directly:
https://www.phoronix.com/scan.php?page=news_item&px=NetworkManager-WiFi-Power-Save

Appears to be in post 1.0 versions of NetworkManager, which is not yet in use in armbian. But might be a new path for handling this in the future (or another thing to account for when managing it).

ThomasKaiser, 2016-11-07:
Thank you for testing. I think this is some sort of misbehaviour in NM now (I also checked whether the scripts are called with a `post-up` case but that's only available with VPNs it seems). Did you test with Jessie or Xenial (important since this package isn't part of Armbian but of the 'real' distribution and we already run into problems that stuff works somewhat differently when using Wheezy or Trusty).

rjbrown99, 2016-11-07:
Originally it was Xenial where I first observed the behavior, but over the weekend I reinstalled with Jessie and that's what I am testing with now. My 'hey it works now' comment earlier in this thread referred to testing on Jessie only and with the 3.x kernel since I can't get the 4.x to boot.

ThomasKaiser, 2016-11-07:
Great, since I tested with Xenial the two important distros are covered. I'll close this now since the fix seems to work. Already on my TODO to test through this with a few more Wi-Fi chips we currently support.

closed 2016-11-07 by ThomasKaiser