armbian/build #537: User specific synced folder in Vagrantfile

popstas, 2016-11-05:
```
config.vm.synced_folder "/home/faddat/git/lib/output", "/vagrant/output"
```

This line breaks `vagrant up`, after comment it script `compile.sh` warns that repository sources was modified. Maybe remove it?

popstas, 2016-11-06:
@zador-blood-stained, sorry for incomplete information, if just comment this mapping, compilation will stop with error such `output: read only file system`. To fix it I created directory `output` in project root and added this mapping:
```
config.vm.synced_folder "./output", "/vagrant/output"
```

I think that you can add empty directory with .gitkeep to repository and then add it to .gitignore, or just add instructions to README.md

Then all errors with this mapping was resolved.

zador-blood-stained, 2016-11-06:
[Documentation](https://www.vagrantup.com/docs/synced-folders/basic_usage.html) says that auto-creation is possible, IMO it will be better than adding an empty directory.

zador-blood-stained, 2016-11-06:
Please test if this works:
```
config.vm.synced_folder "./output", "/vagrant/output", create: true
```

popstas, 2016-11-06:
Confirm, it works.

zador-blood-stained, 2016-11-09:
0db052451c3dfb4d5e03a9732c460108136417aa

commit 998c2f6d (2016-11-06): Remove custom path in Vagrantfile
Fixes #537
closed 2016-11-09 by zador-blood-stained