armbian/build #538: enhancement: default USB OTG to g_ether where applicable

jadr2ddude, 2016-11-10:


WereCatf, 2016-11-11:
Personally, I would rather default to g_serial instead. It doesn't require you to know the IP-address of the board or to install Zeroconf under Windows for, and g_ether doesn't work out-of-the-box with an up-to-date Windows 10-host due to driver-issues.

g_serial, on the other hand, is pretty much plug-and-play, doesn't require you to know any IP-addresses or anything or to even have a working SSH-/telnet-/whatever-server running. It's the safer default. One can always switch to g_ether later on, if they wish.

ThomasKaiser, 2016-11-11:
Currently we enable a serial console through `g_serial` only on the 2 non-Ethernet boards we support (OPi Lite, NanoPi Air). I thought yesterday about adding this to OPi Zero as well since I already forgot to add a serial console there since I was so used to getting console over the USB cable that also powered the board from my MacBook.

But I think we should try to be consistent and as I understood this neither works on every platform nor with every kernel version we support? Please lets keep in mind that Armbian is not an Orange Pi subsidiary but an open project supporting +40 SBC with (sometimes more than) 2 kernel variants each.

I've to admit that I don't know much regarding state of `g_serial` when it's not sun8i legacy. We might collect information first. But at least it's obvious that `g_ether` isn't the way to go by default.

WereCatf, 2016-11-11:
Well, do you mean "consistent" as in "no USB OTG by default at all on any boards" or as in "USB OTG by default on as many boards as possible?"

I do really like having a serial-port over OTG easily-accessible, it's just so much handier than messing with USB TTL-adapters or whatnot, but I don't know how well OTG works across the various boards and whether or not it introduces any unpleasant side-effects.

ThomasKaiser, 2016-11-11:
@WereCatf I'm only thinking about side-effects. @zador-blood-stained reported that with one USB gadget module loaded it might become impossible to load the next one (even unloading the first doesn't do always the job). People might want to use `g_ether` later but then they had to learn to do so they've to remove `g_serial` from `/etc/modules` first and reboot.

Also I've no idea whether this currently works with mainline kernel which would be desirable since otherwise we implement now an enhancement that will not work anymore when throwing the crappy Android kernel away and use mainline instead.

And then I also think about the other boards out there. Though we can also differentiate between 'device classes' we support, claiming that some boards are used as servers (where people do the usual SSH setup) while those targeting IoT (all being small H3 boards right now) need different default behaviour.

My goal was to collect some knowledge first. Do we even know where these USB gadget modules work? At least I only used them with sun8i legacy and no other family or kernel variant.

WereCatf, 2016-11-11:
@ThomasKaiser I just got around to testing USB OTG on my Orange Pi PC with mainline-kernel and it works beautifully, as long as you enable the OTG-stuff in dev-tree and fix musb-initialization. Ie. I just copied the relevant parts from megous's fork and merged them with montjoie's ethernet-fork.

jadr2ddude, 2016-11-11:
@WereCatf Thanks!

I am currently testing it out. I am also going to try to make a ConfigFS setup to consider.

ThomasKaiser, 2016-11-11:
> I just copied the relevant parts from megous's fork and merged them with montjoie's ethernet-fork

I switched back to megi's branch this morning, still using our kernel config:

	KERNELSOURCE='https://github.com/megous/linux'
	KERNELBRANCH='branch:orange-pi-4.9'

I've no freezes, cpufreq scaling works but thermal readouts not (therefore also no throttling). OTG works too (just a quick test using `g_serial`).

jadr2ddude, 2016-11-11:
Hmm. I have modprobed sunxi and g_serial, but dmesg says it couldn't find a UDC. Is there another kernel option I need to enable?

I am using an Orange Pi One, so could it be something in the PC DTB that isn't in the One DTB?

WereCatf, 2016-11-11:
Yes, copy the relevant parts from OPi PCs DTB

Hmm. I have modprobed sunxi and g_serial, but dmesg says it couldn't find a 
UDC. Is there another kernel option I need to enable?

I am using an Orange Pi One, so could it be something in the PC DTB that 
isn't in the One DTB?

jadr2ddude, 2016-11-11:
Still not sure what I am doing wrong. I built for orange pi pc, and ran it on an orange pi pc, but am still getting the same error.

jadr2ddude, 2016-11-11:
Oh wait nvm i forgot to change from one to pc in my copy-files command

jadr2ddude, 2016-11-11:
It is still not working

jadr2ddude, 2016-11-11:
Can you send me your kernel config or image?

jadr2ddude, 2016-11-11:
Ok so this is really weird. All of the modules load and report success in dmesg, but nothing can detect the USB device

jadr2ddude, 2016-11-11:
Nvm it works, the 2 usb cables I tested it with were broken

jadr2ddude, 2016-11-12:
Now networkmanager is complaining about it not being in udev

zador-blood-stained, 2016-12-30:
I think g_serial for selected boards and nothing on other boards will stay as default, as for H3 mainline - it will be tested in the future (and it's not related to the title anyway)

closed 2016-12-30 by zador-blood-stained