armbian/build #547: g_ether in dev kernel doesn't work consistently connected to Linux computers

jadr2ddude, 2016-11-19:
It sometimes does/doesn't work with Debian or Arch. Other distros may have this problem, but those are the two that I have tested. Here are some related dmesg outputs from the host:

http://pastebin.com/s8vWBTzp
http://pastebin.com/vNVsjRUg

The same thing happens with or without the USB hub. I will see if I can find any errors on the Orange Pi that may be causing this.

zador-blood-stained, 2016-11-20:
> I think it is a bug with the kernel running on the Orange Pi.

which is expected since it's a work-in-progress kernel and the only thing we can do is waiting for official release of OTG support and then reporting any issues.

jadr2ddude, 2016-11-20:
Ok so should I open the issue on the repo for the WIP kernel?

closed 2016-11-23 by jadr2ddude