armbian/build #548: Adding SATA Port Multiplier to boot.scr automatic?

LarsNorgaard, 2016-11-20:
Hi,

I've used most of my weekend to figure out why my new JMB321 based SATA Port Multiplier worked fine with Bananaian but not on armbian, even when compiling my own kernel with PMP enabled.

Found out that I needed to edit boot.cmd.

Couldn't the boot script look for vendor id or something and then add the ahci_sunxi line itself?

## FIX:
`nano /boot/boot.cmd`
Go to line that starts with `setenv bootargs` and add `ahci_sunxi.enable_pmp=1` to the end of it.
recompile with `mkimage -C none -A arm -T script -d /boot/boot.cmd /boot/boot.scr`
`reboot`
http://forum.lemaker.org/forum.php?mod=viewthread&tid=9207&page=3#pid93406
Thanks to Kaiser and Paccer.

It's probably not a bug, but I wanted to share my findings.

zador-blood-stained, 2016-11-20:
> Couldn't the boot script look for vendor id or something and then add the ahci_sunxi line itself?

No, I don't think this information is accessible from u-boot (especially via conditional expressions), and even if it was, maintaining a vendor/product table would not be a good idea.

ThomasKaiser, 2016-11-20:
@LarsNorgaard please send a pull request with information how to enable PM support with A20 devices here: https://github.com/igorpecovnik/lib.docs/blob/master/docs/Hardware_Allwinner-A20.md

zador-blood-stained, 2016-11-29:
https://github.com/igorpecovnik/lib.docs/commit/69baae006e7d426460354ad3f7cdd08e5d65bd37

closed 2016-11-29 by zador-blood-stained

hsch0, 2017-09-27:
@LarsNorgaard I'm looking for an image of Armbian with the SATA_Multiplier enabled. can you provide me download link?

LarsNorgaard, 2017-09-28:
@hsch0 I haven't looked into this for a while actually. The PMP board I got gave me sata errors every 4th hour or so, so I stopped using it. 

You're not able to apply anything to boot.cmd ?

ThomasKaiser, 2017-09-28:
Just to inform others stumbling accross this issue and thinking Allwinner SATA especially combined with port multipliers would be worth a try... it's not, Allwinner SATA is slow as hell and when combined with a port multiplier (especially the cheap crap like JMB321) it gets even worse.

Take your time to read through:

- http://linux-sunxi.org/SATA#PMP_support_-_using_SATA_port_multipliers_with_sunxi_devices
- https://forum.armbian.com/index.php?/topic/1925-some-storage-benchmarks-on-sbcs/&do=findComment&comment=34192
- https://forum.armbian.com/index.php?/topic/5199-banana-pi-m2u-wont-boot/&do=findComment&comment=39590

It's 2017 now, please wake up. A20/R40/V40 and PM --> unfortunately you fooled yourself.