armbian/build #549: sunxi-mmc 1c10000.mmc: fatal err update clk timeout on orange pi pc

loqutus, 2016-11-20:
Hi. Having this error when building with the latest dev or stable kernel, then it works for a little while, but when trying to install some packages, then, after a while, `Card stuck in programming state! mmc_do_erase` appears in the code. Could you give some advice how to get some new kernel working? I need at least 3.10 to run docker, and ethernet working. Thanks!

jadr2ddude, 2016-11-20:
I can send you my image with docker in an hour or so

loqutus, 2016-11-20:
thanks, it would be cool!

loqutus, 2016-11-20:
i had it working on some 4.7.something kernel, but lost that image, and can't reproduce the build

jadr2ddude, 2016-11-20:
I am getting this error too, but it is still booting anyway

loqutus, 2016-11-20:
for me it works for a while, and then looks like it stops writing to a card

jadr2ddude, 2016-11-20:
Oh yeah that is systemd. Cant remember how i got past it, but it works

loqutus, 2016-11-20:
oh, cool)

loqutus, 2016-11-20:
btw what distribution it is?

jadr2ddude, 2016-11-20:
Xenial w/ btrfs

loqutus, 2016-11-20:
nice

jadr2ddude, 2016-11-21:
I think this is it (before I installed docker): https://transfer.sh/z3wNC/btrfsopiv5.img.xz

loqutus, 2016-11-21:
it works, thanks!

closed 2016-11-21 by loqutus